coolsnowwolf/lede #13015: 今天中午编译出现的问题

nihaoa15555, 2024-11-15:
### 详细叙述

这个模块似乎有问题？刚更新不久

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

小米cr660x

### 详细日志

make[3] -C toolchain/binutils compile
    ERROR: toolchain/binutils failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 1
WARNING: Makefile 'package/feeds/packages/prometheus-node-exporter-lua/Makefile' has a dependency on 'bmx6', which does not exist
WARNING: Makefile 'package/feeds/packages/prometheus-node-exporter-lua/Makefile' has a dependency on 'bmx6-json', which does not exist
make[2]: Entering directory '/workdir/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/workdir/openwrt/scripts/config'
make[1]: Entering directory '/workdir/openwrt'
make[2]: Entering directory '/workdir/openwrt'
+ mkdir -p /workdir/openwrt/staging_dir/target-mipsel_24kc_musl
+ cd /workdir/openwrt/staging_dir/target-mipsel_24kc_musl
+ mkdir -p bin lib stamp usr/include usr/lib
mkdir -p /workdir/openwrt/build_dir/target-mipsel_24kc_musl/stamp
touch /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/.prepared
make[3]: Entering directory '/workdir/openwrt/toolchain/gdb'
make[3]: Nothing to be done for 'compile'.
make[3]: Leaving directory '/workdir/openwrt/toolchain/gdb'
time: toolchain/gdb/compile#0.00#0.00#0.01
make[3]: Entering directory '/workdir/openwrt/toolchain/binutils'
mkdir -p /workdir/openwrt/dl
SHELL= flock /workdir/openwrt/tmp/.binutils-2.42.tar.xz.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "binutils-2.42.tar.xz" "x" "" "@GNU/binutils/"    '
Cannot find appropriate hash command, ensure the provided hash is either a MD5 or SHA256 checksum.
make[3]: *** [Makefile:112: /workdir/openwrt/dl/binutils-2.42.tar.xz] Error 255
make[3]: Leaving directory '/workdir/openwrt/toolchain/binutils'
time: toolchain/binutils/compile#0.02#0.00#0.03
    ERROR: toolchain/binutils failed to build.
make[2]: *** [toolchain/Makefile:97: toolchain/binutils/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [toolchain/Makefile:[93](https://github.com/nihaoa15555/HI/actions/runs/11851528894/job/33028433214#step:18:94): /workdir/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

guhill1, 2024-11-15:
+1 等待修复

dfbucy090, 2024-11-15:
等待修复

notgp2engine, 2024-11-15:
toolchain/binutils/Makefile加入这段代码
ifeq ($(PKG_VERSION),2.42)
  PKG_HASH:=f6e4d41fd5fc778b06b7891457b3620da5ecea1006c6a4a41ae998109f85a800
endif

hyyz17200, 2024-11-15:
> toolchain/binutils/Makefile加入这段代码 ifeq ($(PKG_VERSION),2.42) PKG_HASH:=f6e4d41fd5fc778b06b7891457b3620da5ecea1006c6a4a41ae998109f85a800 endif

可以了，临时解决 👍

LIUBANGJIAN, 2024-11-15:
原来大伙都同样出现编译问题啊，还以为自己又手贱改了什么。

LIUBANGJIAN, 2024-11-15:
> DIY里添加 sed -i 's/820d9724f020a3e69cb337893a0b63c2db161dadcb0e06fc11dc29eb1e84a32c/f6e4d41fd5fc778b06b7891457b3620da5ecea1006c6a4a41ae998109f85a800/g' toolchain/binutils/Makefile sed -i 's|^820d9724f020a3e69cb337893a0b63c2db161dadcb0e06fc11dc29eb1e84a32c|f6e4d41fd5fc778b06b7891457b3620da5ecea1006c6a4a41ae998109f85a800|' toolchain/binutils/Makefile

不及，做大L大官方修复。

closed 2024-11-15 by coolsnowwolf