coolsnowwolf/lede #13017: x86 docker版本编译不通过，编译了6个小时，都不成功。

lysgwl, 2024-11-16:
### 详细叙述

配置文件文件如下：
[seed.txt](https://github.com/user-attachments/files/17782712/seed.txt)

最终出现的错误如下：
[image]

编译x86 docker版本时，不通过。去除了docker后，就没有问题。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

pve虚拟机设备

### 详细日志

[image]

qingtian110, 2024-11-16:
你都超时了。GitHub action 不适用于插件全家桶、高大全对、内置插件越多越牛逼系列。

lysgwl, 2024-11-16:
> 你都超时了。GitHub action 不适用于插件全家桶、高大全对、内置插件越多越牛逼系列。

问题我找见了，编译没几个插件，问题出在golang的版本上。引用外部插件造成的。再试试看行不行。

lysgwl, 2024-11-17:
> 你都超时了。GitHub action 不适用于插件全家桶、高大全对、内置插件越多越牛逼系列。

编译一晚上，还是出错了。如果不加docker，就没问题，就可以编译通过。以下是错误截图：
[image]

以下配置为不包含docker的配置，就可以编译通过：
[seed.txt](https://github.com/user-attachments/files/17788715/seed.txt)

就奇了怪了，前一阵还可以用，这几天就突然不行了。大神帮我瞅瞅把。

lxl6125, 2024-11-18:
磁盘空间不足，换个工作目录吧，/mnt 这么大的空间为啥不用
[image]

closed 2024-11-20 by coolsnowwolf