coolsnowwolf/lede #13020: 固件刷了不分配IP，WiFi也没有，用TTL把启动的日志读出来了，大佬们修修

Kuron88, 2024-11-16:
### 详细叙述

固件刷了不分配IP，WiFi也没有

用TTL把启动的日志读出来了，大佬们修修

编译的过程是没有报错的，就是固件出来之后直接uboot刷fac固件就不启动，从旧版的保存配置升级sys的那个就可以正常用

uboot大雕的新版2024-4-9号的

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

AX1800 PRO亚瑟

### 详细日志

Format: Log Type - Time(microsec) - Message - Optional Info
Log Type: B - Since Boot(Power On Reset),  D - Delta,  S - Statistic
S - QC_IMAGE_VERSION_STRING=BOOT.XF.0.3-00100-IPQ60xxLZB-1
S - IMAGE_VARIANT_STRING=IPQ6018LA
S - OEM_IMAGE_VERSION_STRING=crm-ubuntu124
S - Boot Interface: eMMC
S - Secure Boot: Off
S - Boot Config @ 0x000a602c = 0x000002e3
S - JTAG ID @ 0x000a607c = 0x001390e1
S - OEM ID @ 0x000a6080 = 0x00000000
S - Serial Number @ 0x000a4128 = 0x3cdbf81e
S - OEM Config Row 0 @ 0x000a4188 = 0x0000000000000000
S - OEM Config Row 1 @ 0x000a4190 = 0x0000000000000000
S - Feature Config Row 0 @ 0x000a4130 = 0x0000800018200021
S - Feature Config Row 1 @ 0x000a4138 = 0x02c3e83783000009
S - PBL Patch Ver: 1
S - I-cache: On
S - D-cache: On
B -      3413 - PBL, Start
B -       592 - bootable_media_detect_entry, Start
B -      4339 - bootable_media_detect_success, Start
B -     91750 - elf_loader_entry, Start
B -     91923 - auth_hash_seg_entry, Start
B -     92904 - auth_hash_seg_exit, Start
B -     93398 - elf_segs_hash_verify_entry, Start
B -    119327 - elf_segs_hash_verify_exit, Start
B -    123517 - auth_xbl_sec_hash_seg_entry, Start
B -    123661 - auth_xbl_sec_hash_seg_exit, Start
B -    130208 - xbl_sec_segs_hash_verify_entry, Start
B -    130209 - xbl_sec_segs_hash_verify_exit, Start
B -    131138 - PBL, End
B -    109617 - SBL1, Start
B -    252540 - GCC [RstStat:0x0, RstDbg:0x600000] WDog Stat : 0x4
B -    254980 - clock_init, Start
D -      3782 - clock_init, Delta
B -    264618 - boot_flash_init, Start
D -     52246 - boot_flash_init, Delta
B -    319335 - sbl1_ddr_set_default_params, Start
D -       244 - sbl1_ddr_set_default_params, Delta
B -    325770 - boot_config_data_table_init, Start
D -      2165 - boot_config_data_table_init, Delta - (575 Bytes)
B -    334890 - CDT Version:2,Platform ID:8,Major ID:3,Minor ID:2,Subtype:1
B -    340441 - Image Load, Start
D -      6618 - OEM_MISC Image Loaded, Delta - (0 Bytes)
B -    349774 - Image Load, Start
D -      5063 - PMIC Image Loaded, Delta - (0 Bytes)
B -    357673 - sbl1_ddr_set_params, Start
B -    362645 - CPR configuration: 0x555
B -    365847 - Pre_DDR_clock_init, Start
D -       213 - Pre_DDR_clock_init, Delta
[... 371 lines cut ...]
[   13.553278] usbcore: registered new interface driver ums-sddr55
[   13.559250] usbcore: registered new interface driver ums-usbat
[   13.566091] usbcore: registered new interface driver usbserial_generic
[   13.569701] usbserial: USB Serial support registered for generic
[   13.601418] xt_time: kernel timezone is -0000
[   13.602768] usbcore: registered new interface driver cdc_mbim
[   13.648303] ath11k c000000.wifi: ipq6018 hw1.0
[   13.648345] ath11k c000000.wifi: FW memory mode: 1
[   13.652014] remoteproc remoteproc0: powering up cd00000.remoteproc
[   13.670817] remoteproc remoteproc0: Booting fw image IPQ6018/q6_fw.mdt, size 7340
[   14.499176] remoteproc remoteproc0: remote processor cd00000.remoteproc is now up
[   14.500463] ath11k c000000.wifi: qmi ignore invalid mem req type 3
[   14.506105] usbcore: registered new interface driver option
[   14.506336] ath11k c000000.wifi: chip_id 0x0 chip_family 0x4 board_id 0xff soc_id 0xffffffff
[   14.511785] usbserial: USB Serial support registered for GSM modem (1-port)
[   14.517220] ath11k c000000.wifi: fw_version 0x250a04b8 fw_build_timestamp 2021-12-20 07:20 fw_build_id WLAN.HK.2.5.0.1-01208-QCAHKSWPL_SILICONZ-1
[   14.532921] usbcore: registered new interface driver qcserial
[   14.545884] usbserial: USB Serial support registered for Qualcomm USB modem
[   14.551850] kmodloader: done loading kernel modules from /etc/modules.d/*
[   14.728812] ath11k c000000.wifi: qmi failed to load CAL data file:cal-ahb-c000000.wifi.bin
[   14.728891] ath11k c000000.wifi: failed to load board data file: -12
[   40.700446] ECM init
[   40.700522] ECM database jhash random seed: 0x2545ccc4
[   40.703082] ECM init complete
[3383]: <msg> ModemManager (version 1.22.0) starting in system bus...

closed 2024-11-16 by Kuron88

xchename, 2025-03-04:
你的问题解决了吗？这边编译成功，刷rec进去，也是不分配地址，网络无法识别

Kuron88, 2025-03-04:
这个大雕修复了

xchename, 2025-03-04:
> 这个大雕修复了

大神，在问下你的luci切换指的时feed源里的切换还是makemenu里的切换？