coolsnowwolf/lede #13032: 云编译hostapd失败

weipeng11, 2024-11-16:
### 详细叙述

云编译时，hostapd编译失败，请大神指点！

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86-64

### 详细日志

[image]

oppen321, 2024-11-16:
加入
sudo -E apt-get -y install clang
试试

coolsnowwolf, 2024-11-17:
看不到日志

weipeng11, 2024-11-17:
> 看不到日志

 make[3] -C feeds/packages/net/trojan compile
 make[3] -C package/network/services/hostapd compile
    ERROR: package/network/services/hostapd failed to build (build variant: wpad-full-openssl).
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/home/runner/work/OpenWrt/OpenWrt/openwrt/include/toplevel.mk:231: world] Error 1
make[2]: Entering directory '/home/runner/work/OpenWrt/OpenWrt/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/runner/work/OpenWrt/OpenWrt/openwrt/scripts/config'
 make[1] world
 make[2] target/compile
 make[3] -C target/linux compile
 make[2] buildinfo
 make[3] diffconfig buildversion feedsversion
 make[4] scripts/config/conf
 make[2] package/cleanup
 make[2] package/compile
 make[3] -C package/libs/libjson-c host-compile
 make[3] -C package/libs/libubox host-compile
 make[3] -C package/system/opkg host-compile
 make[3] -C package/libs/toolchain compile
 make[3] -C package/libs/libnl-tiny compile
 make[3] -C package/libs/libjson-c compile
 make[3] -C package/utils/lua compile
 make[3] -C package/libs/libubox compile
 make[3] -C package/system/ubus compile
 make[3] -C package/system/uci compile
 make[3] -C package/network/config/netifd compile
 make[3] -C package/firmware/cypress-firmware compile
 make[3] -C package/firmware/linux-firmware compile
 make[3] -C package/kernel/linux compile
 make[3] -C package/system/ubox compile
 make[3] -C package/libs/ncurses host-compile
 make[3] -C package/libs/zlib compile
 make[3] -C package/libs/ncurses compile
 make[3] -C package/utils/util-linux compile
 make[3] -C package/system/fstools compile
 make[3] -C package/system/fwtool host-compile
 make[3] -C package/system/fwtool compile
 make[3] -C package/system/usign host-compile
 make[3] -C package/system/ucert host-compile
 make[3] -C package/utils/jsonfilter compile
 make[3] -C package/system/openwrt-keyring compile
 make[3] -C package/system/usign compile
 make[3] -C package/base-files compile
 make[3] -C package/boot/grub2 host-compile
 make[3] -C package/boot/grub2 compile
 make[3] -C package/boot/grub2 compile
 make[3] -C package/boot/grub2 compile
 make[3] -C feeds/luci/modules/luci-base host-compile
 make[3] -C package/kernel/cryptodev-linux compile
 make[3] -C package/libs/openssl compile
 make[3] -C package/libs/ustream-ssl compile
 make[3] -C package/libs/uclient compile
 make[3] -C package/system/opkg compile
 make[3] -C package/utils/lua host-compile
 make[3] -C feeds/luci/applications/luci-app-opkg compile
 make[3] -C package/utils/ucode compile
 make[3] -C feeds/luci/contrib/package/lucihttp compile
 make[3] -C package/network/utils/iwinfo compile
 make[3] -C package/system/rpcd compile
 make[3] -C feeds/luci/libs/rpcd-mod-luci compile
 make[3] -C feeds/luci/contrib/package/ucode-mod-html compile
 make[3] -C feeds/packages/net/cgi-io compile
 make[3] -C feeds/luci/modules/luci-base compile
 make[3] -C feeds/luci/libs/luci-lib-ip compile
 make[3] -C feeds/luci/libs/luci-lib-jsonc compile
 make[3] -C feeds/luci/libs/luci-lib-nixio compile
 make[3] -C feeds/luci/libs/luci-lib-base compile
[... 472 lines cut ...]
patching file src/radius/radius_das.h
patching file src/ap/hostapd.c
patching file src/radius/radius_das.c
Applying ./patches/800-acs-don-t-select-indoor-channel-on-outdoor-operation.patch using plaintext: 
patching file src/ap/acs.c
patching file src/ap/dfs.c
Applying ./patches/990-ctrl-make-WNM_AP-functions-dependant-on-CONFIG_AP.patch using plaintext: 
patching file wpa_supplicant/ctrl_iface.c
Applying ./patches/992-openssl-include-rsa.patch using plaintext: 
patching file src/crypto/crypto_openssl.c
Applying ./patches/993-fix-mbo-module-build.patch using plaintext: 
patching file wpa_supplicant/Makefile
Hunk #1 FAILED at 339.
1 out of 1 hunk FAILED -- saving rejects to file wpa_supplicant/Makefile.rej
Patch failed!  Please fix ./patches/993-fix-mbo-module-build.patch!
make[3]: *** [Makefile:790: /home/runner/work/OpenWrt/OpenWrt/openwrt/build_dir/target-x86_64_musl/hostapd-wpad-full-openssl/hostapd-2022-07-29-b704dc72/.prepared_a5886a1d314a65f1ff8cc7826fd90ae8_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/home/runner/work/OpenWrt/OpenWrt/openwrt/package/network/services/hostapd'
time: package/network/services/hostapd/wpad-full-openssl/compile#0.68#0.46#0.98
    ERROR: package/network/services/hostapd failed to build (build variant: wpad-full-openssl).
make[2]: *** [package/Makefile:116: package/network/services/hostapd/compile] Error 1
make[2]: Leaving directory '/home/runner/work/OpenWrt/OpenWrt/openwrt'
make[1]: *** [package/Makefile:110: /home/runner/work/OpenWrt/OpenWrt/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/runner/work/OpenWrt/OpenWrt/openwrt'
make: *** [/home/runner/work/OpenWrt/OpenWrt/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

weipeng11, 2024-11-17:
> 加入 sudo -E apt-get -y install clang 试试

谢谢回复，问题依旧。

coolsnowwolf, 2024-11-17:
验证了，没问题

closed 2024-11-19 by coolsnowwolf