coolsnowwolf/lede #13046: 太乙开启firewall4时qca nss ecm编译报错

ftkey, 2024-11-17:
### 详细叙述

/home/runner/work/WRT-CI-IPQ60XX/WRT-CI-IPQ60XX/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/qca-nss-ecm-2023-10-20-82b27915/ecm_conntrack_notifier.c:404:19: note: (near initialization for 'ecm_conntrack_notifier.ct_event')
/home/runner/work/WRT-CI-IPQ60XX/WRT-CI-IPQ60XX/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/qca-nss-ecm-2023-10-20-82b27915/ecm_conntrack_notifier.c: In function 'ecm_conntrack_notifier_exit':
/home/runner/work/WRT-CI-IPQ60XX/WRT-CI-IPQ60XX/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/qca-nss-ecm-2023-10-20-82b27915/ecm_conntrack_notifier.c:490:9: error: too many arguments to function 'nf_conntrack_unregister_notifier'
  490 |         nf_conntrack_unregister_notifier(&init_net, &ecm_conntrack_notifier);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ./include/net/netfilter/nf_conntrack_core.h:18,
                 from /home/runner/work/WRT-CI-IPQ60XX/WRT-CI-IPQ60XX/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/qca-nss-ecm-2023-10-20-82b27915/ecm_conntrack_notifier.c:58:
./include/net/netfilter/nf_conntrack_ecache.h:76:6: note: declared here
   76 | void nf_conntrack_unregister_notifier(struct net *net);
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[5]: *** [scripts/Makefile.build:250: /home/runner/work/WRT-CI-IPQ60XX/WRT-CI-IPQ60XX/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/qca-nss-ecm-2023-10-20-82b27915/ecm_conntrack_notifier.o] Error 1
make[4]: *** [Makefile:2009: /home/runner/work/WRT-CI-IPQ60XX/WRT-CI-IPQ60XX/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/qca-nss-ecm-2023-10-20-82b27915] Error 2
make[3]: *** [Makefile:164: /home/runner/work/WRT-CI-IPQ60XX/WRT-CI-IPQ60XX/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/qca-nss-ecm-2023-10-20-82b27915/.built] Error 2
make[4]: Leaving directory '/home/runner/work/WRT-CI-IPQ60XX/WRT-CI-IPQ60XX/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/linux-6.1.112'
make[3]: Leaving directory '/home/runner/work/WRT-CI-IPQ60XX/WRT-CI-IPQ60XX/openwrt/package/qca/qca-nss-ecm'
time: package/qca/qca-nss-ecm/compile#0.91#0.27#1.16
    ERROR: package/qca/qca-nss-ecm failed to build.
make[2]: *** [package/Makefile:116: package/qca/qca-nss-ecm/compile] Error 1
make[2]: Leaving directory '/home/runner/work/WRT-CI-IPQ60XX/WRT-CI-IPQ60XX/openwrt'
make[1]: *** [package/Makefile:110: /home/runner/work/WRT-CI-IPQ60XX/WRT-CI-IPQ60XX/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/runner/work/WRT-CI-IPQ60XX/WRT-CI-IPQ60XX/openwrt'
make: *** [/home/runner/work/WRT-CI-IPQ60XX/WRT-CI-IPQ60XX/openwrt/include/toplevel.mk:231: world] Error 2

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

太乙

### 详细日志

/home/runner/work/WRT-CI-IPQ60XX/WRT-CI-IPQ60XX/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/qca-nss-ecm-2023-10-20-82b27915/ecm_conntrack_notifier.c:404:19: note: (near initialization for 'ecm_conntrack_notifier.ct_event')
/home/runner/work/WRT-CI-IPQ60XX/WRT-CI-IPQ60XX/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/qca-nss-ecm-2023-10-20-82b27915/ecm_conntrack_notifier.c: In function 'ecm_conntrack_notifier_exit':
/home/runner/work/WRT-CI-IPQ60XX/WRT-CI-IPQ60XX/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/qca-nss-ecm-2023-10-20-82b27915/ecm_conntrack_notifier.c:490:9: error: too many arguments to function 'nf_conntrack_unregister_notifier'
  490 |         nf_conntrack_unregister_notifier(&init_net, &ecm_conntrack_notifier);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ./include/net/netfilter/nf_conntrack_core.h:18,
                 from /home/runner/work/WRT-CI-IPQ60XX/WRT-CI-IPQ60XX/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/qca-nss-ecm-2023-10-20-82b27915/ecm_conntrack_notifier.c:58:
./include/net/netfilter/nf_conntrack_ecache.h:76:6: note: declared here
   76 | void nf_conntrack_unregister_notifier(struct net *net);
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[5]: *** [scripts/Makefile.build:250: /home/runner/work/WRT-CI-IPQ60XX/WRT-CI-IPQ60XX/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/qca-nss-ecm-2023-10-20-82b27915/ecm_conntrack_notifier.o] Error 1
make[4]: *** [Makefile:2009: /home/runner/work/WRT-CI-IPQ60XX/WRT-CI-IPQ60XX/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/qca-nss-ecm-2023-10-20-82b27915] Error 2
make[3]: *** [Makefile:164: /home/runner/work/WRT-CI-IPQ60XX/WRT-CI-IPQ60XX/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/qca-nss-ecm-2023-10-20-82b27915/.built] Error 2
make[4]: Leaving directory '/home/runner/work/WRT-CI-IPQ60XX/WRT-CI-IPQ60XX/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/linux-6.1.112'
make[3]: Leaving directory '/home/runner/work/WRT-CI-IPQ60XX/WRT-CI-IPQ60XX/openwrt/package/qca/qca-nss-ecm'
time: package/qca/qca-nss-ecm/compile#0.91#0.27#1.16
    ERROR: package/qca/qca-nss-ecm failed to build.
make[2]: *** [package/Makefile:116: package/qca/qca-nss-ecm/compile] Error 1
make[2]: Leaving directory '/home/runner/work/WRT-CI-IPQ60XX/WRT-CI-IPQ60XX/openwrt'
make[1]: *** [package/Makefile:110: /home/runner/work/WRT-CI-IPQ60XX/WRT-CI-IPQ60XX/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/runner/work/WRT-CI-IPQ60XX/WRT-CI-IPQ60XX/openwrt'
make: *** [/home/runner/work/WRT-CI-IPQ60XX/WRT-CI-IPQ60XX/openwrt/include/toplevel.mk:231: world] Error 2

coolsnowwolf, 2024-11-17:
CONFIG_IPV6 要选上

closed 2024-11-20 by coolsnowwolf

jw10126121, 2024-12-13:
确认CONFIG_IPV6=y配置有加上，但还是出现这个问题，不知道怎么搞了

ftkey, 2024-12-13:
> 确认CONFIG_IPV6=y配置有加上，但还是出现这个问题，不知道怎么搞了

不要刻意删除默认配置, 先加上这个配置项CONFIG_TARGET_PER_DEVICE_ROOTFS=y  .
这个参数加上去,默认配置 就算写上=n也会等同于=m ,编译试试

jw10126121, 2024-12-13:
> > 确认CONFIG_IPV6=y配置有加上，但还是出现这个问题，不知道怎么搞了
> 
> 不要刻意删除默认配置, 先加上这个配置项CONFIG_TARGET_PER_DEVICE_ROOTFS=y . 这个参数加上去,默认配置 就算写上=n也会等同于=m ,编译试试

非常感谢，我正是借鉴了您的云编译，CONFIG_TARGET_PER_DEVICE_ROOTFS=y是有加的，我再排查一下，再次感谢。

ftkey, 2024-12-13:
> > > 确认CONFIG_IPV6=y配置有加上，但还是出现这个问题，不知道怎么搞了
> > 
> > 
> > 不要刻意删除默认配置, 先加上这个配置项CONFIG_TARGET_PER_DEVICE_ROOTFS=y . 这个参数加上去,默认配置 就算写上=n也会等同于=m ,编译试试
> 
> 非常感谢，我正是借鉴了您的云编译，CONFIG_TARGET_PER_DEVICE_ROOTFS=y是有加的，我再排查一下，再次感谢。

那你再重新fork一下吧， 都正常编译的

jw10126121, 2024-12-17:
> > > > 确认CONFIG_IPV6=y配置有加上，但还是出现这个问题，不知道怎么搞了
> > > 
> > > 
> > > 不要刻意删除默认配置, 先加上这个配置项CONFIG_TARGET_PER_DEVICE_ROOTFS=y . 这个参数加上去,默认配置 就算写上=n也会等同于=m ,编译试试
> > 
> > 
> > 非常感谢，我正是借鉴了您的云编译，CONFIG_TARGET_PER_DEVICE_ROOTFS=y是有加的，我再排查一下，再次感谢。
> 
> 那你再重新fork一下吧， 都正常编译的

经过插件一个一个对比，发现是
CONFIG_PACKAGE_luci-app-ipsec-vpnd=y
CONFIG_PACKAGE_luci-i18n-ipsec-vpnd-zh-cn=y
这个插件，设置成y，就会出现这个错误。

ftkey, 2024-12-17:
> 经过插件一个一个对比，发现是 CONFIG_PACKAGE_luci-app-ipsec-vpnd=y CONFIG_PACKAGE_luci-i18n-ipsec-vpnd-zh-cn=y 这个插件，设置成y，就会出现这个错误。

刚刚试了下 确实这个插件不兼容.

jw10126121, 2024-12-17:
> 刚刚试了下 确实这个插件不兼容.

不知道是不是跟其他插件冲突了

ftkey, 2024-12-17:
> 不知道是不是跟其他插件冲突了

这种软件直接放弃吧,  openvpn wg都比这个强 好用

jw10126121, 2024-12-17:
好的，我是看您的配置里面有这个，才加上的，哈哈哈