coolsnowwolf/lede #13071: 拉取最新源码编译报错

bi4tms, 2024-11-17:
### 详细叙述

拉取最新源码编译报错

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

R2S/R4S/R66S

### 详细日志

Applying /home/r4s.6.6/lede/target/linux/rockchip/patches-6.6/205-rockchip-rk3328-add-support-for-FriendlyARM-NanoPi-Neo3.patch using plaintext: 
patching file arch/arm64/boot/dts/rockchip/Makefile
Hunk #1 FAILED at 17.
1 out of 1 hunk FAILED -- saving rejects to file arch/arm64/boot/dts/rockchip/Makefile.rej
patching file arch/arm64/boot/dts/rockchip/rk3328-nanopi-neo3.dts
Patch failed!  Please fix /home/r4s.6.6/lede/target/linux/rockchip/patches-6.6/205-rockchip-rk3328-add-support-for-FriendlyARM-NanoPi-Neo3.patch!
make[4]: *** [Makefile:34: /home/r4s.6.6/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.6.61/.prepared_b0b866533f34dda4429dde5ebdcb1925] Error 1
make[4]: Leaving directory '/home/r4s.6.6/lede/target/linux/rockchip'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/r4s.6.6/lede/target/linux'
time: target/linux/compile#17.15#11.18#22.87
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/home/r4s.6.6/lede'
make[1]: *** [target/Makefile:23: /home/r4s.6.6/lede/staging_dir/target-aarch64_generic_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/r4s.6.6/lede'
make: *** [/home/r4s.6.6/lede/include/toplevel.mk:231：world] 错误 2

yanxin152133, 2024-11-18:
https://github.com/coolsnowwolf/lede/issues/9039

oppen321, 2024-11-18:
看你的报错信息应该是205-rockchip-rk3328-add-support-for-FriendlyARM-NanoPi-Neo3.patch的问题

ccjcu, 2024-11-18:
不是R2S Neo3, 先把Neo3删了，这样可以编译
find . -name 205-rockchip-rk3328-add-support-for-FriendlyARM-NanoPi-Neo3.patch | xargs rm

oppen321, 2024-11-18:
移除Neo3补丁文件
find . -name 205-rockchip-rk3328-add-support-for-FriendlyARM-NanoPi-Neo3.patch | xargs rm
移除neo3
sed -i '/^define Device/friendlyarm_nanopi-neo3$/,/^endef$/d;/^TARGET_DEVICES += friendlyarm_nanopi-neo3$/d' target/linux/rockchip/image/armv8.mk

closed 2024-11-18 by coolsnowwolf