coolsnowwolf/lede #13074: RK3588编译报错

qq761861, 2024-11-18:
### 详细叙述

RK35588编译报错

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

RK3588

### 详细日志

[image]
Applying /home/user/lede/target/linux/rockchip/patches-6.6/106-rockchip-rock-pi-4.patch using plaintext:
patching file arch/arm64/boot/dts/rockchip/Makefile
Hunk #1 succeeded at 39 (offset 1 line).
Hunk #2 succeeded at 62 (offset 1 line).
patching file arch/arm64/boot/dts/rockchip/rk3399-rock-pi-4.dts

Applying /home/user/lede/target/linux/rockchip/patches-6.6/107-mmc-core-set-initial-signal-voltage-on-power-off.patch using plaintext:
patching file drivers/mmc/core/core.c

Applying /home/user/lede/target/linux/rockchip/patches-6.6/110-arm64-rk3568-update-gicv3-its-and-pci-msi-map.patch using plaintext:
patching file arch/arm64/boot/dts/rockchip/rk3568.dtsi
patching file arch/arm64/boot/dts/rockchip/rk356x.dtsi

Applying /home/user/lede/target/linux/rockchip/patches-6.6/111-irqchip-gic-v3-add-hackaround-for-rk3568-its.patch using plaintext:
patching file drivers/irqchip/irq-gic-common.h
patching file drivers/irqchip/irq-gic-v3-its.c

Applying /home/user/lede/target/linux/rockchip/patches-6.6/112-arm64-dts-rockchip-rk3568-Add-xpcs-support.patch using plaintext:
patching file arch/arm64/boot/dts/rockchip/rk3568.dtsi
patching file arch/arm64/boot/dts/rockchip/rk356x.dtsi

Applying /home/user/lede/target/linux/rockchip/patches-6.6/113-ethernet-stmicro-stmmac-Add-SGMII-QSGMII-support.patch using plaintext:
patching file drivers/net/ethernet/stmicro/stmmac/dwmac-rk.c

Applying /home/user/lede/target/linux/rockchip/patches-6.6/114-arm64-dts-rockchip-Update-LED-properties-for-Orange-.patch using plaintext:
patching file arch/arm64/boot/dts/rockchip/rk3328-orangepi-r1-plus.dts

Applying /home/user/lede/target/linux/rockchip/patches-6.6/115-arm64-dts-rockchip-add-LED-configuration-to-Orange-P.patch using plaintext:
patching file arch/arm64/boot/dts/rockchip/rk3328-orangepi-r1-plus.dts

Applying /home/user/lede/target/linux/rockchip/patches-6.6/201-rockchip-rk3328-add-i2c0-controller-for-nanopi-r2s.patch using plaintext:
patching file arch/arm64/boot/dts/rockchip/rk3328-nanopi-r2s.dts

Applying /home/user/lede/target/linux/rockchip/patches-6.6/205-rockchip-rk3328-add-support-for-FriendlyARM-NanoPi-Neo3.patch using plaintext:
patching file arch/arm64/boot/dts/rockchip/Makefile
Hunk #1 FAILED at 17.
1 out of 1 hunk FAILED -- saving rejects to file arch/arm64/boot/dts/rockchip/Makefile.rej
patching file arch/arm64/boot/dts/rockchip/rk3328-nanopi-neo3.dts
Patch failed!  Please fix /home/user/lede/target/linux/rockchip/patches-6.6/205-rockchip-rk3328-add-support-for-FriendlyARM-NanoPi-Neo3.patch!
make[4]: *** [Makefile:34: /home/user/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.6.61/.prepared_b96e7e3dfa273851a02dce1a4eb20b51] Error 1
make[4]: Leaving directory '/home/user/lede/target/linux/rockchip'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/user/lede/target/linux'
time: target/linux/compile#14.90#21.07#31.67
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/home/user/lede'
make[1]: *** [target/Makefile:23: /home/user/lede/staging_dir/target-aarch64_generic_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/user/lede'
make: *** [/home/user/lede/include/toplevel.mk:231: world] Error 2

yanxin152133, 2024-11-18:
是不是缺少这个rk3328-nanopi-neo3.dts，参考下下面这个。
https://github.com/coolsnowwolf/lede/issues/5911

Travel2Here, 2024-11-18:
rk3399-nanopi-r4s.dts 也没有～遇到的错误性质同楼主

oppen321, 2024-11-18:
移除Neo3补丁文件
find . -name 205-rockchip-rk3328-add-support-for-FriendlyARM-NanoPi-Neo3.patch | xargs rm
移除neo3
sed -i '/^define Device\/friendlyarm_nanopi-neo3$/,/^endef$/d;/^TARGET_DEVICES \+= friendlyarm_nanopi-neo3$/d' target/linux/rockchip/image/armv8.mk

closed 2024-11-18 by coolsnowwolf