coolsnowwolf/lede #13075: json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)   x64

d975xbx, 2024-11-18:
### 详细叙述

x64固件正常输出，mt6000没有这样的报错

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x64

### 详细日志

WORK_DIR=/home/parallels/x64/build_dir/target-x86_64_musl/json_info_files /home/parallels/x64/scripts/json_overview_image_info.py /home/parallels/x64/bin/targets/x86/64/profiles.json 
Traceback (most recent call last):
  File "/home/parallels/x64/scripts/json_overview_image_info.py", line 35, in <module>
    output = get_initial_output(image_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/parallels/x64/scripts/json_overview_image_info.py", line 25, in get_initial_output
    profiles = json.loads(output_path.read_text())
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)
make[2]: *** [Makefile:102: /home/parallels/x64/bin/targets/x86/64/profiles.json] Error 1
make[2]: Leaving directory '/home/parallels/x64'
make[1]: *** [Makefile:130: world] Error 2
make[1]: Leaving directory '/home/parallels/x64'
make: *** [/home/parallels/x64/include/toplevel.mk:231：world] 错误 2
[image]
[image]

coolsnowwolf, 2024-11-19:
固件体积设置过大了

d975xbx, 2024-11-19:
> 固件体积设置过大了

默认的448M啊，是勾选的东西太多了？

closed 2024-11-20 by coolsnowwolf