coolsnowwolf/lede #13078: R4S 编译失败

BenjaminX, 2024-11-18:
### 详细叙述

最后的 mark install 阶段

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

R4S

### 详细日志

2024-11-18T14:34:36.3072810Z Configuring ppp-mod-pppoe.
2024-11-18T14:34:36.3073352Z Configuring luci-i18n-nlbwmon-zh-cn.
2024-11-18T14:34:36.3073877Z Configuring arp-scan.
2024-11-18T14:34:36.3074305Z Configuring rkbin-rk3399.
2024-11-18T14:34:36.3074835Z make[2]: Leaving directory '/workdir/openwrt'
2024-11-18T14:34:36.3076492Z make[1]: *** [package/Makefile:111: /workdir/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_install] Error 2
2024-11-18T14:34:36.3077645Z make[1]: Leaving directory '/workdir/openwrt'
2024-11-18T14:34:36.3080752Z make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
2024-11-18T14:34:36.3099773Z ##[error]Process completed with exit code 2.

Travel2Here, 2024-11-19:
+1，清过dir和配置，全新依然同上

coolsnowwolf, 2024-11-19:
日志呢

BenjaminX, 2024-11-19:
> 日志呢

[logs_31043831399.zip](https://github.com/user-attachments/files/17809683/logs_31043831399.zip)

coolsnowwolf, 2024-11-19:
2024-11-19T02:38:24.2311950Z  * check_data_file_clashes: Package nano-plus wants to install file /workdir/openwrt/build_dir/target-aarch64_generic_musl/root-rockchip/usr/bin/nano
2024-11-19T02:38:24.2313221Z 	But that file is already provided by package  * nano
2024-11-19T02:38:24.2313764Z  * opkg_install_cmd: Cannot install package nano-plus.

coolsnowwolf, 2024-11-19:
明显是选了互相冲突的包

BenjaminX, 2024-11-19:
> 明显是选了互相冲突的包

谢谢 LEDE 哒，我去 close

closed 2024-11-19 by BenjaminX