coolsnowwolf/lede #13099: 难道就我一个人有PHP7的编译错误吗？

qqsir-dev, 2024-11-20:
### 详细叙述

PHP7编译错误

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

软路由云编译

### 详细日志

make[3] -C feeds/packages/lang/php7 compile
    ERROR: package/feeds/packages/php7 failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/home/runner/work/AutoLede/AutoLede/openwrt/include/toplevel.mk:231: world] Error 1
WARNING: Makefile 'package/luci-app-serverchan/Makefile' has a dependency on 'wrtbwmon', which does not exist
make[2]: Entering directory '/home/runner/work/AutoLede/AutoLede/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/runner/work/AutoLede/AutoLede/openwrt/scripts/config'
make[1]: Entering directory '/home/runner/work/AutoLede/AutoLede/openwrt'
make[2]: Entering directory '/home/runner/work/AutoLede/AutoLede/openwrt'
make[3]: Entering directory '/home/runner/work/AutoLede/AutoLede/openwrt/target/linux'
make[4]: Entering directory '/home/runner/work/AutoLede/AutoLede/openwrt/target/linux/x86'
rm -f /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.62/localversion
/home/runner/work/AutoLede/AutoLede/openwrt/scripts/kconfig.pl  + + /home/runner/work/AutoLede/AutoLede/openwrt/target/linux/generic/config-6.6 /home/runner/work/AutoLede/AutoLede/openwrt/target/linux/x86/config-6.6 /home/runner/work/AutoLede/AutoLede/openwrt/target/linux/x86/64/config-6.6 > /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.62/.config.target
awk '/^(#[[:space:]]+)?CONFIG_KERNEL/{sub("CONFIG_KERNEL_","CONFIG_");print}' /home/runner/work/AutoLede/AutoLede/openwrt/.config >> /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.62/.config.target
echo "# CONFIG_KALLSYMS_EXTRA_PASS is not set" >> /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.62/.config.target
echo "# CONFIG_KALLSYMS_ALL is not set" >> /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.62/.config.target
echo "CONFIG_KALLSYMS_UNCOMPRESSED=y" >> /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.62/.config.target
/home/runner/work/AutoLede/AutoLede/openwrt/scripts/package-metadata.pl kconfig /home/runner/work/AutoLede/AutoLede/openwrt/tmp/.packageinfo /home/runner/work/AutoLede/AutoLede/openwrt/.config 6.6 > /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.62/.config.override
/home/runner/work/AutoLede/AutoLede/openwrt/scripts/kconfig.pl 'm+' '+' /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.62/.config.target /dev/null /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.62/.config.override > /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.62/.config.set
mv /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.62/.config.set /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.62/.config.old
grep -v INITRAMFS /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.62/.config.old > /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.62/.config.set
echo 'CONFIG_INITRAMFS_SOURCE=""' >> /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.62/.config.set
echo '# CONFIG_INITRAMFS_FORCE is not set' >> /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.62/.config.set
echo "# CONFIG_INITRAMFS_PRESERVE_MTIME is not set" >> /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.62/.config.set
rm -rf /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/modules
cmp -s /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.62/.config.set /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.62/.config.prev || { cp /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.62/.config.set /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.62/.config; cp /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.62/.config.set /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.62/.config.prev; }
export MAKEFLAGS= ; [ -d /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.62/user_headers ] || make -C /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.62 KCFLAGS="-fmacro-prefix-map=/home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl=target-x86_64_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/runner/work/AutoLede/AutoLede/openwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Tue Nov 19 04:30:32 2024" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/runner/work/AutoLede/AutoLede/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=6.6.62 CC="x86_64-openwrt-linux-musl-gcc"  INSTALL_HDR_PATH=/home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.62/user_headers headers_install
grep '=[ym]' /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.62/.config.set | LC_ALL=C sort | /home/runner/work/AutoLede/AutoLede/openwrt/staging_dir/host/bin/mkhash md5 > /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.62/.vermagic
touch /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.62/.configured
rm -f /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.62/vmlinux /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.62/System.map
make -C /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.62 KCFLAGS="-fmacro-prefix-map=/home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl=target-x86_64_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/runner/work/AutoLede/AutoLede/openwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Tue Nov 19 04:30:32 2024" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/runner/work/AutoLede/AutoLede/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=6.6.62 CC="x86_64-openwrt-linux-musl-gcc" olddefconfig
make[5]: Entering directory '/home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.62'
#
# No change to .config
#
make[5]: Leaving directory '/home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.62'
make -C /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.62 KCFLAGS="-fmacro-prefix-map=/home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl=target-x86_64_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/runner/work/AutoLede/AutoLede/openwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Tue Nov 19 04:30:32 2024" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/runner/work/AutoLede/AutoLede/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=6.6.62 CC="x86_64-openwrt-linux-musl-gcc" bzImage modules
make[5]: Entering directory '/home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.62'
  DESCEND objtool
  INSTALL libsubcmd_headers
[... cut ...]
  LD      .tmp_vmlinux.kallsyms1
  NM      .tmp_vmlinux.kallsyms1.syms
  KSYMS   .tmp_vmlinux.kallsyms1.S
  AS      .tmp_vmlinux.kallsyms1.o
  LD      .tmp_vmlinux.kallsyms2
  NM      .tmp_vmlinux.kallsyms2.syms
  KSYMS   .tmp_vmlinux.kallsyms2.S
  AS      .tmp_vmlinux.kallsyms2.o
  LD      vmlinux
  NM      System.map
  SORTTAB vmlinux
  VOFFSET arch/x86/boot/compressed/../voffset.h
  CC      arch/x86/boot/compressed/misc.o
  OBJCOPY arch/x86/boot/compressed/vmlinux.bin
  XZKERN  arch/x86/boot/compressed/vmlinux.bin.xz

qqsir-dev, 2024-11-20:
/usr/include/bits/socket2.h:61:1: note: previous definition of 'recvfrom' with type 'ssize_t(int,  void * restrict,  size_t,  int,  __SOCKADDR_ARG,  socklen_t * restrict)' {aka 'long int(int,  void * restrict,  long unsigned int,  int,  __SOCKADDR_ARG,  unsigned int * restrict)'}
   61 | recvfrom (int __fd, void *__restrict __buf, size_t __n, int __flags,
      | ^~~~~~~~
In file included from /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/php-7.4.30/Zend/zend_virtual_cwd.h:31,
                 from /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/php-7.4.30/main/php.h:439,
                 from /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/php-7.4.30/ext/sqlite3/sqlite3.c:23:
/usr/include/utime.h:39:5: error: unknown type name '__time64_t'
   39 |     __time64_t actime;          /* Access time.  */
      |     ^~~~~~~~~~
/usr/include/utime.h:40:5: error: unknown type name '__time64_t'
   40 |     __time64_t modtime;         /* Modification time.  */
      |     ^~~~~~~~~~
In file included from /usr/include/stdio.h:36,
                 from /home/runner/work/AutoLede/AutoLede/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify/stdio.h:22,
                 from /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/php-7.4.30/Zend/zend_portability.h:49,
                 from /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/php-7.4.30/Zend/zend_types.h:25,
                 from /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/php-7.4.30/Zend/zend.h:27,
                 from /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/php-7.4.30/main/php.h:33,
                 from /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/php-7.4.30/ext/sqlite3/sqlite3.c:23:
/home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/php-7.4.30/ext/sqlite3/sqlite3.c: In function 'php_sqlite3_error':
/home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/php-7.4.30/ext/sqlite3/sqlite3.c:50:18: warning: passing argument 1 of '__builtin_va_start' makes pointer from integer without a cast [-Wint-conversion]
   50 |         va_start(arg, format);
      |                  ^~~
      |                  |
      |                  va_list {aka int}
/home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/php-7.4.30/ext/sqlite3/sqlite3.c:50:18: note: expected '__va_list_tag *' but argument is of type 'va_list' {aka 'int'}
/home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/php-7.4.30/ext/sqlite3/sqlite3.c:52:16: warning: passing argument 1 of '__builtin_va_end' makes pointer from integer without a cast [-Wint-conversion]
   52 |         va_end(arg);
      |                ^~~
      |                |
      |                va_list {aka int}
/home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/php-7.4.30/ext/sqlite3/sqlite3.c:52:16: note: expected '__va_list_tag *' but argument is of type 'va_list' {aka 'int'}
make[4]: *** [Makefile:568: ext/sqlite3/sqlite3.lo] Error 1
make[4]: Leaving directory '/home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/php-7.4.30'
make[3]: *** [Makefile:621: /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/php-7.4.30/.built] Error 2
make[3]: Leaving directory '/home/runner/work/AutoLede/AutoLede/openwrt/feeds/packages/lang/php7'
time: package/feeds/packages/php7/compile#0.41#0.12#0.49
    ERROR: package/feeds/packages/php7 failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/php7/compile] Error 1
make[2]: Leaving directory '/home/runner/work/AutoLede/AutoLede/openwrt'
make[1]: *** [package/Makefile:110: /home/runner/work/AutoLede/AutoLede/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/runner/work/AutoLede/AutoLede/openwrt'
make: *** [/home/runner/work/AutoLede/AutoLede/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

qqsir-dev, 2024-11-20:
难道又是gcc版本不行？

qqsir-dev, 2024-11-22:
@coolsnowwolf 救命，这个怎么办呀

coolsnowwolf, 2024-11-22:
就你一个人

qqsir-dev, 2024-11-22:
真服了，好几天了，之前也没动哪，难道是云编译的环境有什么变化？

binge8, 2024-11-22:
还有我  好几天了

binge8, 2024-11-22:
大概找到问题了，之前因为helloword依赖gn编译报错，添加了clang，gn成功了，就会有这个报错

qqsir-dev, 2024-11-22:
看来不是我一个人，可是这个怎么能解决呢，我好像也没有加clang

freeAhao, 2024-11-24:
一样的问题

binge8, 2024-11-24:
插件少的编译没问题，选多了就出现这个报错，一模一样，暂时不会确定是哪个插件导致的

coolsnowwolf, 2024-11-24:
和你们的 hostos 有关，别选 gcc14 就好

binge8, 2024-11-25:
> 和你们的 hostos 有关，别选 gcc14 就好

应该也不是，我是gcc13

zsedc800, 2024-11-25:
我也遇到了，错误信息也是sqlite3那个问题。我本地的gcc是11.4 lede里面的gcc是11.3不知道是不是这个差异导致的。切换版本太麻烦了，我干脆在 menuconfig languages选项里面将php7关了换成php8可以编译通过

binge8, 2024-11-25:
> 我也遇到了，错误信息也是sqlite3那个问题。我本地的gcc是11.4 lede里面的gcc是11.3不知道是不是这个差异导致的。切换版本太麻烦了，我干脆在 menuconfig languages选项里面将php7关了换成php8可以编译通过

大佬您好，刚才看了看换php8，对小白来说有点复杂，还有别的办法吗

zsedc800, 2024-11-25:
> > 我也遇到了，错误信息也是sqlite3那个问题。我本地的gcc是11.4 lede里面的gcc是11.3不知道是不是这个差异导致的。切换版本太麻烦了，我干脆在 menuconfig languages选项里面将php7关了换成php8可以编译通过
> 
> 大佬您好，刚才看了看换php8，对小白来说有点复杂，还有别的办法吗

[image]
[image]
我就是在这里参考php7里面的选项配置php8,然后关掉php7的选择

zyimm, 2024-11-26:
使用docker  来安装php 似乎是更好的选择

sxml, 2024-11-26:
make[4]: *** [Makefile:568: ext/sqlite3/sqlite3.lo] Error 1 

5135make[3]: *** [Makefile:621: /workdir/openwrt/build_dir/target-aarch64_generic_musl/php-7.4.30/.built] Error 2 

5136make[2]: *** [package/Makefile:116: package/feeds/packages/php7/compile] Error 1 

5137make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2 

5138make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2 

5139make[4]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/php-7.4.30' 

5140make[3]: Leaving directory '/workdir/openwrt/feeds/packages/lang/php7' 

5141time: package/feeds/packages/php7/compile#0.51#0.12#0.59 

5142 ERROR: package/feeds/packages/php7 failed to build. 

5143make[2]: Leaving directory '/workdir/openwrt' 

5144make[1]: Leaving directory '/workdir/openwrt' 

5145Error: Process completed with exit code 2.

云编译也报错卡在这php7  换成18.06luci 就没事？？？

binge8, 2024-11-28:
```bash
2024-11-28T05:35:20.1971676Z make[4]: Entering directory '/mnt/openwrt/build_dir/target-x86_64_musl/php-7.4.30'
2024-11-28T05:35:20.1981244Z /bin/bash /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/libtool --silent --preserve-dup-deps --mode=compile x86_64-openwrt-linux-musl-gcc -DZEND_ENABLE_STATIC_TSRMLS_CACHE=1 -Iext/sqlite3/ -I/home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/ext/sqlite3/ -DPHP_ATOM_INC -I/home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/include -I/home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/main -I/home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30 -I/home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/ext/date/lib -I/home/runner/work/op/op/openwrt/staging_dir/target-x86_64_musl/usr/include/libxml2 -I/home/runner/work/op/op/openwrt/staging_dir/target-x86_64_musl/usr/include -I/home/runner/work/op/op/openwrt/staging_dir/target-x86_64_musl/usr/include/freetype2 -I/home/runner/work/op/op/openwrt/staging_dir/target-x86_64_musl/usr/include/libpng16 -I/home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/ext/mbstring/libmbfl -I/home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/ext/mbstring/libmbfl/mbfl -I/home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/TSRM -I/home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/Zend  -I/home/runner/work/op/op/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/runner/work/op/op/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/runner/work/op/op/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include  -I/usr/include -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30=php-7.4.30 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fvisibility=hidden -Wall -Wno-strict-aliasing -DZEND_SIGNALS   -prefer-pic -c /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/ext/sqlite3/sqlite3.c -o ext/sqlite3/sqlite3.lo 
2024-11-28T05:35:20.3046567Z In file included from /usr/include/sys/types.h:129,
2024-11-28T05:35:20.3047229Z                  from /usr/include/stdlib.h:395,
2024-11-28T05:35:20.3048633Z                  from /home/runner/work/op/op/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify/stdlib.h:22,
2024-11-28T05:35:20.3049926Z                  from /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/include/../main/php_config.h:2327,
2024-11-28T05:35:20.3051137Z                  from /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/Zend/zend_config.h:1,
2024-11-28T05:35:20.3052314Z                  from /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/Zend/zend_portability.h:43,
2024-11-28T05:35:20.3053448Z                  from /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/Zend/zend_types.h:25,
2024-11-28T05:35:20.3054546Z                  from /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/Zend/zend.h:27,
2024-11-28T05:35:20.3055764Z                  from /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/main/php.h:33,
2024-11-28T05:35:20.3056852Z                  from /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/ext/sqlite3/sqlite3.c:23:
2024-11-28T05:35:20.3057825Z /usr/include/bits/types/time_t.h:8:9: error: unknown type name '__time64_t'
2024-11-28T05:35:20.3058448Z     8 | typedef __time64_t time_t;
2024-11-28T05:35:20.3058825Z       |         ^~~~~~~~~~
2024-11-28T05:35:20.3059278Z In file included from /usr/include/bits/endian.h:35,
2024-11-28T05:35:20.3059803Z                  from /usr/include/endian.h:24,
2024-11-28T05:35:20.3060266Z                  from /usr/include/sys/types.h:176,
2024-11-28T05:35:20.3060946Z                  from /usr/include/stdlib.h:395,
2024-11-28T05:35:20.3061821Z                  from /home/runner/work/op/op/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify/stdlib.h:22,
2024-11-28T05:35:20.3063158Z                  from /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/include/../main/php_config.h:2327,
2024-11-28T05:35:20.3064406Z                  from /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/Zend/zend_config.h:1,
2024-11-28T05:35:20.3065557Z                  from /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/Zend/zend_portability.h:43,
2024-11-28T05:35:20.3066675Z                  from /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/Zend/zend_types.h:25,
2024-11-28T05:35:20.3067750Z                  from /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/Zend/zend.h:27,
2024-11-28T05:35:20.3068841Z                  from /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/main/php.h:33,
2024-11-28T05:35:20.3070016Z                  from /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/ext/sqlite3/sqlite3.c:23:
2024-11-28T05:35:20.3070993Z /usr/include/bits/endianness.h:9: warning: "__BYTE_ORDER" redefined
2024-11-28T05:35:20.3071612Z     9 | #define __BYTE_ORDER __LITTLE_ENDIAN
2024-11-28T05:35:20.3072026Z       | 
[... cut ...]
2024-11-28T05:35:20.4863425Z /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/ext/sqlite3/sqlite3.c:52:16: note: expected '__va_list_tag *' but argument is of type 'va_list' {aka 'int'}
2024-11-28T05:35:20.5133978Z make[4]: *** [Makefile:568: ext/sqlite3/sqlite3.lo] Error 1
2024-11-28T05:35:20.5134436Z make[4]: Leaving directory '/mnt/openwrt/build_dir/target-x86_64_musl/php-7.4.30'
2024-11-28T05:35:20.5136424Z make[3]: *** [Makefile:621: /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/.built] Error 2
2024-11-28T05:35:20.5136902Z make[3]: Leaving directory '/mnt/openwrt/feeds/packages/lang/php7'
2024-11-28T05:35:20.5

sxml, 2024-11-28:
> ```shell
> 2024-11-28T05:35:20.1971676Z make[4]: Entering directory '/mnt/openwrt/build_dir/target-x86_64_musl/php-7.4.30'
> 2024-11-28T05:35:20.1981244Z /bin/bash /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/libtool --silent --preserve-dup-deps --mode=compile x86_64-openwrt-linux-musl-gcc -DZEND_ENABLE_STATIC_TSRMLS_CACHE=1 -Iext/sqlite3/ -I/home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/ext/sqlite3/ -DPHP_ATOM_INC -I/home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/include -I/home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/main -I/home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30 -I/home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/ext/date/lib -I/home/runner/work/op/op/openwrt/staging_dir/target-x86_64_musl/usr/include/libxml2 -I/home/runner/work/op/op/openwrt/staging_dir/target-x86_64_musl/usr/include -I/home/runner/work/op/op/openwrt/staging_dir/target-x86_64_musl/usr/include/freetype2 -I/home/runner/work/op/op/openwrt/staging_dir/target-x86_64_musl/usr/include/libpng16 -I/home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/ext/mbstring/libmbfl -I/home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/ext/mbstring/libmbfl/mbfl -I/home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/TSRM -I/home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/Zend  -I/home/runner/work/op/op/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/runner/work/op/op/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/runner/work/op/op/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include  -I/usr/include -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30=php-7.4.30 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fvisibility=hidden -Wall -Wno-strict-aliasing -DZEND_SIGNALS   -prefer-pic -c /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/ext/sqlite3/sqlite3.c -o ext/sqlite3/sqlite3.lo 
> 2024-11-28T05:35:20.3046567Z In file included from /usr/include/sys/types.h:129,
> 2024-11-28T05:35:20.3047229Z                  from /usr/include/stdlib.h:395,
> 2024-11-28T05:35:20.3048633Z                  from /home/runner/work/op/op/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify/stdlib.h:22,
> 2024-11-28T05:35:20.3049926Z                  from /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/include/../main/php_config.h:2327,
> 2024-11-28T05:35:20.3051137Z                  from /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/Zend/zend_config.h:1,
> 2024-11-28T05:35:20.3052314Z                  from /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/Zend/zend_portability.h:43,
> 2024-11-28T05:35:20.3053448Z                  from /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/Zend/zend_types.h:25,
> 2024-11-28T05:35:20.3054546Z                  from /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/Zend/zend.h:27,
> 2024-11-28T05:35:20.3055764Z                  from /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/main/php.h:33,
> 2024-11-28T05:35:20.3056852Z                  from /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/ext/sqlite3/sqlite3.c:23:
> 2024-11-28T05:35:20.3057825Z /usr/include/bits/types/time_t.h:8:9: error: unknown type name '__time64_t'
> 2024-11-28T05:35:20.3058448Z     8 | typedef __time64_t time_t;
> 2024-11-28T05:35:20.3058825Z       |         ^~~~~~~~~~
> 2024-11-28T05:35:20.3059278Z In file included from /usr/include/bits/endian.h:35,
> 2024-11-28T05:35:20.3059803Z                  from /usr/include/endian.h:24,
> 2024-11-28T05:35:20.3060266Z                  from /usr/include/sys/types.h:176,
> 2024-11-28T05:35:20.3060946Z                  from /usr/include/stdlib.h:395,
> 2024-11-28T05:35:20.3061821Z                  from /home/runner/work/op/op/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify/stdlib.h:22,
> 2024-11-28T05:35:20.3063158Z                  from /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/include/../main/php_config.h:2327,
> 2024-11-28T05:35:20.3064406Z                  from /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/Zend/zend_config.h:1,
> 2024-11-28T05:35:20.3065557Z                  from /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/Zend/zend_portability.h:43,
> 2024-11-28T05:35:20.3066675Z                  from /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/Zend/zend_types.h:25,
> 2024-11-28T05:35:20.3067750Z                  from /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/Zend/zend.h:27,
> 2024-11-28T05:35:20.3068841Z                  from /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/main/php.h:33,
> 2024-11-28T05:35:20.3070016Z                  from /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/ext/sqlite3/sqlite3.c:23:
> 2024-11-28T05:35:20.3070993Z /usr/include/bits/endianness.h:9: warning: "__BYTE_ORDER" redefined
[... cut ...]
> 2024-11-28T05:35:20.4344163Z                  from /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/php-7.4.30/ext/sqlite3/sqlite3.c:23:
> 2024-11-28T05:35:20.4344330Z /usr/include/utime.h:39:5: error: unknown type name '__time64_t'
> 2024-11-28T05:35:20.4344439Z    39 |     __time64_t actime;          /* Access time.  */
> 2024-11-28T05:35:20.4344510Z       |     ^~~~~~~~~~
> 2024-11-28T05:35:20.4344669Z /usr/include/utime.h:40:5: error: unknown type name '__time64_t'
> 2024-11-28T05:35:20.4344796Z    40 |     __time64

binge8, 2024-11-28:
我也有这个可道云，去掉试试

sxml, 2024-11-28:
> 我也有这个可道云，去掉试试

去掉luci-app-kodexplorer 可道云过了 没再报错

binge8, 2024-11-30:
> > 我也有这个可道云，去掉试试
> 
> 去掉luci-app-kodexplorer 可道云过了 没再报错

去掉这个可道云确实成功了

binge8, 2024-12-01:
> > 我也有这个可道云，去掉试试
> 
> 去掉luci-app-kodexplorer 可道云过了 没再报错

去掉可道云后，armv8架构没问题了，但是x86架构还不行啊

sxml, 2024-12-05:
> 这个问题已经解决了，有可道云也可以正常编译了

23.05大雕有修改 Makefile里之前是php7

closed 2024-12-06 by qqsir-dev