coolsnowwolf/lede #13101: error

magi-Z, 2024-11-20:
### 详细叙述

/usr/bin/ld: /home/magi/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/gcc/config/i386/i386.md:20955:(.text+0xebc7d): undefined reference to `recog_data'
/usr/bin/ld: /home/magi/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/gcc/config/i386/i386.md:20957:(.text+0xebca8): undefined reference to `recog_data'
collect2: error: ld returned 1 exit status
make[5]: *** [/home/magi/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/gcc/c/Make-lang.in:87: cc1] Error 1
make[5]: Leaving directory '/home/magi/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-initial/gcc'
make[4]: *** [Makefile:4428: all-gcc] Error 2
make[4]: Leaving directory '/home/magi/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-initial'
make[3]: *** [Makefile:37: /home/magi/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-initial/.built] Error 2
make[3]: Leaving directory '/home/magi/lede/toolchain/gcc/initial'
time: toolchain/gcc/initial/compile#3.94#3.06#8.07
    ERROR: toolchain/gcc/initial failed to build.
make[2]: *** [toolchain/Makefile:97: toolchain/gcc/initial/compile] Error 1
make[2]: Leaving directory '/home/magi/lede'
make[1]: *** [toolchain/Makefile:93: /home/magi/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/magi/lede'
make: *** [/home/magi/lede/include/toplevel.mk:231：world] 错误 2

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

/usr/bin/ld: /home/magi/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/gcc/config/i386/i386.md:20955:(.text+0xebc7d): undefined reference to `recog_data'
/usr/bin/ld: /home/magi/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/gcc/config/i386/i386.md:20957:(.text+0xebca8): undefined reference to `recog_data'
collect2: error: ld returned 1 exit status
make[5]: *** [/home/magi/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/gcc/c/Make-lang.in:87: cc1] Error 1
make[5]: Leaving directory '/home/magi/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-initial/gcc'
make[4]: *** [Makefile:4428: all-gcc] Error 2
make[4]: Leaving directory '/home/magi/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-initial'
make[3]: *** [Makefile:37: /home/magi/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-initial/.built] Error 2
make[3]: Leaving directory '/home/magi/lede/toolchain/gcc/initial'
time: toolchain/gcc/initial/compile#3.94#3.06#8.07
    ERROR: toolchain/gcc/initial failed to build.
make[2]: *** [toolchain/Makefile:97: toolchain/gcc/initial/compile] Error 1
make[2]: Leaving directory '/home/magi/lede'
make[1]: *** [toolchain/Makefile:93: /home/magi/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/magi/lede'
make: *** [/home/magi/lede/include/toplevel.mk:231：world] 错误 2

yanxin152133, 2024-11-21:
https://github.com/coolsnowwolf/lede/issues/11328

coolsnowwolf, 2026-09-05:
这是 #11328 同类的旧工具链/脏构建目录问题，当前 master 已无法复现 issue 中的 GCC 11.3 initial 错误：

- #11328 的报告者已确认更新并重建 toolchain 后解决
- 当前默认工具链已升级为 GCC `13.3.0`、binutils `2.42`，不再使用日志中的 GCC 11.3.0
- 当前 `x86_64-openwrt-linux-musl-gcc` 能正常启动 `cc1` 和 assembler，并成功编译测试对象；本轮多个 x86_64 包也已用该工具链实编成功

这类 `recog_data`/LTO 符号异常通常来自更新源码后继续复用旧 `build_dir`/`staging_dir`。应在同一源码版本下清理并完整重建 toolchain，不能混用旧产物。关闭为重复且已由当前工具链更新解决。

closed 2026-09-05 by coolsnowwolf