coolsnowwolf/lede #13103: 新版固件mwan3配置不生效

Elec4d0, 2024-11-21:
### 详细叙述

新版固件mwan3配置不生效，以如下配置为例，并未能走到指定接口，而是按主路由表走。
旧版mwan3可以，希望大佬能排查一下
[image]

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

暂无

Elec4d0, 2024-11-21:
另外，这些mwan管理诊断器执行不了，是否mwan3接口变化，而前端页面未更新？
[image]

coolsnowwolf, 2024-11-21:
更新了 js 的版本了

closed 2024-11-22 by coolsnowwolf

Elec4d0, 2024-11-22:
大佬你好！似乎问题没有得到解决，mwan3规则还是不生效

Elec4d0, 2024-11-22:
目前我回退到了2024.11.01的那个commit进行编译，目前一切正常了，最新合并进来的源码mwan3似乎还是无法正确工作。

编译版本：
[image]
1.1.1.1 单独指定出口测试的配置已生效
[image]

zx09, 2025-04-17:
LZ你解决了吗  我也是启动失败 这是恩山的人说的 似乎是脚本启动顺序有问题 
“其实你想过是为什么么，是因为mwan3设定启动优先级为19、dnsmasq也是19，而mwan3-helper为60，导致dnsmasq启动在mwan3-helper前，相关路径都还没有被mwan3-helper创建。
事实上在19时整个系统网络环境都还没有完整建立（因为network启动在20，odhcpd在35，dbus在60），不清楚为何作者会把mwan3优先级设到19，此时mwan3根本无法工作，你可以明确看到syslog里有mwan3的运行报错。
只需要把mwan3-helper改为先于dnsmasq启动就行了。
最science办法是改helper的启动脚本”

https://www.right.com.cn/forum/forum.php?mod=viewthread&tid=8296669&page=1#pid19335965

大佬能不能重新编译一下

Elec4d0, 2025-04-17:
> LZ你解决了吗  我也是启动失败 这是恩山的人说的 似乎是脚本启动顺序有问题
> “其实你想过是为什么么，是因为mwan3设定启动优先级为19、dnsmasq也是19，而mwan3-helper为60，导致dnsmasq启动在mwan3-helper前，相关路径都还没有被mwan3-helper创建。 [...]

那次之后我的lede版本就稳定使用了，没有再进行编译和升级替换……，如果您现在也是mwan3不生效，建议您会退24.11.01那个commit进行编译即可，lean这边看起来没有关心这个事情