coolsnowwolf/lede #13104: make menuconfig 后出现递归依赖提示，怎么解决

xiaomayi755259, 2024-11-21:
### 详细叙述

Collecting package info: done
tmp/.config-package.in:100412:error: recursive dependency detected!
tmp/.config-package.in:100412:	symbol PACKAGE_iptables is selected by PACKAGE_miniupnpd
tmp/.config-package.in:101250:	symbol PACKAGE_miniupnpd is selected by PACKAGE_luci-app-upnp
tmp/.config-package.in:64554:	symbol PACKAGE_luci-app-upnp depends on PACKAGE_rpcd
tmp/.config-package.in:1142:	symbol PACKAGE_rpcd is selected by PACKAGE_rpcd-mod-luci
tmp/.config-package.in:60983:	symbol PACKAGE_rpcd-mod-luci is selected by PACKAGE_luci-base
tmp/.config-package.in:61358:	symbol PACKAGE_luci-base is selected by PACKAGE_luci-lua-runtime
tmp/.config-package.in:61510:	symbol PACKAGE_luci-lua-runtime is selected by PACKAGE_luci-compat
tmp/.config-package.in:61499:	symbol PACKAGE_luci-compat is selected by PACKAGE_luci-app-bypass
tmp/.config-package.in:62038:	symbol PACKAGE_luci-app-bypass depends on PACKAGE_iptables
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86/64

### 详细日志

Collecting package info: done
tmp/.config-package.in:100412:error: recursive dependency detected!
tmp/.config-package.in:100412:	symbol PACKAGE_iptables is selected by PACKAGE_miniupnpd
tmp/.config-package.in:101250:	symbol PACKAGE_miniupnpd is selected by PACKAGE_luci-app-upnp
tmp/.config-package.in:64554:	symbol PACKAGE_luci-app-upnp depends on PACKAGE_rpcd
tmp/.config-package.in:1142:	symbol PACKAGE_rpcd is selected by PACKAGE_rpcd-mod-luci
tmp/.config-package.in:60983:	symbol PACKAGE_rpcd-mod-luci is selected by PACKAGE_luci-base
tmp/.config-package.in:61358:	symbol PACKAGE_luci-base is selected by PACKAGE_luci-lua-runtime
tmp/.config-package.in:61510:	symbol PACKAGE_luci-lua-runtime is selected by PACKAGE_luci-compat
tmp/.config-package.in:61499:	symbol PACKAGE_luci-compat is selected by PACKAGE_luci-app-bypass
tmp/.config-package.in:62038:	symbol PACKAGE_luci-app-bypass depends on PACKAGE_iptables
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

coolsnowwolf, 2024-11-21:
删除 luci-app-bypass

xiaomayi755259, 2024-11-21:
> 删除 luci-app-bypass

大佬，出现递归依赖是怎么回事？怎么彻底才能解决，以前编译没出现过问题，即使删除整个目录重新编译也可以。

HollyMadCow, 2024-11-26:
删除 luci-app-upnp就可以了

xiaomayi755259, 2024-11-26:
> 删除 luci-app-upnp就可以了

多谢多谢了，确实没报错了

closed 2025-01-27 by aiamadeus