coolsnowwolf/lede #13105: 吃螃蟹，kernel 6.12版本编译失败

GreatMichaelLee, 2024-11-21:
### 详细叙述

缺ko依赖

mkdir -p /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/stamp

SHELL= flock /home/builder/lede_x86/tmp/.root-copy.flock -c 'cp -fpR /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kernel/. /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/'
touch /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/stamp/.kernel_installed
Package kmod-tpm is missing dependencies for the following libraries:
ecdh_generic.ko
libaescfb.ko
make[3]: *** [modules/other.mk:1229: /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-tpm_6.12-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/builder/lede_x86/package/kernel/linux'
time: package/kernel/linux/compile#2.32#0.50#2.81
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:110: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:231：world] 错误 2

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

见上

coolsnowwolf, 2024-11-21:
小问题，我来助你~

GreatMichaelLee, 2024-11-21:
> 小问题，我来助你~

深夜惊动大佬，等米下锅^^

coolsnowwolf, 2024-11-21:
好了

GreatMichaelLee, 2024-11-21:
新问题，ksmbd 又过不了

# CC [M]  /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-2023.12.22/unicode.o
  x86_64-openwrt-linux-musl-gcc -Wp,-MMD,/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-2023.12.22/.unicode.o.d -nostdinc -I./arch/x86/include -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/compiler-version.h -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -fmacro-prefix-map=./= -std=gnu11 -fshort-wchar -funsigned-char -fno-common -fno-PIE -fno-strict-aliasing -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -fcf-protection=none -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -mindirect-branch-cs-prefix -fno-jump-tables -fno-delete-null-pointer-checks -O2 -fno-allow-store-data-races -fstack-protector -fno-omit-frame-pointer -fno-optimize-sibling-calls -fno-stack-clash-protection -falign-functions=16 -fstrict-flex-arrays=3 -fno-strict-overflow -fno-stack-check -fconserve-stack -Wall -Wundef -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Werror=strict-prototypes -Wno-format-security -Wno-trigraphs -Wno-frame-address -Wno-address-of-packed-member -Wmissing-declarations -Wmissing-prototypes -Wframe-larger-than=2048 -Wno-main -Wno-dangling-pointer -Wvla -Wno-pointer-sign -Wcast-function-type -Wno-stringop-overflow -Wno-array-bounds -Wno-alloc-size-larger-than -Wimplicit-fallthrough=5 -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -Wenum-conversion -Wextra -Wunused -Wno-unused-but-set-variable -Wno-unused-const-variable -Wno-packed-not-aligned -Wno-format-overflow -Wno-format-truncation -Wno-stringop-truncation -Wno-override-init -Wno-missing-field-initializers -Wno-type-limits -Wno-shift-negative-value -Wno-maybe-uninitialized -Wno-sign-compare -Wno-unused-parameter -g -fno-var-tracking -femit-struct-debug-baseonly -fmacro-prefix-map=/home/builder/lede_x86/build_dir/target-x86_64_musl=target-x86_64_musl -fno-caller-saves -DCONFIG_SMB_INSECURE_SERVER=1 -I/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12/arch/arc/kernel  -DMODULE  -DKBUILD_BASENAME='"unicode"' -DKBUILD_MODNAME='"ksmbd"' -D__KBUILD_MODNAME=kmod_ksmbd -c -o /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-2023.12.22/unicode.o /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-2023.12.22/unicode.c   ; ./tools/objtool/objtool --hacks=jump_label --hacks=noinstr --retpoline --stackval --static-call --uaccess   --module /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-2023.12.22/unicode.o
/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-2023.12.22/unicode.c:11:10: fatal error: asm/unaligned.h: No such file or directory
   11 | #include <asm/unaligned.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
make[5]: *** [scripts/Makefile.build:229: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-2023.12.22/unicode.o] Error 1
make[4]: *** [/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12/Makefile:1936: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-2023.12.22] Error 2
make[3]: *** [Makefile:224: __sub-make] Error 2
make[3]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12'
make[2]: *** [Makefile:78: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-2023.12.22/.built] Error 2
make[2]: Leaving directory '/home/builder/lede_x86/package/kernel/ksmbd'
time: package/kernel/ksmbd/compile#0.31#0.30#0.60
    ERROR: package/kernel/ksmbd failed to build.
make[1]: *** [package/Makefile:116: package/kernel/ksmbd/compile] Error 1
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:231：package/kernel/ksmbd/compile] 错误 2

coolsnowwolf, 2024-11-22:
好了，ksmbd oot 还不支持 6.12 内核

GreatMichaelLee, 2024-11-22:
感谢大佬，new errors,  xtables-addson

ons/rtsp/nf_conntrack_rtsp.c   ; ./tools/objtool/objtool --hacks=jump_label --hacks=noinstr --retpoline --stackval --static-call --uaccess   --module /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24/extensions/rtsp/nf_conntrack_rtsp.o
/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24/extensions/rtsp/nf_conntrack_rtsp.c: In function 'init':
/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24/extensions/rtsp/nf_conntrack_rtsp.c:738:17: error: implicit declaration of function 'strlcpy'; did you mean 'strncpy'? [-Werror=implicit-function-declaration]
  738 |                 strlcpy(hlpr->name, tmpname, sizeof(hlpr->name));
      |                 ^~~~~~~
      |                 strncpy
cc1: some warnings being treated as errors
make[10]: *** [scripts/Makefile.build:229: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24/extensions/rtsp/nf_conntrack_rtsp.o] Error 1
make[9]: *** [scripts/Makefile.build:478: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24/extensions/rtsp] Error 2
make[8]: *** [/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12/Makefile:1936: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24/extensions] Error 2
make[7]: *** [Makefile:224: __sub-make] Error 2
make[7]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12'
make[6]: *** [Makefile:466: modules] Error 2
make[6]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24/extensions'
make[5]: *** [Makefile:623: all-recursive] Error 1
make[5]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24'
make[4]: *** [Makefile:395: all] Error 2
make[4]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24'
make[3]: *** [Makefile:213: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.24/.built] Error 2
make[3]: Leaving directory '/home/builder/lede_x86/feeds/packages/net/xtables-addons'
time: package/feeds/packages/xtables-addons/compile#0.71#0.33#1.05
    ERROR: package/feeds/packages/xtables-addons failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/xtables-addons/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:110: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:231：world] 错误 2

coolsnowwolf, 2024-11-22:
已修复支持，xtables-addons 上游还没支持 linux 6.12

GreatMichaelLee, 2024-11-22:
> 已修复支持，xtables-addons 上游还没支持 linux 6.12

刚编了一个6.12出来，upgrade之后直接给我kernel panic 无限重启了，我都给整哭了，哈哈，esxi里console闪太快，没看清哪个模块，好像是nfs还是ksmbd. 这螃蟹吃的：）

GreatMichaelLee, 2024-11-22:
6.6.62 ksmbd 出这个。。。。

 -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/compiler-version.h -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -fmacro-prefix-map=./= -std=gnu11 -fshort-wchar -funsigned-char -fno-common -fno-PIE -fno-strict-aliasing -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -fcf-protection=none -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -mindirect-branch-cs-prefix -fno-jump-tables -fno-delete-null-pointer-checks -O2 -fno-allow-store-data-races -fstack-protector -fno-omit-frame-pointer -fno-optimize-sibling-calls -fno-stack-clash-protection -falign-functions=16 -fstrict-flex-arrays=3 -fno-strict-overflow -fno-stack-check -fconserve-stack -Wall -Wundef -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Werror=strict-prototypes -Wno-format-security -Wno-trigraphs -Wno-frame-address -Wno-address-of-packed-member -Wframe-larger-than=2048 -Wno-main -Wno-unused-but-set-variable -Wno-unused-const-variable -Wno-dangling-pointer -Wvla -Wno-pointer-sign -Wcast-function-type -Wno-array-bounds -Wno-alloc-size-larger-than -Wimplicit-fallthrough=5 -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -Wenum-conversion -Wno-unused-but-set-variable -Wno-unused-const-variable -Wno-restrict -Wno-packed-not-aligned -Wno-format-overflow -Wno-format-truncation -Wno-stringop-overflow -Wno-stringop-truncation -Wno-missing-field-initializers -Wno-type-limits -Wno-shift-negative-value -Wno-maybe-uninitialized -Wno-sign-compare -g -fno-var-tracking -femit-struct-debug-baseonly -fmacro-prefix-map=/home/builder/lede_x86/build_dir/target-x86_64_musl=target-x86_64_musl -fno-caller-saves  -DMODULE  -DKBUILD_BASENAME='"vfs"' -DKBUILD_MODNAME='"ksmbd"' -D__KBUILD_MODNAME=kmod_ksmbd -c -o /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-2024.10.14/vfs.o /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-2024.10.14/vfs.c   ; ./tools/objtool/objtool --hacks=jump_label --hacks=noinstr --retpoline --stackval --static-call --uaccess   --module /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-2024.10.14/vfs.o
/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-2024.10.14/vfs.c:2666:20: error: redefinition of 'is_dot_dotdot'
 2666 | static inline bool is_dot_dotdot(const char *name, size_t len)
      |                    ^~~~~~~~~~~~~
In file included from /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-2024.10.14/vfs.c:8:
./include/linux/fs.h:2852:20: note: previous definition of 'is_dot_dotdot' with type 'bool(const char *, size_t)' {aka '_Bool(const char *, long unsigned int)'}
 2852 | static inline bool is_dot_dotdot(const char *name, size_t len)
      |                    ^~~~~~~~~~~~~
make[6]: *** [scripts/Makefile.build:243: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-2024.10.14/vfs.o] Error 1
make[5]: *** [/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.62/Makefile:1921: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-2024.10.14] Error 2
make[4]: *** [Makefile:234: __sub-make] Error 2
make[4]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.62'
make[3]: *** [Makefile:54: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-2024.10.14/.built] Error 2
make[3]: Leaving directory '/home/builder/lede_x86/package/kernel/ksmbd'
time: package/kernel/ksmbd/compile#0.52#0.34#0.85
    ERROR: package/kernel/ksmbd failed to build.
make[2]: *** [package/Makefile:116: package/kernel/ksmbd/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:110: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:231：world] 错误 2
builder@Build-Server:/home/builder/lede_x86$

xiangfeidexiaohuo, 2024-11-22:
还有 ntfs 这个模块

closed 2024-11-23 by coolsnowwolf