coolsnowwolf/lede #13111: 昨天新编译的版本中，SSRP只能使用SS节点无法使用V2RAY节点

henryyifan, 2024-11-22:
### 详细叙述

实在不好意思，太小白了，搞不定不得不提问打扰各位了。
今年2月之后就一直没编译过新版本，昨天编译了新版本后，发现SSR+无论国内外都无法连接。
后来遇到L大回我说是XRAY大改，要reset，于是按照他的命令 /etc/init.d/shadowsocksr reset 后，国内和SS节点可以连接，但vmess节点仍然不行。
也试过：https://github.com/coolsnowwolf/lede/issues/12617  这里的方法，问题依旧。也试过不升级直接重装，问题依旧。
SSR+用的是helloword的master的分支。
不确定是否是编译时少选了什么，据我浅薄的知识应该是有个XRAY-CORE就行了？
现已刷回老版本R24.2.2，内核6.1.77。
不好意思打扰各位了，能否再多提点一二？
先感谢各位！

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

X86

### 详细日志

暂无

coolsnowwolf, 2024-11-22:
xray 版本的变动可能也需要服务器改变，单单是描述看不出来的

henryyifan, 2024-11-22:
> xray 版本的变动可能也需要服务器改变，单单是描述看不出来的

感谢L大的热心回复！
服务器那边的确说不清，看来我只能继续先用当前的旧版本了。

suizhenmin, 2024-11-22:
遇到同样的情况，就是vmess的节点无法连接，ssr中节点可用性检查是通的，就是启动不了服务

henryyifan, 2024-11-22:
> 遇到同样的情况，就是vmess的节点无法连接，ssr中节点可用性检查是通的，就是启动不了服务

L大很热心的回复了，而且我看到他在passwall那个帖子说SSR+正常，所以暂时不知道该如何处理，只能暂时用旧版本的同时等其他人帮忙了。
也不知道该提供哪些log之类的暂时

abclai, 2024-11-22:
我告诉你们是什么问题啦！前几天刚解决啦！
就是这个软件大改啦，原来的设置文件不能用啦！需要重新设置。
方法1.直接重置所有设置。
方法2.用WinSCP登录路由器吧congfig里面SS的设置删除然后重新刷新固件就可以啦！
一定要把原来的设置文件删除干净再重新设置就行啦！

abclai, 2024-11-22:
> > 我告诉你们是什么问题啦！前几天刚解决啦！ 就是这个软件大改啦，原来的设置文件不能用啦！需要重新设置。 方法1.直接重置所有设置。 方法2.用WinSCP登录路由器吧congfig里面SS的设置删除然后重新刷新固件就可以啦！ 一定要把原来的设置文件删除干净再重新设置就行啦！
> 
> 试过这个方法，但是没有把所有配置都重置，只是重置和删除了SSR+相关的，明天试试看，感谢

你可以试试这样 先软件列表哪里把ssr+软件卸载，之后WinSCP登录进去路由器把config里面一个SS的设置文件给删除。之后在系统升级里面重新升级下固件啦！升级完之后重新添加服务器信息啦 应该就可以 我是这样解决啦

henryyifan, 2024-11-23:
> > > 我告诉你们是什么问题啦！前几天刚解决啦！ 就是这个软件大改啦，原来的设置文件不能用啦！需要重新设置。 方法1.直接重置所有设置。 方法2.用WinSCP登录路由器吧congfig里面SS的设置删除然后重新刷新固件就可以啦！ 一定要把原来的设置文件删除干净再重新设置就行啦！
> > 
> > 
> > 试过这个方法，但是没有把所有配置都重置，只是重置和删除了SSR+相关的，明天试试看，感谢
> 
> 你可以试试这样 先软件列表哪里把ssr+软件卸载，之后WinSCP登录进去路由器把config里面一个SS的设置文件给删除。之后在系统升级里面重新升级下固件啦！升级完之后重新添加服务器信息啦 应该就可以 我是这样解决啦

你好，这个方法我又试了一下，结果还是不行，感谢你的帮助~~~

closed 2024-11-23 by coolsnowwolf