coolsnowwolf/lede #13115: 无法登录IP

dl250, 2024-11-23:
### 详细叙述

编译前我已经把源码IP地址改了，编译完用新的IP登录发现登录不了，通过命令发现IP还是默认192.168.1.1，而不是我编译前修改的IP，我通过命令改了IP还是无法登录

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

[image]

dl250, 2024-11-23:
网卡改为e1000,还是无法登录

henryyifan, 2024-11-23:
windows里的ipv4看看在不在同一网段？不在的话改成同一网段？

dl250, 2024-11-23:
> windows里的ipv4看看在不是同一网段？不在的话改成同一网段？

网段没问题

coolsnowwolf, 2024-11-23:
因为不需要改源码  
[image]

fyngd, 2024-11-23:
> 因为不需要改源码 [...]

老大，找不到 Preinit configuration options 这个选项在哪个菜单下啊

dl250, 2024-11-23:
又重新拉源码，还是不行，没添加插件，IP还是登录不上，IP也按上边的方法改了

19901019, 2024-11-23:
package/base-files/luci2/bin/config_generate

fyngd, 2024-11-23:
> package/base-files/luci2/bin/config_generate

谢谢大佬！请问知道在哪里改默认主题吗？