coolsnowwolf/lede #13119: 5g 频段wifi缺少频段设置选项

kcro227, 2024-11-24:
### 详细叙述

如图，5Ghz的wifi设置中AC模式缺少频带设置选项
[image]

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

极路由B70

### 详细日志

make不报错

binext, 2024-12-02:
同样的问题，这是luci的bug吗？设置之后5g wifi就启动不了了。

kcro227, 2024-12-07:
> 同样的问题，这是luci的bug吗？设置之后5g wifi就启动不了了。

是的，你用ssh进终端，修改/etc/config/wireless的内容，添加关于5G频段的选项就可以

ipconfig25, 2024-12-19:
ASUS RT-AC42U 同样问题 设置之后  5g wifi就启动不了了

kcro227, 2024-12-19:
> ASUS RT-AC42U 同样问题 设置之后 5g wifi就启动不了了

实测，把luci换回master分支就正常了

coolsnowwolf, 2026-09-05:
这个旧版 LuCI 显示问题已在当前分支修复。

核对结果：
- 旧逻辑在选择框只有一个选项时使用 `sel.options.length <= 1` 隐藏整行，因此 802.11ac 只有 5 GHz 一个 band 时看不到“频带”
- LuCI 提交 `50ff91533fca30dbd1c2bb50879b961689912821` 已调整为 band 选择框只要有可用选项就显示
- 当前 `wireless.js` 为 AC 明确生成 `ac: [ 5g, 5 GHz, ... ]`，并依据实际 5 GHz channel availability 启用
- x86_64 下执行 `make package/feeds/luci/luci-mod-network/compile -j30 CONFIG_PACKAGE_luci-mod-network=y` 成功，生成 `luci-mod-network_26.233.28523~ee9eee0_all.ipk`（85553 bytes）

因此无需再手工修改 `/etc/config/wireless`；使用当前 `openwrt-25.12` LuCI 分支即可。关闭此 issue。

closed 2026-09-05 by coolsnowwolf