coolsnowwolf/lede #13122: UPnP 不显示活动的重定向

zjjxwhh, 2024-11-25:
### 详细叙述

编译日期：2024.11.24
固件版本：LEDE R24.11.11 / LuCI openwrt-23.05 branch git-24.326.41750-70addca
内核版本：6.6.63

软件包版本：
- luci-app-upnp：git-24.326.41750-70addca
- luci-i18n-upnp-zh-cn：git-24.326.41750-70addca
- miniupnpd：2.0.20170421-3

LuCI 源为 [coolsnowwolf/luci (openwrt-23.05)](https://github.com/coolsnowwolf/luci/tree/openwrt-23.05)，packages 源为 [coolsnowwolf/packages (master)](https://github.com/coolsnowwolf/packages/tree/master)

---

UPnP 界面不显示 活动的 UPnP 重定向：

[image]

但是用 upnpc-static 测试，映射可成功创建：

[image]

另一个问题是测试时必须带 -i 参数，否则会提示 `No valid UPNP Internet Gateway Device found.`

[image]

---

UPnP 配置：

```
$ cat /etc/config/upnpd

config upnpd 'config'
        option download '1024'
        option upload '512'
        option internal_iface 'lan'
        option port '5000'
        option upnp_lease_file '/var/upnp.leases'
        option enabled '1'
        option uuid '382eec07-ef49-444e-a23b-8cbb25c0a195'
        option log_output '1'
        option secure_mode '0'

config perm_rule
        option action 'allow'
        option ext_ports '1024-65535'
        option int_addr '0.0.0.0/0'
        option int_ports '1024-65535'
        option comment 'Allow high ports'

config perm_rule
        option action 'deny'
        option ext_ports '0-65535'
        option int_addr '0.0.0.0/0'
        option int_ports '0-65535'
        option comment 'Default deny'
```

---

类似 issue：
- https://github.com/coolsnowwolf/lede/issues/12554
- https://github.com/coolsnowwolf/lede/issues/12550

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86_64

### 详细日志

Mon Nov 25 18:33:54 2024 daemon.notice miniupnpd[11553]: shutting down MiniUPnPd
Mon Nov 25 18:33:54 2024 daemon.notice miniupnpd[19645]: HTTP listening on port 5000
Mon Nov 25 18:33:54 2024 daemon.notice miniupnpd[19645]: Listening for NAT-PMP/PCP traffic on port 5351
Mon Nov 25 18:33:55 2024 daemon.err miniupnpd[19645]: upnp_event_recv: recv(): Connection reset by peer

EdoQ, 2024-11-28:
同样的问题

nb369, 2024-11-28:
6.1.112也有这问题，测试upnp开启了！但是不显示活动连接

coffeelb, 2024-12-03:
老版本luci正常
2305luci的UPNP一直有问题

mygithub1981, 2025-01-21:
同样问题

conscloud, 2025-01-23:
一样的问题

aiamadeus, 2025-01-27:
修好了，update feeds

closed 2025-01-31 by zjjxwhh

1WuNeiGui, 2025-03-02:
我在别的平台上看有人说不编译eqos之类的插件就好了，我自己在RAX3000M上试了一下，确实可以了，upnp显示了，23.05的luci，