coolsnowwolf/lede #13124: 无网环境下编译报错

GiottoHuangjiyi, 2024-11-26:
### 详细叙述

目前需要在无网络的环境下编译。dl文件夹中的包已经预下载好了，
但是编译glib2过程中会提示cmake和pkg-config找不到pcre2然后进行下载，因为无网原因下载失败导致编译中断。
试过将pcre2-10.40.tar.bz2手动下载并放到dl文件夹中，但是仍然会出现这样的问题。
请问我应该将pcre2-10.40.tar.bz2放在哪里才能让这个编译流程顺利的跑下去？

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

none

### 详细日志

Run-time dependency iconv found: YES
Found pkg-config: /home/Giotto/workspace/ow/staging_dir/host/bin/pkg-config (1.8.0)
Found CMake: /home/Giotto/workspace/ow/staging_dir/host/bin/cmake (3.22.3)
Run-time dependency libpcre2-8 found: NO (tried pkgconfig and cmake)
Run-time dependency libpcre2-8 found: NO (tried pkgconfig and cmake)
Looking for a fallback subproject for the dependency libpcre2-8
Downloading pcre2 source from https://github.com/PhilipHazel/pcre2/releases/download/pcre2-10.40/pcre2-10.40.tar.bz2
<urlopen error [Errno -3] Temporary failure in name resolution>
A fallback URL could be specified using source_fallback_url key in the wrap file

../../../../build_dir/hostpkg/glib-2.74.0/meson.build:2062:2: ERROR: could not get https://github.com/PhilipHazel/pcre2/releases/download/pcre2-10.40/pcre2-10.40.tar.bz2 is the internet available?

A full log can be found at /home/Giotto/workspace/ow/build_dir/hostpkg/glib-2.74.0/openwrt-build/meson-logs/meson-log.txt
make[3]: *** [Makefile:123: /home/Giotto/workspace/ow/build_dir/hostpkg/glib-2.74.0/.configured] Error 1
make[3]: Leaving directory '/home/Giotto/workspace/ow/cz_external/packages/libs/glib2'
time: package/feeds/packages/glib2/host-compile#6.28#2.43#19.56
    ERROR: package/feeds/packages/glib2 [host] failed to build.
make[2]: *** [package/Makefile:118: package/feeds/packages/glib2/host/compile] Error 1

xingshu1990, 2024-11-26:
猜测式回答：
放到dl目录，然后make clean以后
再重新执行编译命令

coolsnowwolf, 2026-09-05:
当前 LEDE 源码已具备正确的离线构建依赖关系，这个问题不能在当前 master 复现：

- `feeds/packages/libs/glib2/Makefile` 明确声明 `HOST_BUILD_DEPENDS:=pcre2/host libffi/host`
- 当前 PCRE2 host 版本为 `10.42`，缓存文件名应为 `dl/pcre2-10.42.tar.bz2`，不是旧日志中的 10.40
- 清理 GLib host 构建目录后，将 HTTP/HTTPS/ALL_PROXY 全部指向不可用的 `127.0.0.1:9`，执行 `make package/feeds/packages/glib2/host/compile -j30` 成功
- Meson 日志确认：`Run-time dependency libpcre2-8 found: YES 10.42`，没有进入 fallback 下载

issue 日志中的包路径是 `cz_external/packages/libs/glib2`，属于外部自定义包；其依赖声明没有保证先构建 `pcre2/host`。使用当前官方 feeds 包，并先执行 `make download -j30` 准备完整 `dl/` 后即可离线构建。关闭此 issue。

closed 2026-09-05 by coolsnowwolf