coolsnowwolf/lede #13125: serverchan/Server酱没有了，如何手动添加呢？

Nicepaul, 2024-11-26:
### 详细叙述

在feeds里加入src-git wechatpush https://github.com/tty228/luci-app-wechatpush.git
但是./scripts/feeds update -a && ./scripts/feeds install -a时报错
julien@julien:~/x64/lede$ ./scripts/feeds update -a && ./scripts/feeds install -a
Updating feed 'packages' from 'https://github.com/coolsnowwolf/packages' ...
Already up to date.
Updating feed 'luci' from 'https://github.com/coolsnowwolf/luci.git;openwrt-23.05' ...
Already up to date.
Updating feed 'routing' from 'https://github.com/coolsnowwolf/routing' ...
Already up to date.
Updating feed 'telephony' from 'https://github.com/coolsnowwolf/telephony.git' ...
Already up to date.
Updating feed 'passwall_packages' from 'https://github.com/xiaorouji/openwrt-passwall-packages.git;main' ...
Already up to date.
Updating feed 'passwall' from 'https://github.com/xiaorouji/openwrt-passwall.git;main' ...
Already up to date.
Updating feed 'nezha' from 'https://github.com/Erope/openwrt_nezha' ...
Already up to date.
Updating feed 'wechatpush' from 'https://github.com/tty228/luci-app-wechatpush.git' ...
Cloning into './feeds/wechatpush'...
remote: Enumerating objects: 58, done.
remote: Counting objects: 100% (58/58), done.
remote: Compressing objects: 100% (42/42), done.
remote: Total 58 (delta 2), reused 40 (delta 1), pack-reused 0 (from 0)
Unpacking objects: 100% (58/58), 102.76 KiB | 312.00 KiB/s, done.
Create index file './feeds/packages.index' 
Create index file './feeds/luci.index' 
Create index file './feeds/routing.index' 
Create index file './feeds/telephony.index' 
Create index file './feeds/passwall_packages.index' 
Create index file './feeds/passwall.index' 
Create index file './feeds/nezha.index' 
Create index file './feeds/wechatpush.index' 
grep: feeds/wechatpush/Makefile:#: No such file or directory
grep: call: No such file or directory
grep: BuildPackage: No such file or directory
grep: OpenWrt: No such file or directory
grep: buildroot: No such file or directory
grep: signature/Makefile: No such file or directory
/home/julien/x64/lede/feeds/wechatpush.tmp/info/.files-packageinfo.mk:1: *** target pattern contains no '%'.  Stop.
Collecting target info: done
Ignoring feed 'wechatpush' - index missing
Installing all packages from feed packages.
Installing all packages from feed luci.
Installing all packages from feed routing.
Installing all packages from feed telephony.
Installing all packages from feed passwall_packages.
Installing all packages from feed passwall.
Installing all packages from feed nezha.
Installing all packages from feed wechatpush.
Ignoring feed 'wechatpush' - index missing
Ignoring feed 'wechatpush' - index missing
Ignoring feed 'wechatpush' - index missing
julien@julien:~/x64/lede$

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x64

### 详细日志

julien@julien:~/x64/lede$ ./scripts/feeds update -a && ./scripts/feeds install -a
Updating feed 'packages' from 'https://github.com/coolsnowwolf/packages' ...
Already up to date.
Updating feed 'luci' from 'https://github.com/coolsnowwolf/luci.git;openwrt-23.05' ...
Already up to date.
Updating feed 'routing' from 'https://github.com/coolsnowwolf/routing' ...
Already up to date.
Updating feed 'telephony' from 'https://github.com/coolsnowwolf/telephony.git' ...
Already up to date.
Updating feed 'passwall_packages' from 'https://github.com/xiaorouji/openwrt-passwall-packages.git;main' ...
Already up to date.
Updating feed 'passwall' from 'https://github.com/xiaorouji/openwrt-passwall.git;main' ...
Already up to date.
Updating feed 'nezha' from 'https://github.com/Erope/openwrt_nezha' ...
Already up to date.
Updating feed 'wechatpush' from 'https://github.com/tty228/luci-app-wechatpush.git' ...
Cloning into './feeds/wechatpush'...
remote: Enumerating objects: 58, done.
remote: Counting objects: 100% (58/58), done.
remote: Compressing objects: 100% (42/42), done.
remote: Total 58 (delta 2), reused 40 (delta 1), pack-reused 0 (from 0)
Unpacking objects: 100% (58/58), 102.76 KiB | 312.00 KiB/s, done.
Create index file './feeds/packages.index' 
Create index file './feeds/luci.index' 
Create index file './feeds/routing.index' 
Create index file './feeds/telephony.index' 
Create index file './feeds/passwall_packages.index' 
Create index file './feeds/passwall.index' 
Create index file './feeds/nezha.index' 
Create index file './feeds/wechatpush.index' 
grep: feeds/wechatpush/Makefile:#: No such file or directory
grep: call: No such file or directory
grep: BuildPackage: No such file or directory
grep: OpenWrt: No such file or directory
grep: buildroot: No such file or directory
grep: signature/Makefile: No such file or directory
/home/julien/x64/lede/feeds/wechatpush.tmp/info/.files-packageinfo.mk:1: *** target pattern contains no '%'.  Stop.
Collecting target info: done
Ignoring feed 'wechatpush' - index missing
Installing all packages from feed packages.
Installing all packages from feed luci.
Installing all packages from feed routing.
Installing all packages from feed telephony.
Installing all packages from feed passwall_packages.
Installing all packages from feed passwall.
Installing all packages from feed nezha.
Installing all packages from feed wechatpush.
Ignoring feed 'wechatpush' - index missing
Ignoring feed 'wechatpush' - index missing
Ignoring feed 'wechatpush' - index missing
julien@julien:~/x64/lede$

NothingNA, 2024-12-04:
新版不是改名叫wechatpush了么？

sgx800, 2024-12-07:
现在叫wechatpush，可以自己添加，期待Lean大集成
https://github.com/tty228/luci-app-wechatpush

mrclock, 2024-12-11:
我也是同样的问题

coolsnowwolf, 2026-09-05:
这个问题在当前源码中已经解决。`luci-app-wechatpush` 已直接集成到 LEDE 使用的 LuCI feed，不需要再把单包仓库作为 feed 添加。

已验证：
- 当前 `feeds/luci` 的 `openwrt-25.12` 分支包含 `luci-app-wechatpush 3.6.12`
- x86_64 下执行 `make package/feeds/luci/luci-app-wechatpush/compile -j30 CONFIG_PACKAGE_luci-app-wechatpush=m` 成功
- 生成 `luci-app-wechatpush_3.6.12-r1_all.ipk`（81718 bytes）

原日志报错是因为 `tty228/luci-app-wechatpush` 是单包仓库，直接写成 `src-git wechatpush ...` 后，feed 索引器会把仓库根目录内容误当作 feed 目录扫描。当前直接在菜单中选择 LuCI → Applications → luci-app-wechatpush 即可。关闭此 issue。

closed 2026-09-05 by coolsnowwolf