coolsnowwolf/lede #13128: 新版LEDE的改动，没有办法修改默认的IP，主机名了

Cyanro, 2024-11-26:
### 详细叙述

大概是10月份左右，我编译了一次LEDE，但是结果和以往都不一样，用的是git action，
1.默认IP无法更改成功了
2.默认主机名也无法修改了
3.代码文件里面的主题名称也不见了，只要编译时勾选了argon, 它就是默认的主题了，不用手动切换，这和以往都不一样
git action的流程是没有变的，之前也是成功的，然后我在本地编译结果也是一样的，
所以我改怎么确定到底啥什么问题，其它认有遇见过吗

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86_64

### 详细日志

sed -i 's/192.168.1.1/192.168.2.1/g' package/base-files/files/bin/config_generate
sed -i 's/luci-theme-bootstrap/luci-theme-argon/g' feeds/luci/collections/luci/Makefile
sed -i 's/OpenWrt/CyanTec/g' package/base-files/files/bin/config_generate

Cyanro, 2024-11-26:
太纠结了，到底是代码的问题，还是我的问题啊，我的action流程根本没有改动，而且我也本地编译了，默认IP任然是192.168.1.1， 这个再之前改起来很简单的，真不知道为什么

tonylee2022, 2024-11-26:
> package/base-files/luci2/bin/config_generate 试试这里

改这里也没用，另luci默认23.05分支后，samba4,aria2等不能正确分类到菜单目录下，都在服务项下面。修改luci-app中的Makefile可以改变安装菜单吗？

closed 2024-12-01 by Cyanro