coolsnowwolf/lede #13130: mt76 无线驱动模块在内核6.12中编译出错

ntbowen, 2024-11-27:
### 详细叙述

mt76 无线驱动模块在内核6.12中编译出错

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

X86

### 详细日志

make[3]: Entering directory '/home/ntbowen/openwrt/lede/package/kernel/mt76'
rm -f /home/ntbowen/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/mt76-2023-08-14-b14c2351/.built
touch /home/ntbowen/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/mt76-2023-08-14-b14c2351/.built_check
make -C /home/ntbowen/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.1 KCFLAGS="-fmacro-prefix-map=/home/ntbowen/openwrt/lede/build_dir/target-x86_64_musl=target-x86_64_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/ntbowen/openwrt/lede/staging_dir/host/include -I/home/ntbowen/openwrt/lede/staging_dir/hostpkg/include -I/home/ntbowen/openwrt/lede/staging_dir/target-x86_64_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Tue Nov 26 15:00:20 2024" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/ntbowen/openwrt/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/home/ntbowen/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/button-hotplug.symvers /home/ntbowen/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/cryptodev-linux.symvers /home/ntbowen/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/dmx_usb_module.symvers /home/ntbowen/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/fullconenat.symvers /home/ntbowen/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/mac80211.symvers /home/ntbowen/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/macremapper.symvers /home/ntbowen/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/oaf.symvers /home/ntbowen/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/r8101.symvers /home/ntbowen/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/r8125.symvers /home/ntbowen/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/r8126.symvers /home/ntbowen/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/r8152.symvers /home/ntbowen/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/r8168.symvers /home/ntbowen/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/siit.symvers /home/ntbowen/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/v4l2loopback.symvers" KERNELRELEASE=6.12.1  CONFIG_MT76_USB=m CONFIG_MT76_CONNAC_LIB=m CONFIG_MT76_SDIO=m CONFIG_MT792x_LIB=m CONFIG_MT792x_USB=m CONFIG_MT7921_COMMON=m CONFIG_MT7921U=m CONFIG_MT7921S=m CONFIG_MT7921E=m M="/home/ntbowen/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/mt76-2023-08-14-b14c2351" NOSTDINC_FLAGS="-nostdinc  -isystem /home/ntbowen/openwrt/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/11.3.0/include -I/home/ntbowen/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/mt76-2023-08-14-b14c2351 -I/home/ntbowen/openwrt/lede/staging_dir/target-x86_64_musl/usr/include/mac80211-backport/uapi -I/home/ntbowen/openwrt/lede/staging_dir/target-x86_64_musl/usr/include/mac80211-backport -I/home/ntbowen/openwrt/lede/staging_dir/target-x86_64_musl/usr/include/mac80211/uapi -I/home/ntbowen/openwrt/lede/staging_dir/target-x86_64_musl/usr/include/mac80211 -include backport/autoconf.h -include backport/backport.h -DCONFIG_MAC80211_MESH" modules
make[4]: Entering directory '/home/ntbowen/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.1'
  CC [M]  /home/ntbowen/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/mt76-2023-08-14-b14c2351/mac80211.o
/home/ntbowen/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/mt76-2023-08-14-b14c2351/mac80211.c: In function 'mt76_create_page_pool':
/home/ntbowen/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/mt76-2023-08-14-b14c2351/mac80211.c:573:26: error: 'PP_FLAG_PAGE_FRAG' undeclared (first use in this function)
  573 |                 .flags = PP_FLAG_PAGE_FRAG,
      |                          ^~~~~~~~~~~~~~~~~
/home/ntbowen/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/mt76-2023-08-14-b14c2351/mac80211.c:573:26: note: each undeclared identifier is reported only once for each function it appears in
make[6]: *** [scripts/Makefile.build:229: /home/ntbowen/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/mt76-2023-08-14-b14c2351/mac80211.o] Error 1
make[5]: *** [/home/ntbowen/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.1/Makefile:1936: /home/ntbowen/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/mt76-2023-08-14-b14c2351] Error 2
make[4]: *** [Makefile:224: __sub-make] Error 2
make[4]: Leaving directory '/home/ntbowen/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.12.1'
make[3]: *** [Makefile:670: /home/ntbowen/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/mt76-2023-08-14-b14c2351/.built] Error 2
make[3]: Leaving directory '/home/ntbowen/openwrt/lede/package/kernel/mt76'
time: package/kernel/mt76/compile#0.50#0.32#0.81
    ERROR: package/kernel/mt76 failed to build.
make[2]: *** [package/Makefile:116: package/kernel/mt76/compile] Error 1
make[2]: Leaving directory '/home/ntbowen/openwrt/lede'
make[1]: *** [package/Makefile:110: /home/ntbowen/openwrt/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ntbowen/openwrt/lede'
make: *** [/home/ntbowen/openwrt/lede/include/toplevel.mk:231：world] 错误 2

coolsnowwolf, 2024-11-29:
已添加 6.12 内核支持

ntbowen, 2024-11-29:
> 已添加 6.12 内核支持

编译出现新问题：
make[3]: Entering directory '/home/ntbowen/openwrt/lede/package/kernel/mt76'
touch /home/ntbowen/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/mt76-2024-04-04-8f301a5c/.prepared_4d022cb42f072d5221acf5b92c2b1614_6664517399ebbbc92a37c5bb081b5c53_check
. /home/ntbowen/openwrt/lede/include/shell.sh; xzcat /home/ntbowen/openwrt/lede/dl/mt76-2024-04-04-8f301a5c.tar.xz | tar -C /home/ntbowen/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/mt76-2024-04-04-8f301a5c/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /home/ntbowen/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/mt76-2024-04-04-8f301a5c

Applying ./patches-6.x/001-allow-vht-on-2g.patch using plaintext: 
patching file mac80211.c
Hunk #1 succeeded at 293 (offset 11 lines).
Hunk #2 succeeded at 360 (offset 11 lines).
Hunk #3 succeeded at 523 (offset 15 lines).
Hunk #4 succeeded at 709 (offset 18 lines).

Applying ./patches-6.x/002-mt7603-fix-mixed-code.patch using plaintext: 
patching file mt7603/dma.c

Applying ./patches-6.x/003-wifi-mt76-mt7915-fix-oops.patch using plaintext: 
patching file mac80211.c

Applying ./patches-6.x/101-fixes-mt7925-build-error.patch using plaintext: 
patching file mt7925/main.c
patching file mt7925/mcu.c

Applying ./patches-6.x/200-fix-kernel-6.12-support.patch using plaintext: 
patching file mac80211.c
Hunk #1 succeeded at 581 (offset 11 lines).
touch /home/ntbowen/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/mt76-2024-04-04-8f301a5c/.prepared_4d022cb42f072d5221acf5b92c2b1614_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/ntbowen/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/mt76-2024-04-04-8f301a5c/.configured_*
rm -f /home/ntbowen/openwrt/lede/staging_dir/target-x86_64_musl/stamp/.mt76_installed
mkdir -p /home/ntbowen/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/mt76-2024-04-04-8f301a5c/tools
(cd /home/ntbowen/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/mt76-2024-04-04-8f301a5c/tools; CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/ntbowen/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/mt76-2024-04-04-8f301a5c=mt76-2024-04-04-8f301a5c -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/ntbowen/openwrt/lede/staging_dir/target-x86_64_musl/usr/include/libnl-tiny " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/ntbowen/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/mt76-2024-04-04-8f301a5c=mt76-2024-04-04-8f301a5c -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/ntbowen/openwrt/lede/staging_dir/target-x86_64_musl/usr/include/libnl-tiny " LDFLAGS="-L/home/ntbowen/openwrt/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/home/ntbowen/openwrt/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -znow -zrelro " cmake -DCMAKE_SYSTEM_NAME=Linux -DCMAKE_SYSTEM_VERSION=1 -DCMAKE_SYSTEM_PROCESSOR=x86_64 -DCMAKE_BUILD_TYPE=Release -DCMAKE_C_FLAGS_RELEASE="-DNDEBUG" -DCMAKE_CXX_FLAGS_RELEASE="-DNDEBUG" -DCMAKE_C_COMPILER_LAUNCHER="" -DCMAKE_C_COMPILER="/home/ntbowen/openwrt/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/bin/x86_64-openwrt-linux-musl-gcc" -DCMAKE_CXX_COMPILER_LAUNCHER="" -DCMAKE_CXX_COMPILER="/home/ntbowen/openwrt/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/bin/x86_64-openwrt-linux-musl-g++" -DCMAKE_ASM_COMPILER_LAUNCHER="" -DCMAKE_ASM_COMPILER="/home/ntbowen/openwrt/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/bin/x86_64-openwrt-linux-musl-gcc" -DCMAKE_EXE_LINKER_FLAGS:STRING="-L/home/ntbowen/openwrt/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/home/ntbowen/openwrt/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -znow -zrelro" -DCMAKE_MODULE_LINKER_FLAGS:STRING="-L/home/ntbowen/openwrt/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/home/ntbowen/openwrt/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -znow -zrelro -Wl,-Bsymbolic-functions" -DCMAKE_SHARED_LINKER_FLAGS:STRING="-L/home/ntbowen/openwrt/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/home/ntbowen/openwrt/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -znow -zrelro -Wl,-Bsymbolic-functions" -DCMAKE_AR="/home/ntbowen/openwrt/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/bin/x86_64-openwrt-linux-musl-gcc-ar" -DCMAKE_NM="/home/ntbowen/openwrt/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/bin/x86_64-openwrt-linux-musl-gcc-nm" -DCMAKE_RANLIB="/home/ntbowen/openwrt/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/bin/x86_64-openwrt-linux-musl-gcc-ranlib" -DCMAKE_FIND_ROOT_PATH="/home/ntbowen/openwrt/lede/staging_dir/target-x86_64_musl/usr;/home/ntbowen/openwrt/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl" -DCMAKE_FIND_ROOT_PATH_MODE_PROGRAM=BOTH -DCMAKE_FIND_ROOT_PATH_MODE_LIBRARY=ONLY -DCMAKE_FIND_ROOT_PATH_MODE_INCLUDE=ONLY -DCMAKE_STRIP=: -DCMAKE_INSTALL_PREFIX=/usr -DDL_LIBRARY=/home/ntbowen/openwrt/lede/staging_dir/target-x86_64_musl -DCMAKE_PREFIX_PATH=/home/ntbowen/openwrt/lede/staging_dir/target-x86_64_musl -DCMAKE_SKIP_RPATH=TRUE -DCMAKE_EXPORT_PACKAGE_REGISTRY=FALSE -DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=TRUE -DCMAKE_FIND_USE_PACKAGE_REGISTRY=FALSE -DCMAKE_FIND_PACKAGE_NO_PACKAGE_REGISTRY=TRUE -DCMAKE_FIND_USE_SYSTEM_PACKAGE_REGISTRY=FALSE -DCMAKE_FIND_PACKAGE_NO_SYSTEM_PACKAGE_REGISTRY=TRUE  /home/ntbowen/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/mt76-2024-04-04-8f3 [... cut ...]
make[3]: Leaving directory '/home/ntbowen/openwrt/lede/package/kernel/mt76'
time: package/kernel/mt76/compile#7.63#2.08#9.68
    ERROR: package/kernel/mt76 failed to build.
make[2]: *** [package/Makefile:116: package/kernel/mt76/compile] Error 1
make[2]: Leaving directory '/home/ntbowen/openwrt/lede'
make[1]: *** [package/Makefile:110: /home/ntbowen/openwrt/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ntbowen/openwrt/lede'
make: *** [/home/ntbowen/openwrt/lede/include/toplevel.mk:231：world] 错误 2

qingtian110, 2024-12-01:
对于 Linux 6.12  如果有一定动手能力的话。删除源码自带的 mac80211 和 mt76 

之后

使用这个 80211 https://github.com/sbwml/package_kernel_mac80211

使用 openwrt 上游最新的 mt76 https://github.com/openwrt/openwrt/tree/openwrt-24.10/package/kernel/mt76
加上这两个 patch  https://github.com/sbwml/r4s_build_script/tree/master/openwrt/patch/mt76/patches

你就能完美编译 所有 mt76 和 mac80211 的包在 linux 6.12 上。不过操作起来真的繁杂，但对于需求明确的情况，肯定不是难事。

这个方式有成功案例的   https://kmod.cooluc.com/x86_64/6.12.1~b31cc72d830df5c5b99c2048a5e26a5c-r1/

[image]

ntbowen, 2024-12-01:
> mac80211

感谢你的建议，但是动mac80211牵扯的面就大了，基本上所有无线网卡的驱动都会受影响。我不仅仅是要编译mt76，事实上我根据你的建议改动mac80211后再次编译就会出现ath11的驱动编译出错，这在改mac80211之前是可以正常编译的。

ntbowen, 2024-12-01:
> 对于 Linux 6.12 如果有一定动手能力的话。删除源码自带的 mac80211 和 mt76
>
> 之后
>
> 使用这个 80211 https://github.com/sbwml/package_kernel_mac80211
>
> 使用 openwrt 上游最新的 mt76 [...]

感谢你的建议，但是动mac80211牵扯的面就大了，基本上所有无线网卡的驱动都会受影响。我不仅仅是要编译mt76，事实上我根据你的建议改动mac80211后再次编译就会出现ath11的驱动编译出错，这在改mac80211之前是可以正常编译的。

coolsnowwolf, 2026-09-05:
已在当前 master 重新验证，这个问题已经由新版 mt76 解决：

- 当前 mt76：`2026-09-01-be5ce791`（升级提交 `8f7aa30c44b3`）
- 当前 mac80211/backports：`6.18.39`，并保留旧内核使用的 `6.6.15` 分支
- 使用 x86_64 / Linux `6.12.108`，显式选中 `kmod-mt7603` 执行 `make package/kernel/mt76/compile -j30 CONFIG_PACKAGE_kmod-mt7603=m`，编译成功
- 产物：`mt7603e.ko`（725528 bytes）及 `kmod-mt7603_6.12.108+2026-09-01-be5ce791-1_x86_64.ipk`（104650 bytes）

issue 中的 `PP_FLAG_PAGE_FRAG` 和 `.remove` 回调类型错误均来自旧版 mt76，当前源码已不再复现。关闭此 issue。

closed 2026-09-05 by coolsnowwolf