coolsnowwolf/lede #13132: 3588aib installation NFA765 network card cannot be edited

JDKCHINA, 2024-11-27:
### 详细叙述

[image]
TypeError
Cannot read properties of undefined (reading

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

seewo-scrm3588_io

### 详细日志

It's not log,is luci question,please fix it,thanks.

JDKCHINA, 2024-11-27:
编译日志无报错，就是luci编辑wifi报错，无法编辑，无日志

jidanyunmian, 2024-11-27:
你编译的时候选的什么驱动 我也是这个网卡 编译出来没有无线的选项

JDKCHINA, 2024-11-27:
> 你编译的时候选的什么驱动 我也是这个网卡 编译出来没有无线的选项

>
[image]

jidanyunmian, 2024-11-27:
> ![Image_594238499698418](https://private-user-images.githubusercontent.com/14959641/390410049-d3af9e63-64cb-427e-858d-7fefc123a292.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI3MTQ1MDAsIm5iZiI6MTczMjcxNDIwMCwicGF0aCI6Ii8xNDk1OTY0MS8zOTA0MTAwNDktZDNhZjllNjMtNjRjYi00MjdlLTg1OGQtN2ZlZmMxMjNhMjkyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTI3VDEzMzAwMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWUxM2ZhNjM4MGIzZjAzNTkyZjUyYzE0YTA1NzVlOTBmNTY1MWZmMWVkMmQ5MDVlODdlODRmNThhNjQ1MTMxOTQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.WRCQxE-ncNHzRnU5CXu5G4iNq6LQ-c8e1Sc3iH0Wf4A) 这3个驱动

好的谢谢 我去编译下

jidanyunmian, 2024-11-28:
> ![Image_594238499698418](https://private-user-images.githubusercontent.com/14959641/390410049-d3af9e63-64cb-427e-858d-7fefc123a292.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29t [...]

我发现问题 目前最新内核版本6.6 我改成5.15会编译不过 报错    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
搜了一下发现是5.15的问题

jidanyunmian, 2024-11-29:
编译atk11k 6.1内核的版本后 一样出错 错误如楼主图 一模一样

jidanyunmian, 2024-12-01:
> > 编译atk11k 6.1内核的版本后 一样出错 错误如楼主图 一模一样
> 
> 我的x86 6.1内核版本正常使用

方便提供一份 .config 么 我用你的配置编译一遍看看

wherss1, 2024-12-25:
7922也和你一样报错

closed 2025-01-31 by aiamadeus