coolsnowwolf/lede #13140: tools_compile_nyyynyynnyyynynyyyyyyynyyynnynyynnynnyyyyyyyyyyyyyyyyynynnyyyyyyyy] 错误

LIUBANGJIAN, 2024-12-01:
### 详细叙述

mkdir -p /home/runner/work/lede/lede/staging_dir/host/stamp
touch /home/runner/work/lede/lede/build_dir/host/cmake-3.25.2/.built
touch /home/runner/work/lede/lede/staging_dir/host/stamp/.cmake_installed
make[3]: Leaving directory '/home/runner/work/lede/lede/tools/cmake'
time: tools/cmake/compile#1410.53#129.30#502.14
make[2]: Leaving directory '/home/runner/work/lede/lede'
make[1]: *** [tools/Makefile:218: /home/runner/work/lede/lede/staging_dir/host/stamp/.tools_compile_nyyynyynnyyynynyyyyyyynyyynnynyynnynnyyyyyyyyyyyyyyyyynynnyyyyyyyy] Error 2
make[1]: Leaving directory '/home/runner/work/lede/lede'
make: *** [/home/runner/work/lede/lede/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

没改编译配置，这两天又编译错误了。请问是什么地方有问题呢？

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

mkdir -p /home/runner/work/lede/lede/staging_dir/host/stamp
touch /home/runner/work/lede/lede/build_dir/host/cmake-3.25.2/.built
touch /home/runner/work/lede/lede/staging_dir/host/stamp/.cmake_installed
make[3]: Leaving directory '/home/runner/work/lede/lede/tools/cmake'
time: tools/cmake/compile#1410.53#129.30#502.14
make[2]: Leaving directory '/home/runner/work/lede/lede'
make[1]: *** [tools/Makefile:218: /home/runner/work/lede/lede/staging_dir/host/stamp/.tools_compile_nyyynyynnyyynynyyyyyyynyyynnynyynnynnyyyyyyyyyyyyyyyyynynnyyyyyyyy] Error 2
make[1]: Leaving directory '/home/runner/work/lede/lede'
make: *** [/home/runner/work/lede/lede/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

没改编译配置，这两天又编译错误了。请问是什么地方有问题呢？

Kuron88, 2024-12-01:
ax1800 pro一样的问题

coolsnowwolf, 2024-12-01:
rm -rf ./build_dir , rm -rf ./staging_dir , 然后重新编译就好了

LIUBANGJIAN, 2024-12-01:
> rm -rf ./build_dir , rm -rf ./staging_dir , 然后重新编译就好了

github action编译也是要这样操作吗？

LIUBANGJIAN, 2024-12-01:
> rm -rf ./build_dir , rm -rf ./staging_dir , 然后重新编译就好了

在什么目录执行呢？

coolsnowwolf, 2024-12-01:
发现了问题，上游的 ci 改了 ubuntu 24.10.1 导致 flex 工具编译失败。 ci 要改回 24.04 lts 版本

LIUBANGJIAN, 2024-12-01:
> 发现了问题，上游的 ci 改了 ubuntu 24.10.1 导致 flex 工具编译失败。 ci 要改回 24.04 lts 版本
辛苦大佬

closed 2024-12-01 by coolsnowwolf