coolsnowwolf/lede #13141: 编译报错

Kuron88, 2024-12-01:
### 详细叙述

编译报错

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

AX1800 PRO亚瑟

### 详细日志

checking for GNU gettext in libc... yes
checking whether to use NLS... yes
checking where the gettext function comes from... libc
checking for bison... bison -y
checking for flex... flex
checking lex output file root... lex.yy
checking lex library... -lfl
checking whether yytext is a pointer... yes
checking whether the C compiler works... no
configure: error: in `/home/mi/lede/build_dir/host/flex-2.6.4':
configure: error: C compiler cannot create executables
See `config.log' for more details
make[3]: *** [Makefile:39: /home/mi/lede/build_dir/host/flex-2.6.4/.configured] Error 77
make[3]: Leaving directory '/home/mi/lede/tools/flex'
time: tools/flex/compile#12.46#2.52#15.76
    ERROR: tools/flex failed to build.
make[2]: *** [tools/Makefile:222: tools/flex/compile] Error 1
make[2]: Leaving directory '/home/mi/lede'
make[1]: *** [tools/Makefile:218: /home/mi/lede/staging_dir/host/stamp/.tools_compile_nyyynyynnyyynynyyyyyyynyyynnynyynnynnyyyyyyyyyyyyyyyyynynnyyyyyyyy] Error 2
make[1]: Leaving directory '/home/mi/lede'
make: *** [/home/mi/lede/include/toplevel.mk:231：world] 错误 2

binge8, 2024-12-01:
这个问题在已经关闭的issue里发过了，也已经修复了，你再试试吧

Kuron88, 2024-12-01:
我刚刚试了就是报这个错，更新到最新的源码了

Kuron88, 2024-12-01:
我用高通的没修复

evenwangcn, 2024-12-01:
+1，编译X86_64的报同样的错误

coolsnowwolf, 2024-12-01:
发现了问题，上游的 ci 改了 ubuntu 24.10.1 导致 flex 工具编译失败。 ci 要改回 24.04 lts 版本

LIUBANGJIAN, 2024-12-02:
> 发现了问题，上游的 ci 改了 ubuntu 24.10.1 导致 flex 工具编译失败。 ci 要改回 24.04 lts 版本

大佬，这是你这边改，还是我们这边改？

lj267lj267, 2024-12-02:
请问题主AX1800 PRO亚瑟 这个型号在哪个系列里选择？没有ipq6000这个系列啊？

LIUBANGJIAN, 2024-12-02:
编译了一天还是错误

evenwangcn, 2024-12-02:
ci重新改成了ubuntu-latest，看日志已经改回到24.04.5了，可以编译了

closed 2024-12-02 by coolsnowwolf