coolsnowwolf/lede #13142: openwrt还是使用cgroup v1吗？

gitbruc, 2024-12-01:
### 详细叙述

现在docker使用的是cgroup v2，导致docker的内存信息无法正常显示
docker info | grep Cgroup
 Cgroup Driver: cgroupfs
 Cgroup Version: 2

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86,旧版luci

### 详细日志

docker stats
CONTAINER ID   NAME                               CPU %     MEM USAGE / LIMIT   MEM %     NET I/O           BLOCK I/O   PIDS
1dc864a268a2   cadvisor                           0.00%     0B / 0B             0.00%     214kB / 8.49MB    0B / 0B     0
39e193d96e5d   redis2                             0.00%     0B / 0B             0.00%     9.52kB / 0B       0B / 0B     0
8c1e167266ba   node-exporter                      0.00%     0B / 0B             0.00%     97.3kB / 939kB    0B / 0B     0

coolsnowwolf, 2024-12-01:
对，可以考虑转移到 v2 了

gitbruc, 2024-12-01:
> 对，可以考虑转移到 v2 了

要换23.05吗L大

liwenjie119, 2024-12-01:
转吧，就是一大堆app估计没人更新维护了。不过也不能死守旧版了，该与时俱进了

gitbruc, 2024-12-01:
> 转吧，就是一大堆app估计没人更新维护了。不过也不能死守旧版了，该与时俱进了

好吧，我就是怕很多软件没法用了😄️

closed 2024-12-01 by gitbruc