coolsnowwolf/lede #13144: libc.so.6: version `GLIBC_2.33' not found

lj267lj267, 2024-12-02:
### 详细叙述

借宝地问问openwrt22.03.6 官方sdk用ubuntu 24.04 和 22.04 编译都出现libc.so.6: version `GLIBC_2.33' not found 这个错误编译不过去是怎么回事。谁能帮帮忙

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

wndr3800

### 详细日志

touch /home/lj/sdk/staging_dir/target-mips_24kc_musl/stamp/.usign_installed
touch -r /home/lj/sdk/build_dir/target-mips_24kc_musl/usign-2020-05-23-f1f65026/.built /home/lj/sdk/build_dir/target-mips_24kc_musl/usign-2020-05-23-f1f65026/.autoremove 2>/dev/null >/dev/null
find /home/lj/sdk/build_dir/target-mips_24kc_musl/usign-2020-05-23-f1f65026 -mindepth 1 -maxdepth 1 -not '(' -type f -and -name '.*' -and -size 0 ')' -and -not -name '.pkgdir'  -print0 | xargs -r -0 rm -rf
make[3]: Leaving directory '/home/lj/sdk/feeds/base/package/system/usign'
time: package/feeds/base/usign/compile#2.33#1.20#2.98
make[3]: Entering directory '/home/lj/sdk/feeds/base/package/base-files'
touch /home/lj/sdk/build_dir/target-mips_24kc_musl/linux-ath79_generic/base-files/.prepared_7302616adc3fcfd59569538dc880250b_18f1e190c5d53547fed41a3eaa76e9e9_check
mkdir -p /home/lj/sdk/build_dir/target-mips_24kc_musl/linux-ath79_generic/base-files
touch /home/lj/sdk/build_dir/target-mips_24kc_musl/linux-ath79_generic/base-files/.prepared_7302616adc3fcfd59569538dc880250b_18f1e190c5d53547fed41a3eaa76e9e9
rm -f /home/lj/sdk/build_dir/target-mips_24kc_musl/linux-ath79_generic/base-files/.configured_*
rm -f /home/lj/sdk/staging_dir/target-mips_24kc_musl/stamp/.base-files_installed
[ -s /home/lj/sdk/key-build -a -s /home/lj/sdk/key-build.pub ] || /home/lj/sdk/staging_dir/host/bin/usign -G -s /home/lj/sdk/key-build -p /home/lj/sdk/key-build.pub -c "Local build key"
[ -s /home/lj/sdk/key-build.ucert ] || /home/lj/sdk/staging_dir/host/bin/ucert -I -c /home/lj/sdk/key-build.ucert -p /home/lj/sdk/key-build.pub -s /home/lj/sdk/key-build
/home/lj/sdk/staging_dir/host/bin/ucert: /home/lj/sdk/staging_dir/host/lib/libc.so.6: version `GLIBC_2.33' not found (required by /home/lj/sdk/staging_dir/host/bin/ucert)
/home/lj/sdk/staging_dir/host/bin/ucert: /home/lj/sdk/staging_dir/host/lib/libc.so.6: version `GLIBC_2.34' not found (required by /home/lj/sdk/staging_dir/host/bin/ucert)
/home/lj/sdk/staging_dir/host/bin/ucert: /home/lj/sdk/staging_dir/host/lib/libc.so.6: version `GLIBC_2.33' not found (required by /home/lj/sdk/staging_dir/host/lib/libblobmsg_json.so)
make[3]: *** [Makefile:241: /home/lj/sdk/build_dir/target-mips_24kc_musl/linux-ath79_generic/base-files/.configured_0ffb4af4d14f246af9ebdccba2f49a11_8e081b74cf069e1e6800a5bbcbb282f0] Error 1
make[3]: Leaving directory '/home/lj/sdk/feeds/base/package/base-files'
time: package/feeds/base/base-files/compile#0.62#0.34#0.80
    ERROR: package/feeds/base/base-files failed to build.
make[2]: *** [package/Makefile:120: package/feeds/base/base-files/compile] Error 1
make[2]: Leaving directory '/home/lj/sdk'
make[1]: *** [package/Makefile:114: /home/lj/sdk/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lj/sdk'
make: *** [/home/lj/sdk/include/toplevel.mk:223：world] 错误 2

yanxin152133, 2024-12-02:
这个原因？
[image]

lj267lj267, 2024-12-02:
> 这个原因？ [...]

lj@vm:~$ ldd --version
ldd (Ubuntu GLIBC 2.35-0ubuntu3.8) 2.35
Copyright (C) 2022 自由软件基金会。
这是自由软件；复制本软件需满足的条件请参见源代码。本软件不提供任何
品质保证；甚至不保证适销性或是适用于某种特定用途。
由 Roland McGrath 和 Ulrich Drepper 编写。
lj@vm:~$ 
可是我用的22.04和24.04 libc6版本都在2.35以上了啊？

lj267lj267, 2024-12-02:
> 这个原因？ [...]

我用的22.04和24.04版本的ubunt,glibc6版本都在2.35以上了啊，怎么还会要升级？
lj@vm:~$ ldd --version
ldd (Ubuntu GLIBC 2.35-0ubuntu3.8) 2.35
Copyright (C) 2022 自由软件基金会。
这是自由软件；复制本软件需满足的条件请参见源代码。本软件不提供任何
品质保证；甚至不保证适销性或是适用于某种特定用途。
由 Roland McGrath 和 Ulrich Drepper 编写。
lj@vm:~$

coolsnowwolf, 2024-12-03:
strings /lib/x86_64-linux-gnu/libc.so.6 |grep GLIBC_
看看，肯定就是缺了

closed 2024-12-03 by lj267lj267