coolsnowwolf/lede #13146: Hunk #1 FAILED at 2303. Hunk #2 succeeded at 2378 (offset 1 line).  我也是服了，openwrt的官方sdk换几个系统仍编译不过去

lj267lj267, 2024-12-02:
### 详细叙述

patching file drivers/net/ethernet/mediatek/mtk_eth_soc.c
Hunk #1 FAILED at 2303.
Hunk #2 succeeded at 2378 (offset 1 line).
Hunk #3 succeeded at 3109 (offset 1 line).
Hunk #4 succeeded at 3190 (offset 1 line).
Hunk #5 succeeded at 3199 (offset 1 line).

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

wndr3800

### 详细日志

patching file drivers/net/ethernet/mediatek/mtk_eth_soc.c
patching file drivers/net/ethernet/mediatek/mtk_eth_soc.h

Applying /home/lj/sdk/feeds/base/package/kernel/bpf-headers/../../../target/linux/generic/backport-5.10/610-v5.13-32-net-ethernet-mtk_eth_soc-add-support-for-initializin.patch using plaintext: 
patching file drivers/net/ethernet/mediatek/Makefile
patching file drivers/net/ethernet/mediatek/mtk_eth_soc.c
Hunk #1 FAILED at 2303.
Hunk #2 succeeded at 2378 (offset 1 line).
Hunk #3 succeeded at 3109 (offset 1 line).
Hunk #4 succeeded at 3190 (offset 1 line).
Hunk #5 succeeded at 3199 (offset 1 line).
1 out of 5 hunks FAILED -- saving rejects to file drivers/net/ethernet/mediatek/mtk_eth_soc.c.rej
patching file drivers/net/ethernet/mediatek/mtk_eth_soc.h
patching file drivers/net/ethernet/mediatek/mtk_ppe.c
patching file drivers/net/ethernet/mediatek/mtk_ppe.h
patching file drivers/net/ethernet/mediatek/mtk_ppe_debugfs.c
patching file drivers/net/ethernet/mediatek/mtk_ppe_regs.h
Patch failed!  Please fix /home/lj/sdk/feeds/base/package/kernel/bpf-headers/../../../target/linux/generic/backport-5.10/610-v5.13-32-net-ethernet-mtk_eth_soc-add-support-for-initializin.patch!
make[3]: *** [Makefile:124: /home/lj/sdk/build_dir/target-mips_24kc_musl/linux-ath79_generic/bpf-headers/linux-5.10.201/.prepared_97fda69a651f361a02c7c939b7bdc03b_18f1e190c5d53547fed41a3eaa76e9e9] Error 1
make[3]: Leaving directory '/home/lj/sdk/feeds/base/package/kernel/bpf-headers'
time: package/feeds/base/bpf-headers/compile#11.40#9.90#46.74
    ERROR: package/feeds/base/bpf-headers failed to build.
make[2]: *** [package/Makefile:120: package/feeds/base/bpf-headers/compile] Error 1
make[2]: Leaving directory '/home/lj/sdk'
make[1]: *** [package/Makefile:114: /home/lj/sdk/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lj/sdk'
make: *** [/home/lj/sdk/include/toplevel.mk:223：world] 错误 2

coolsnowwolf, 2024-12-02:
?

aiamadeus, 2024-12-02:
？

closed 2024-12-02 by coolsnowwolf