coolsnowwolf/lede #13148: checking whether mknod can create fifo without root privileges... configure: error: in `/home/lj/lede/build_dir/host/tar-1.34': configure: error: you should not run configure as root (set FORCE_UNSAFE_CONFIGURE=1 in environment to bypass this check)

lj267lj267, 2024-12-02:
### 详细叙述

checking whether mknod can create fifo without root privileges... configure: error: in `/home/lj/lede/build_dir/host/tar-1.34':
configure: error: you should not run configure as root (set FORCE_UNSAFE_CONFIGURE=1 in environment to bypass this check)
lede本项目加了small包，就报这个错误。是哪有问题呢？

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

wndr3800

### 详细日志

checking whether mknod can create fifo without root privileges... configure: error: in `/home/lj/lede/build_dir/host/tar-1.34':
configure: error: you should not run configure as root (set FORCE_UNSAFE_CONFIGURE=1 in environment to bypass this check)
See `config.log' for more details
make[3]: *** [Makefile:36: /home/lj/lede/build_dir/host/tar-1.34/.configured] Error 1
make[3]: Leaving directory '/home/lj/lede/tools/tar'
time: tools/tar/compile#15.29#9.06#27.57
    ERROR: tools/tar failed to build.
make[2]: *** [tools/Makefile:222: tools/tar/compile] Error 1
make[2]: Leaving directory '/home/lj/lede'
make[1]: *** [tools/Makefile:218: /home/lj/lede/staging_dir/host/stamp/.tools_compile_nyyynyynnyyynynyyyyyyynyyynnynyynnyynyyyyyyyyyyyyyyyyynynyyyyyyyyy] Error 2
make[1]: Leaving directory '/home/lj/lede'
make: *** [/home/lj/lede/include/toplevel.mk:231：world] 错误 2

yanxin152133, 2024-12-02:
[image]

lj267lj267, 2024-12-02:
> ![image](https://private-user-images.githubusercontent.com/43341537/391457419-fe5914e9-fc54-47b4-adb4-7cc83303a4d3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzMxMjgxMTIsIm5iZiI6MTczMzEyNzgxMiwicGF0aCI6Ii80MzM0MTUzNy8zOTE0NTc0MTktZmU1OTE0ZTktZmM1NC00N2I0LWFkYjQtN2NjODMzMDNhNGQzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjAyVDA4MjMzMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQ4ODNjNTdhNjVlY2FhYmVhMTExMjg5MGY4NzE0NGNhY2E5MWFkZDUyOGZiNDk0YjQ1NTU4ZTk4MzkwOTU0MTkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.UshxEmRqJSh39HzHOxfx-5-d9RUgJDln4COtf6J7u7Y)

用的是sudo 加编译命令运行不行吗？

lj267lj267, 2024-12-02:
不加sudo没有权限

lj267lj267, 2024-12-02:
export FORCE_UNSAFE_CONFIGURE=1 无效

yanxin152133, 2024-12-02:
https://www.cnblogs.com/songshuaiStudy/p/11647441.html
你看下你的文件的拥有者

closed 2024-12-02 by lj267lj267