coolsnowwolf/lede #13150: 编译报错

Kuron88, 2024-12-03:
### 详细叙述

编译报错

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

AX1800 PRO亚瑟

### 详细日志

Applying ./patches/020-locale.patch using plaintext: 
patching file gobject/glib-mkenums.in
touch /home/mi/lede/build_dir/target-aarch64_cortex-a53_musl/glib-2.74.0/.prepared_59a663a93872f3ec9a50ab802507f6cb_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/mi/lede/build_dir/target-aarch64_cortex-a53_musl/glib-2.74.0/.configured_*
rm -f /home/mi/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.glib2_installed
/home/mi/lede/staging_dir/host/bin/sed -e "s|@CC@|'/home/mi/lede/staging_dir/host/bin/gcc',|" -e "s|@CXX@|'/home/mi/lede/staging_dir/host/bin/g++',|" -e "s|@PKGCONFIG@|/home/mi/lede/staging_dir/host/bin/pkg-config|" -e "s|@CMAKE@|/home/mi/lede/staging_dir/host/bin/cmake|" -e "s|@PYTHON@|/home/mi/lede/staging_dir/host/bin/python3|" -e "s|@CFLAGS@|'-O2', '-I/home/mi/lede/staging_dir/host/include', '-I/home/mi/lede/staging_dir/hostpkg/include', '-I/home/mi/lede/staging_dir/target-aarch64_cortex-a53_musl/host/include', '-I/home/mi/lede/staging_dir/host/include', '-I/home/mi/lede/staging_dir/hostpkg/include', '-I/home/mi/lede/staging_dir/target-aarch64_cortex-a53_musl/host/include',|" -e "s|@CXXFLAGS@|'-O2', '-I/home/mi/lede/staging_dir/host/include', '-I/home/mi/lede/staging_dir/hostpkg/include', '-I/home/mi/lede/staging_dir/target-aarch64_cortex-a53_musl/host/include', '-I/home/mi/lede/staging_dir/host/include', '-I/home/mi/lede/staging_dir/hostpkg/include', '-I/home/mi/lede/staging_dir/target-aarch64_cortex-a53_musl/host/include',|" -e "s|@LDFLAGS@|'-L/home/mi/lede/staging_dir/host/lib', '-L/home/mi/lede/staging_dir/hostpkg/lib', '-L/home/mi/lede/staging_dir/target-aarch64_cortex-a53_musl/host/lib',|" -e "s|@PREFIX@|/home/mi/lede/staging_dir/hostpkg|" < /home/mi/lede/staging_dir/host/lib/meson/openwrt-native.txt.in > /home/mi/lede/build_dir/target-aarch64_cortex-a53_musl/glib-2.74.0/openwrt-native.txt
/home/mi/lede/staging_dir/host/bin/sed -e "s|@CC@|'aarch64-openwrt-linux-musl-gcc',|" -e "s|@CXX@|'aarch64-openwrt-linux-musl-g++',|" -e "s|@AR@|aarch64-openwrt-linux-musl-gcc-ar|" -e "s|@STRIP@|aarch64-openwrt-linux-musl-strip|" -e "s|@NM@|aarch64-openwrt-linux-musl-gcc-nm|" -e "s|@PKGCONFIG@|/home/mi/lede/staging_dir/host/bin/pkg-config|" -e "s|@CMAKE@|/home/mi/lede/staging_dir/host/bin/cmake|" -e "s|@PYTHON@|/home/mi/lede/staging_dir/host/bin/python3|" -e "s|@CFLAGS@|'-Os', '-pipe', '-mcpu=cortex-a53', '-fno-caller-saves', '-fno-plt', '-fhonour-copts', '-fmacro-prefix-map=/home/mi/lede/build_dir/target-aarch64_cortex-a53_musl/glib-2.74.0=glib-2.74.0', '-ffunction-sections', '-fdata-sections', '-Wformat', '-Werror=format-security', '-fstack-protector', '-Wl,-z,now', '-Wl,-z,relro', '-I/home/mi/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/usr/include', '-I/home/mi/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/include', '-I/home/mi/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/include/fortify',|" -e "s|@CXXFLAGS@|'-Os', '-pipe', '-mcpu=cortex-a53', '-fno-caller-saves', '-fno-plt', '-fhonour-copts', '-fmacro-prefix-map=/home/mi/lede/build_dir/target-aarch64_cortex-a53_musl/glib-2.74.0=glib-2.74.0', '-ffunction-sections', '-fdata-sections', '-Wformat', '-Werror=format-security', '-fstack-protector', '-Wl,-z,now', '-Wl,-z,relro', '-ffunction-sections', '-fdata-sections', '-I/home/mi/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/usr/include', '-I/home/mi/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/include', '-I/home/mi/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/include/fortify',|" -e "s|@LDFLAGS@|'-L/home/mi/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/usr/lib', '-L/home/mi/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/lib', '-fuse-ld=bfd', '-Wl,--gc-sections', '-znow', '-zrelro',|" -e "s|@ARCH@|"aarch64"|" -e "s|@CPU@|"generic"|" -e "s|@ENDIAN@|little|" < /home/mi/lede/staging_dir/host/lib/meson/openwrt-cross.txt.in > /home/mi/lede/build_dir/target-aarch64_cortex-a53_musl/glib-2.74.0/openwrt-cross.txt
/home/mi/lede/staging_dir/host/bin/python3 /home/mi/lede/staging_dir/host/bin/meson.py  --buildtype plain --native-file /home/mi/lede/build_dir/target-aarch64_cortex-a53_musl/glib-2.74.0/openwrt-native.txt --cross-file /home/mi/lede/build_dir/target-aarch64_cortex-a53_musl/glib-2.74.0/openwrt-cross.txt -Dselinux=disabled -Dlibmount=disabled -Dman=false -Ddtrace=false -Dsystemtap=false -Dsysprof=disabled -Dgtk_doc=false -Dbsymbolic_functions=true -Dforce_posix_threads=true -Dtests=false -Dinstalled_tests=false -Doss_fuzz=disabled -Dglib_debug=disabled -Dglib_assert=false -Dglib_checks=true -Dlibelf=disabled -Dxattr=true -Db_lto=true -Ddefault_library=both -Dnls=disabled /home/mi/lede/build_dir/target-aarch64_cortex-a53_musl/glib-2.74.0/openwrt-build /home/mi/lede/build_dir/target-aarch64_cortex-a53_musl/glib-2.74.0/openwrt-build/..

ERROR: Malformed value in machine file variable 'c_ld': lexer
[@LD@
^.
WARNING: Running the setup command as `meson [options]` instead of `meson setup [options]` is ambiguous and deprecated.
make[3]: *** [Makefile:125: /home/mi/lede/build_dir/target-aarch64_cortex-a53_musl/glib-2.74.0/.configured_efb2d3cf01c707184fb77fd0fe149f94] Error 1
make[3]: Leaving directory '/home/mi/lede/feeds/packages/libs/glib2'
time: package/feeds/packages/glib2/compile#0.64#0.60#1.13
    ERROR: package/feeds/packages/glib2 failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/glib2/compile] Error 1
make[2]: Leaving directory '/home/mi/lede'
make[1]: *** [package/Makefile:110: /home/mi/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/mi/lede'
make: *** [/home/mi/lede/include/toplevel.mk:231：world] 错误 2

Brucedyz, 2024-12-03:
X86 编译报错
make[1]: *** [tools/Makefile:227: /home/***/lede/staging_dir/host/stamp/.tools_compile_nyyynyynnnyyynynyyyyyyynyyynynyynnynnyyyyyynyyyyyyyyyyynynnyyyyynyyy] Error 2
make[1]: Leaving directory '/home/bruce/lede'
make: *** [/home/***/lede/include/toplevel.mk:231: world] Error 2

coolsnowwolf, 2024-12-03:
rm -rf staging_dir build_dir

sxml, 2024-12-03:
> 记得 git pull 下即可解决

云编译 也报错glib2 失败

云编译会有旧文件吗diy-part2.sh脚本加了rm -rf staging_dir build_dir 

加了报其他

ERROR: unknown option aarch64-openwrt-linux-musl-gcc
Try './configure --help' for more information
make[3]: *** [Makefile:62: /workdir/openwrt/build_dir/target-aarch64_generic_musl/liburing-2.3/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/libs/liburing'
time: package/feeds/packages/liburing/compile#0.15#0.05#0.18
    ERROR: package/feeds/packages/liburing failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/liburing/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2