coolsnowwolf/lede #13151: 编译报错

xianfanjie, 2024-12-03:
### 详细叙述

编译过程中报错，求大佬给看看。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

qihoo 360V6

### 详细日志

make[3]: Leaving directory '/workdir/openwrt/feeds/packages/net/nginx-util'
time: package/feeds/packages/nginx-util/compile#0.17#0.06#0.21
make[3]: Entering directory '/workdir/openwrt/feeds/packages/libs/expat'
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/libs/expat'
time: package/feeds/packages/expat/compile#0.13#0.06#0.17
make[3]: Entering directory '/workdir/openwrt/feeds/packages/utils/dbus'
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/utils/dbus'
time: package/feeds/packages/dbus/compile#0.13#0.07#0.17
make[3]: Entering directory '/workdir/openwrt/feeds/packages/libs/libffi'
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/libs/libffi'
time: package/feeds/packages/libffi/compile#0.14#0.09#0.20
make[3]: Entering directory '/workdir/openwrt/feeds/packages/libs/pcre2'
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/libs/pcre2'
time: package/feeds/packages/pcre2/compile#0.15#0.08#0.20
make[3]: Entering directory '/workdir/openwrt/feeds/packages/libs/glib2'
rm -f /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/glib-2.74.0/.configured_*
rm -f /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.glib2_installed
/workdir/openwrt/staging_dir/host/bin/sed -e "s|@CC@|'/workdir/openwrt/staging_dir/host/bin/gcc',|" -e "s|@CXX@|'/workdir/openwrt/staging_dir/host/bin/g++',|" -e "s|@PKGCONFIG@|/workdir/openwrt/staging_dir/host/bin/pkg-config|" -e "s|@CMAKE@|/workdir/openwrt/staging_dir/host/bin/cmake|" -e "s|@PYTHON@|/workdir/openwrt/staging_dir/host/bin/python3|" -e "s|@CFLAGS@|'-O2', '-I/workdir/openwrt/staging_dir/host/include', '-I/workdir/openwrt/staging_dir/hostpkg/include', '-I/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/host/include', '-I/workdir/openwrt/staging_dir/host/include', '-I/workdir/openwrt/staging_dir/hostpkg/include', '-I/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/host/include',|" -e "s|@CXXFLAGS@|'-O2', '-I/workdir/openwrt/staging_dir/host/include', '-I/workdir/openwrt/staging_dir/hostpkg/include', '-I/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/host/include', '-I/workdir/openwrt/staging_dir/host/include', '-I/workdir/openwrt/staging_dir/hostpkg/include', '-I/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/host/include',|" -e "s|@LDFLAGS@|'-L/workdir/openwrt/staging_dir/host/lib', '-L/workdir/openwrt/staging_dir/hostpkg/lib', '-L/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/host/lib',|" -e "s|@PREFIX@|/workdir/openwrt/staging_dir/hostpkg|" < /workdir/openwrt/staging_dir/host/lib/meson/openwrt-native.txt.in > /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/glib-2.74.0/openwrt-native.txt
/workdir/openwrt/staging_dir/host/bin/sed -e "s|@CC@|'aarch64-openwrt-linux-musl-gcc',|" -e "s|@CXX@|'aarch64-openwrt-linux-musl-g++',|" -e "s|@AR@|aarch64-openwrt-linux-musl-gcc-ar|" -e "s|@STRIP@|aarch64-openwrt-linux-musl-strip|" -e "s|@NM@|aarch64-openwrt-linux-musl-gcc-nm|" -e "s|@PKGCONFIG@|/workdir/openwrt/staging_dir/host/bin/pkg-config|" -e "s|@CMAKE@|/workdir/openwrt/staging_dir/host/bin/cmake|" -e "s|@PYTHON@|/workdir/openwrt/staging_dir/host/bin/python3|" -e "s|@CFLAGS@|'-Os', '-pipe', '-mcpu=cortex-a53', '-fno-caller-saves', '-fno-plt', '-fhonour-copts', '-fmacro-prefix-map=/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/glib-2.74.0=glib-2.74.0', '-ffunction-sections', '-fdata-sections', '-Wformat', '-Werror=format-security', '-fstack-protector', '-Wl,-z,now', '-Wl,-z,relro', '-I/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/usr/include', '-I/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/include', '-I/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/include/fortify',|" -e "s|@CXXFLAGS@|'-Os', '-pipe', '-mcpu=cortex-a53', '-fno-caller-saves', '-fno-plt', '-fhonour-copts', '-fmacro-prefix-map=/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/glib-2.74.0=glib-2.74.0', '-ffunction-sections', '-fdata-sections', '-Wformat', '-Werror=format-security', '-fstack-protector', '-Wl,-z,now', '-Wl,-z,relro', '-ffunction-sections', '-fdata-sections', '-I/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/usr/include', '-I/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/include', '-I/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/include/fortify',|" -e "s|@LDFLAGS@|'-L/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/usr/lib', '-L/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/lib', '-fuse-ld=bfd', '-Wl,--gc-sections', '-znow', '-zrelro',|" -e "s|@ARCH@|"aarch64"|" -e "s|@CPU@|"generic"|" -e "s|@ENDIAN@|little|" < /workdir/openwrt/staging_dir/host/lib/meson/openwrt-cross.txt.in > /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/glib-2.74.0/openwrt-cross.txt
/workdir/openwrt/staging_dir/host/bin/python3 /workdir/openwrt/staging_dir/host/bin/meson.py  --buildtype plain --native-file /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/glib-2.74.0/openwrt-native.txt --cross-file /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/glib-2.74.0/openwrt-cross.txt -Dselinux=disabled -Dlibmount=disabled -Dman=false -Ddtrace=false -Dsystemtap=false -Dsysprof=disabled -Dgtk_doc=false -Dbsymbolic_functions=true -Dforce_posix_threads=true -Dtests=false -Dinstalled_tests=false -Doss_fuzz=disabled -Dglib_debug=disabled -Dglib_assert=false -Dglib_checks=true -Dlibelf=disabled -Dxattr=true -Db_lto=true -Ddefault_library=both -Dnls=disabled /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/glib-2.74.0/openwrt-build /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/glib-2.74.0/openwrt-build/..

ERROR: Malformed value in machine file variable 'c_ld': lexer
[@LD@
^.
WARNING: Running the setup command as `meson [options]` instead of `meson setup [options]` is ambiguous and deprecated.
make[3]: *** [Makefile:125: /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/glib-2.74.0/.configured_efb2d3cf01c707184fb77fd0fe149f94] Error 1
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/libs/glib2'
time: package/feeds/packages/glib2/compile#0.73#0.11#0.81
    ERROR: package/feeds/packages/glib2 failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/glib2/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2

coolsnowwolf, 2024-12-03:
rm -f  staging_dir build_dir

xianfanjie, 2024-12-03:
> rm -rf staging_dir 构建目录
感谢大佬，正常了。

closed 2024-12-03 by xianfanjie