coolsnowwolf/lede #13152: 重新fork后，还是编译报错。

LIUBANGJIAN, 2024-12-03:
### 详细叙述

重新fork后，还是编译报错。这次是新的报错。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

make  install-am
make[6]: Entering directory '/home/runner/work/lede/lede/build_dir/target-x86_64_musl/gnutls-3.8.3/src'
make[7]: Entering directory '/home/runner/work/lede/lede/build_dir/target-x86_64_musl/gnutls-3.8.3/src'
 /usr/bin/mkdir -p '/home/runner/work/lede/lede/build_dir/target-x86_64_musl/gnutls-3.8.3/ipkg-install/usr/bin'
  /bin/bash ../libtool   --mode=install /home/runner/work/lede/lede/staging_dir/host/bin/install -c psktool gnutls-cli-debug certtool gnutls-serv gnutls-cli ocsptool '/home/runner/work/lede/lede/build_dir/target-x86_64_musl/gnutls-3.8.3/ipkg-install/usr/bin'
libtool: warning: '../lib/libgnutls.la' has not been installed in '/usr/lib'
OpenWrt-libtool: install: /home/runner/work/lede/lede/staging_dir/host/bin/install -c .libs/psktool /home/runner/work/lede/lede/build_dir/target-x86_64_musl/gnutls-3.8.3/ipkg-install/usr/bin/psktool
libtool: warning: '../lib/libgnutls.la' has not been installed in '/usr/lib'
OpenWrt-libtool: install: /home/runner/work/lede/lede/staging_dir/host/bin/install -c .libs/gnutls-cli-debug /home/runner/work/lede/lede/build_dir/target-x86_64_musl/gnutls-3.8.3/ipkg-install/usr/bin/gnutls-cli-debug
libtool: warning: '../lib/libgnutls.la' has not been installed in '/usr/lib'
libtool: warning: '/home/runner/work/lede/lede/build_dir/target-x86_64_musl/gnutls-3.8.3/lib/libgnutls.la' has not been installed in '/usr/lib'
OpenWrt-libtool: install: /home/runner/work/lede/lede/staging_dir/host/bin/install -c .libs/certtool /home/runner/work/lede/lede/build_dir/target-x86_64_musl/gnutls-3.8.3/ipkg-install/usr/bin/certtool
libtool: warning: '../lib/libgnutls.la' has not been installed in '/usr/lib'
OpenWrt-libtool: install: /home/runner/work/lede/lede/staging_dir/host/bin/install -c .libs/gnutls-serv /home/runner/work/lede/lede/build_dir/target-x86_64_musl/gnutls-3.8.3/ipkg-install/usr/bin/gnutls-serv
libtool: warning: '../lib/libgnutls.la' has not been installed in '/usr/lib'
OpenWrt-libtool: install: /home/runner/work/lede/lede/staging_dir/host/bin/install -c .libs/gnutls-cli /home/runner/work/lede/lede/build_dir/target-x86_64_musl/gnutls-3.8.3/ipkg-install/usr/bin/gnutls-cli
libtool: warning: '../lib/libgnutls.la' has not been installed in '/usr/lib'
OpenWrt-libtool: install: /home/runner/work/lede/lede/staging_dir/host/bin/install -c .libs/ocsptool /home/runner/work/lede/lede/build_dir/target-x86_64_musl/gnutls-3.8.3/ipkg-install/usr/bin/ocsptool
make[7]: Nothing to be done for 'install-data-am'.
make[7]: Leaving directory '/home/runner/work/lede/lede/build_dir/target-x86_64_musl/gnutls-3.8.3/src'
make[6]: Leaving directory '/home/runner/work/lede/lede/build_dir/target-x86_64_musl/gnutls-3.8.3/src'
make[5]: Leaving directory '/home/runner/work/lede/lede/build_dir/target-x86_64_musl/gnutls-3.8.3/src'
make[5]: Entering directory '/home/runner/work/lede/lede/build_dir/target-x86_64_musl/gnutls-3.8.3'
make[6]: Entering directory '/home/runner/work/lede/lede/build_dir/target-x86_64_musl/gnutls-3.8.3'
make[6]: Nothing to be done for 'install-exec-am'.
make[6]: Nothing to be done for 'install-data-am'.
make[6]: Leaving directory '/home/runner/work/lede/lede/build_dir/target-x86_64_musl/gnutls-3.8.3'
make[5]: Leaving directory '/home/runner/work/lede/lede/build_dir/target-x86_64_musl/gnutls-3.8.3'
make[4]: Leaving directory '/home/runner/work/lede/lede/build_dir/target-x86_64_musl/gnutls-3.8.3'
touch /home/runner/work/lede/lede/build_dir/target-x86_64_musl/gnutls-3.8.3/.built
rm -rf /home/runner/work/lede/lede/tmp/stage-gnutls
mkdir -p /home/runner/work/lede/lede/tmp/stage-gnutls/host /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/packages
install -d -m0755 /home/runner/work/lede/lede/tmp/stage-gnutls/usr/include /home/runner/work/lede/lede/tmp/stage-gnutls/usr/lib/pkgconfig
cp -fpR /home/runner/work/lede/lede/build_dir/target-x86_64_musl/gnutls-3.8.3/ipkg-install/usr/lib/*.so* /home/runner/work/lede/lede/tmp/stage-gnutls/usr/lib/
cp -fpR /home/runner/work/lede/lede/build_dir/target-x86_64_musl/gnutls-3.8.3/ipkg-install/usr/include/gnutls /home/runner/work/lede/lede/tmp/stage-gnutls/usr/include/
cp -fpR /home/runner/work/lede/lede/build_dir/target-x86_64_musl/gnutls-3.8.3/ipkg-install/usr/lib/pkgconfig/*.pc /home/runner/work/lede/lede/tmp/stage-gnutls/usr/lib/pkgconfig/
find /home/runner/work/lede/lede/tmp/stage-gnutls -name '*.la' | xargs -r rm -f; 
if [ -f /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/packages/gnutls.list ]; then /home/runner/work/lede/lede/scripts/clean-package.sh "/home/runner/work/lede/lede/staging_dir/target-x86_64_musl/packages/gnutls.list" "/home/runner/work/lede/lede/staging_dir/target-x86_64_musl"; fi
if [ -d /home/runner/work/lede/lede/tmp/stage-gnutls ]; then (cd /home/runner/work/lede/lede/tmp/stage-gnutls; find ./ > /home/runner/work/lede/lede/tmp/stage-gnutls.files); 	SHELL= flock /home/runner/work/lede/lede/tmp/.staging-dir.flock -c ' mv /home/runner/work/lede/lede/tmp/stage-gnutls.files /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/packages/gnutls.list && cp -fpR /home/runner/work/lede/lede/tmp/stage-gnutls/* /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/; '; fi
rm -rf /home/runner/work/lede/lede/tmp/stage-gnutls
touch /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/stamp/.gnutls_installed
mkdir -p /home/runner/work/lede/lede/bin/targets/x86/64/packages /home/runner/work/lede/lede/build_dir/target-x86_64_musl/gnutls-3.8.3/ipkg-x86_64/libgnutls/CONTROL /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/pkginfo
install -d -m0755 /home/runner/work/lede/lede/build_dir/target-x86_64_musl/gnutls-3.8.3/ipkg-x86_64/libgnutls/usr/lib
cp -fpR /home/runner/work/lede/lede/build_dir/target-x86_64_musl/gnutls-3.8.3/ipkg-install/usr/lib/libgnutls.so.* /home/runner/work/lede/lede/build_dir/target-x86_64_musl/gnutls-3.8.3/ipkg-x86_64/libgnutls/usr/lib/
find /home/runner/work/lede/lede/build_dir/target-x86_64_musl/gnutls-3.8.3/ipkg-x86_64/libgnutls -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
export CROSS="x86_64-openwrt-linux-musl-"  NO_RENAME=1 ; NM="x86_64-openwrt-linux-musl-nm" STRIP="/home/runner/work/lede/lede/staging_dir/host/bin/sstrip -z" STRIP_KMOD="/home/runner/work/lede/lede/scripts/strip-kmod.sh" PATCHELF="/home/runner/work/lede/lede/staging_dir/host/bin/patchelf" /home/runner/work/lede/lede/scripts/rstrip.sh /home/runner/work/lede/lede/build_dir/target-x86_64_musl/gnutls-3.8.3/ipkg-x86_64/libgnutls
rstrip.sh: /home/runner/work/lede/lede/build_dir/target-x86_64_musl/gnutls-3.8.3/ipkg-x86_64/libgnutls/usr/lib/libgnutls.so.30.37.1: shared object
(cd /home/runner/work/lede/lede/build_dir/target-x86_64_musl/gnutls-3.8.3/ipkg-x86_64/libgnutls/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]*, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_postinst \$0 \$@"; ) > postinst; ( echo "#!/bin/sh"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_prerm \$0 \$@"; ) > prerm; chmod 0755 postinst prerm;  )
install -d -m0755 /home/runner/work/lede/lede/bin/packages/x86_64/packages
/home/runner/work/lede/lede/staging_dir/host/bin/fakeroot /home/runner/work/lede/lede/staging_dir/host/bin/bash /home/runner/work/lede/lede/scripts/ipkg-build -m "" /home/runner/work/lede/lede/build_dir/target-x86_64_musl/gnutls-3.8.3/ipkg-x86_64/libgnutls /home/runner/work/lede/lede/bin/packages/x86_64/packages
Packaged contents of /home/runner/work/lede/lede/build_dir/target-x86_64_musl/gnutls-3.8.3/ipkg-x86_64/libgnutls into /home/runner/work/lede/lede/bin/packages/x86_64/packages/libgnutls_3.8.3-1_x86_64.ipk
rm -rf /home/runner/work/lede/lede/build_dir/target-x86_64_musl/gnutls-3.8.3/.pkgdir/libgnutls.installed /home/runner/work/lede/lede/build_dir/target-x86_64_musl/gnutls-3.8.3/.pkgdir/libgnutls
mkdir -p /home/runner/work/lede/lede/build_dir/target-x86_64_musl/gnutls-3.8.3/.pkgdir/libgnutls
install -d -m0755 /home/runner/work/lede/lede/build_dir/target-x86_64_musl/gnutls-3.8.3/.pkgdir/libgnutls/usr/lib
[... cut ...]
make[3]: Leaving directory '/home/runner/work/lede/lede/feeds/packages/libs/gnutls'
time: package/feeds/packages/gnutls/compile#252.26#30.61#160.65
make[2]: Leaving directory '/home/runner/work/lede/lede'
make[1]: *** [package/Makefile:110: /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/runner/work/lede/lede'
make: *** [/home/runner/work/lede/lede/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

LIUBANGJIAN, 2024-12-03:
还是用L大自己的openwrt-ci.yml文件

coolsnowwolf, 2024-12-03:
[image]