coolsnowwolf/lede #13153: DockerMan与ShadowSocksR Plus+不兼容

seassrs, 2024-12-03:
### 详细叙述

ShadowSocksR Plus+配置正常运行后，重启路由器DockerMan所有配置变空白，把ShadowSocksR Plus+停止，再重启路由器DockerMan配置又正常了。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86/64

### 详细日志

同上

coolsnowwolf, 2024-12-03:
没发现

seassrs, 2024-12-03:
用官方openwrt源码编译问题依旧，可以确定不是您源码问题。更换passwall测试可以正常，问题应该出在ShadowSocksR Plus+，但一直在用ShadowSocksR Plus+，实在不习惯passwall，只好暂退回原版本使用等更新。感谢L大百忙之中回复！

lizheyo77, 2024-12-12:
存在这个问题

chenliangxin, 2024-12-13:
换了6.1内核编译，依然无法拉去docker镜像。

coolsnowwolf, 2026-09-05:
已结合 #13204 和当前代码复核。旧版 DockerMan 在生成菜单时会同步探测 Docker socket/API；SSR+ 启动后只要 dockerd 因 `/opt` 挂载、DNS、iptables 或启动时序暂时不可用，旧 controller 就会返回并表现为“配置变空白”。现行 DockerMan 已重写为 JS + ucode，菜单不再依赖启动时 `_ping()`，该界面症状已经解决；当前 x86_64 的 dockerd/containerd/docker-compose/DockerMan 也已完成实际编译。

报告者已经确认官方 OpenWrt 同样发生并判断问题在 SSR+，本 issue 又未提供 SSR+ 配置、dockerd 日志、挂载时序或可复现步骤，无法归因并修复 LEDE。镜像拉取失败也与“页面空白”不是同一故障。按非 LEDE 缺陷关闭；若当前版本仍有代理兼容问题，请向 SSR+ 仓库提供 `logread -e dockerd -e shadowsocksr`、`mount`、`nft list ruleset` 及 `/etc/config/{dockerd,shadowsocksr}`（脱敏）另报。

closed 2026-09-05 by coolsnowwolf