coolsnowwolf/lede #13155: 最新代码编译,无法重启

xianfanjie, 2024-12-03:
### 详细叙述

拉取的最新代码
设备：qihoo_360v6
问题:：刷入固件后,无法自动重启,只要重启就会关机，最后只能拔电重启。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

qihoo_360v6

### 详细日志

系统日志：
[系统日志.txt](https://github.com/user-attachments/files/17990710/default.txt)
内核日志：
[内核日志.txt](https://github.com/user-attachments/files/17990715/default.txt)

coolsnowwolf, 2024-12-03:
看不到原因

xianfanjie, 2024-12-04:
> 看不到原因

大佬看看这个
ttl的日志 拔电重新启动到点击重启后的日志
[ttl日志.txt](https://github.com/user-attachments/files/18006125/ttl.txt)

D -         0 - sbl1_ddr_set_params, Delta 到这就没日志了， 电源灯直接就不亮，必须拔电重启。

xianfanjie, 2024-12-08:
> 换uboot

换谁的

xianfanjie, 2025-01-28:
> 现在好了吗？

现在没编译这个仓库的代码了，换其他的在用了。

xianfanjie, 2025-01-30:
> > > 现在好了吗？
> > 
> > 
> > 现在还没编译这个仓库的代码了，换其他的在用了。
> 
> 请教一下，用的那个仓库，方便告知吗？用着稳定吗，lede固件不影响使用，重启拔电这点闹心！

immortalwrt或者openwrt官方

closed 2025-02-13 by xianfanjie

nb369, 2025-03-18:
> > > > > 现在好了吗？
> > > > 
> > > > 
> > > > 现在还没编译这个仓库的代码了，换其他的在用了。
> > > 
> > > 
> > > 请教一下，用的那个仓库，方便告知吗？用着稳定吗，lede固件不影响使用，重启拔电这点闹心！
> > 
> > 
> > immortalwrt或者openwrt官方
> 
> 好消息！用6.6内核已经可以正常重启了！

你好，请问一下，你编的6.6内核无线正常吗？我360v6编译完没有无线，wifi驱动都勾选了！