coolsnowwolf/lede #13156: 缺少luci-app-ksmbd

gssiwmd, 2024-12-03:
### 详细叙述

编译错误

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

  r5s

### 详细日志

Already up to date.
Create index file './feeds/packages.index' 
Create index file './feeds/luci.index' 
Create index file './feeds/routing.index' 
Create index file './feeds/telephony.index' 
Create index file './feeds/gssiwmdpackage.index' 
Create index file './feeds/privatepackage.index' 
WARNING: Makefile 'package/lean/autosamba/Makefile' has a dependency on 'luci-app-ksmbd', which does not exist
Installing all packages from feed packages.
Installing all packages from feed luci.
Installing all packages from feed routing.
Installing all packages from feed telephony.

bi4tms, 2024-12-03:
同样的问题
@coolsnowwolf

coolsnowwolf, 2024-12-03:
改名了

closed 2024-12-03 by gssiwmd