coolsnowwolf/lede #13158: 拉取最新源码编译失败

bi4tms, 2024-12-03:
### 详细叙述

拉取最新源码，编译失败，提示ninja无法编译
@coolsnowwolf 

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

X86/R2S/R4S/R66S

### 详细日志

Found ninja-1.12.1 at /home/r2s.6.6/lede/staging_dir/host/bin/ninja
touch /home/r2s.6.6/lede/build_dir/target-aarch64_generic_musl/glib-2.82.0/.configured_efb2d3cf01c707184fb77fd0fe149f94
rm -f /home/r2s.6.6/lede/build_dir/target-aarch64_generic_musl/glib-2.82.0/.built
touch /home/r2s.6.6/lede/build_dir/target-aarch64_generic_musl/glib-2.82.0/.built_check
MAKEFLAGS="" /home/r2s.6.6/lede/staging_dir/host/bin/ninja  -j1 -C /home/r2s.6.6/lede/build_dir/target-aarch64_generic_musl/glib-2.82.0/openwrt-build 
ninja: Entering directory `/home/r2s.6.6/lede/build_dir/target-aarch64_generic_musl/glib-2.82.0/openwrt-build'
[474/578] Compiling C object gio/gresource.p/gresource-tool.c.o
FAILED: gio/gresource.p/gresource-tool.c.o 
aarch64-openwrt-linux-musl-gcc -Igio/gresource.p -Igio -I../gio -I. -I.. -Iglib -I../glib -Igmodule -I../gmodule -Igobject -I../gobject -I/home/r2s.6.6/lede/staging_dir/target-aarch64_generic_musl/usr/include -I/home/r2s.6.6/lede/staging_dir/toolchain-aarch64_generic_gcc-13.3.0_musl/usr/include -I/home/r2s.6.6/lede/staging_dir/toolchain-aarch64_generic_gcc-13.3.0_musl/include -I/home/r2s.6.6/lede/staging_dir/toolchain-aarch64_generic_gcc-13.3.0_musl/include/fortify -flto=auto -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -Wpedantic -std=gnu99 -D_GNU_SOURCE -fno-strict-aliasing -DG_DISABLE_CAST_CHECKS -DG_DISABLE_ASSERT -Wduplicated-branches -Wimplicit-fallthrough -Wmisleading-indentation -Wmissing-field-initializers -Wnonnull -Wnull-dereference -Wunused -Wno-unused-parameter -Wno-cast-function-type -Wno-pedantic -Wno-format-zero-length -Wno-variadic-macros -Werror=format=2 -Werror=init-self -Werror=missing-include-dirs -Werror=pointer-arith -Werror=unused-result -Wstrict-prototypes -Wno-bad-function-cast -Werror=implicit-function-declaration -Werror=missing-prototypes -Werror=pointer-sign -Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/r2s.6.6/lede/build_dir/target-aarch64_generic_musl/glib-2.82.0=glib-2.82.0 -ffunction-sections -fdata-sections -Wformat -Werror=format-security -fstack-protector -Wl,-z,now -Wl,-z,relro -MD -MQ gio/gresource.p/gresource-tool.c.o -MF gio/gresource.p/gresource-tool.c.o.d -o gio/gresource.p/gresource-tool.c.o -c ../gio/gresource-tool.c
In file included from ../gio/gresource-tool.c:35:
/home/r2s.6.6/lede/staging_dir/target-aarch64_generic_musl/usr/include/gelf.h:86:9: error: unknown type name 'Elf64_Relr'
   86 | typedef Elf64_Relr GElf_Relr;
      |         ^~~~~~~~~~
ninja: build stopped: subcommand failed.
make[3]: *** [Makefile:133: /home/r2s.6.6/lede/build_dir/target-aarch64_generic_musl/glib-2.82.0/.built] Error 1
make[3]: Leaving directory '/home/r2s.6.6/lede/feeds/packages/libs/glib2'
time: package/feeds/packages/glib2/compile#240.96#21.59#181.16
    ERROR: package/feeds/packages/glib2 failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/glib2/compile] Error 1
make[2]: Leaving directory '/home/r2s.6.6/lede'
make[1]: *** [package/Makefile:110: /home/r2s.6.6/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/r2s.6.6/lede'
make: *** [/home/r2s.6.6/lede/include/toplevel.mk:231：world] 错误 2

binge8, 2024-12-03:
[多搜索一下，明明有同样的问题而且给了解决方案，还非得再开一个，就差给你喂嘴里了](https://github.com/coolsnowwolf/lede/issues/13151)

coolsnowwolf, 2024-12-03:
工具链同步了官方 v24.10 ，需要清除之前的缓存

closed 2024-12-04 by coolsnowwolf