coolsnowwolf/lede #13159: zmq报错

binge8, 2024-12-03:
### 详细叙述

rm -rf 应该解决不了，因为全新坏境编译

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x64，armv8

### 详细日志

```bash
2024-12-03T10:34:40.1834881Z make[3]: Entering directory '/mnt/openwrt/feeds/packages/libs/zmq'
2024-12-03T10:34:40.3431238Z rm -f /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/zeromq-nc/zeromq-4.3.4/.built
2024-12-03T10:34:40.3460778Z touch /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/zeromq-nc/zeromq-4.3.4/.built_check
2024-12-03T10:34:40.3486536Z MAKEFLAGS="" /home/runner/work/op/op/openwrt/staging_dir/host/bin/ninja  -j1 -C /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/zeromq-nc/zeromq-4.3.4/openwrt-build 
2024-12-03T10:34:40.3517271Z ninja: Entering directory `/home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/zeromq-nc/zeromq-4.3.4/openwrt-build'
2024-12-03T10:34:40.7521218Z [1/110] Building CXX object CMakeFiles/objects.dir/src/curve_client.cpp.o
2024-12-03T10:34:40.7522132Z FAILED: CMakeFiles/objects.dir/src/curve_client.cpp.o 
2024-12-03T10:34:40.7528356Z /home/runner/work/op/op/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/bin/x86_64-openwrt-linux-musl-g++ -DZMQ_CUSTOM_PLATFORM_HPP -D_GNU_SOURCE -D_REENTRANT -D_THREAD_SAFE -I/home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/zeromq-nc/zeromq-4.3.4/include -I/home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/zeromq-nc/zeromq-4.3.4/openwrt-build -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/zeromq-nc/zeromq-4.3.4=zeromq-4.3.4 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -std=gnu++11 -Wno-tautological-constant-compare -DNDEBUG -fPIC -MD -MT CMakeFiles/objects.dir/src/curve_client.cpp.o -MF CMakeFiles/objects.dir/src/curve_client.cpp.o.d -o CMakeFiles/objects.dir/src/curve_client.cpp.o -c /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/zeromq-nc/zeromq-4.3.4/src/curve_client.cpp
2024-12-03T10:34:40.7534182Z make[3]: *** [Makefile:128: /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/zeromq-nc/zeromq-4.3.4/.built] Error 1
2024-12-03T10:34:40.7535825Z In file included from /home/runner/work/op/op/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/x86_64-openwrt-linux-musl/include/c++/13.3.0/ext/alloc_traits.h:34,
2024-12-03T10:34:40.7537644Z                  from /home/runner/work/op/op/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/x86_64-openwrt-linux-musl/include/c++/13.3.0/bits/stl_tree.h:67,
2024-12-03T10:34:40.7538767Z                  from /home/runner/work/op/op/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/x86_64-openwrt-linux-musl/include/c++/13.3.0/map:62,
2024-12-03T10:34:40.7539663Z                  from /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/zeromq-nc/zeromq-4.3.4/src/metadata.hpp:33,
2024-12-03T10:34:40.7540445Z                  from /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/zeromq-nc/zeromq-4.3.4/src/msg.hpp:40,
2024-12-03T10:34:40.7541453Z                  from /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/zeromq-nc/zeromq-4.3.4/src/curve_client.cpp:35:
2024-12-03T10:34:40.7542810Z /home/runner/work/op/op/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/x86_64-openwrt-linux-musl/include/c++/13.3.0/bits/alloc_traits.h: In instantiation of 'struct std::__allocator_traits_base::__rebind<zmq::secure_allocator_t<unsigned char>, unsigned char, void>':
2024-12-03T10:34:40.7545188Z /home/runner/work/op/op/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/x86_64-openwrt-linux-musl/include/c++/13.3.0/bits/alloc_traits.h:94:11:   required by substitution of 'template<class _Alloc, class _Up> using std::__alloc_rebind = typename std::__allocator_traits_base::__rebind<_Alloc, _Up>::type [with _Alloc = zmq::secure_allocator_t<unsigned char>; _Up = unsigned char]'
2024-12-03T10:34:40.7548539Z /home/runner/work/op/op/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/x86_64-openwrt-linux-musl/include/c++/13.3.0/bits/alloc_traits.h:228:8:   required by substitution of 'template<class _Alloc> template<class _Tp> using std::allocator_traits< <template-parameter-1-1> >::rebind_alloc = std::__alloc_rebind<_Alloc, _Tp> [with _Tp = unsigned char; _Alloc = zmq::secure_allocator_t<unsigned char>]'
2024-12-03T10:34:40.7551547Z /home/runner/work/op/op/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/x86_64-openwrt-linux-musl/include/c++/13.3.0/ext/alloc_traits.h:126:65:   required from 'struct __gnu_cxx::__alloc_traits<zmq::secure_allocator_t<unsigned char>, unsigned char>::rebind<unsigned char>'
2024-12-03T10:34:40.7554108Z /home/runner/work/op/op/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/x86_64-openwrt-linux-musl/include/c++/13.3.0/bits/stl_vector.h:88:21:   required from 'struct std::_Vector_base<unsigned char, zmq::secure_allocator_t<unsigned char> >'
2024-12-03T10:34:40.7556600Z /home/runner/work/op/op/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/x86_64-openwrt-linux-musl/include/c++/13.3.0/bits/stl_vector.h:428:11:   required from 'class std::vector<unsigned char, zmq::secure_allocator_t<unsigned char> >'
2024-12-03T10:34:40.7558297Z /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/zeromq-nc/zeromq-4.3.4/src/curve_client_tools.hpp:64:76:   required from here
2024-12-03T10:34:40.7559990Z /home/runner/work/op/op/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/x86_64-openwrt-linux-musl/include/c++/13.3.0/bits/alloc_traits.h:70:31: error: static assertion failed: allocator_traits<A>::rebind_alloc<A::value_type> must be A
2024-12-03T10:34:40.7564390Z    70 |                         _Tp>::value,
2024-12-03T10:34:40.7564782Z       |                               ^~~~~
2024-12-03T10:34:40.7566270Z /home/runner/work/op/op/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/x86_64-openwrt-linux-musl/include/c++/13.3.0/bits/alloc_traits.h:70:31: note: 'std::integral_constant<bool, false>::value' evaluates to false
2024-12-03T10:34:40.7568082Z cc1plus: note: unrecognized command-line option '-Wno-tautological-constant-compare' may have been intended to silence earlier diagnostics
2024-12-03T10:34:40.7569016Z ninja: build stopped: subcommand failed.
2024-12-03T10:34:40.7569543Z make[3]: Leaving directory '/mnt/openwrt/feeds/packages/libs/zmq'
2024-12-03T10:34:40.7570138Z time: package/feeds/packages/zmq/nc/compile#0.47#0.12#0.57
2024-12-03T10:34:40.7570802Z make[2]: *** [package/Makefile:116: package/feeds/packages/zmq/compile] Error 1
2024-12-03T10:34:40.7571549Z make[1]: *** [package/Makefile:110: /home/runner/work/op/op/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2024-12-03T10:34:40.7572185Z make: *** [/home/runner/work/op/op/openwrt/include/toplevel.mk:231: world] Error 2
```

GreatMichaelLee, 2024-12-03:
patch放这里 feeds/packages/libs/zmq/patches
[030-gcc13-compatibility.patch](https://github.com/user-attachments/files/17993380/030-gcc13-compatibility.patch)

coolsnowwolf, 2024-12-03:
可以顺便发个 PR 么

GreatMichaelLee, 2024-12-03:
> 可以顺便发个 PR 么

我是QA,不发PR的。。。。哈哈

binge8, 2024-12-03:
> 可以顺便发个 PR 么

已发PR

binge8, 2024-12-04:
> > 可以顺便发个 PR 么
> 
> 已发PR

哥哥，赶快合并啊，等着当小白鼠呢

coolsnowwolf, 2024-12-04:
好了，小问题

binge8, 2024-12-04:
辛苦

closed 2024-12-04 by binge8