coolsnowwolf/lede #13160: 编译失败

wangcome, 2024-12-03:
### 详细叙述

checking whether the C compiler works... no
configure: error: in `/home/runner/work/AutoBuild-Actions-BETA/AutoBuild-Actions-BETA/openwrt/build_dir/target-mipsel_24kc_musl/elfutils-0.192':
configure: error: C compiler cannot create executables
See `config.log' for more details
make[3]: *** [Makefile:129: /home/runner/work/AutoBuild-Actions-BETA/AutoBuild-Actions-BETA/openwrt/build_dir/target-mipsel_24kc_musl/elfutils-0.192/.configured_a17fb5ef857664f03cd0ce37cc5ea591] Error 77
make[3]: Leaving directory '/home/runner/work/AutoBuild-Actions-BETA/AutoBuild-Actions-BETA/openwrt/package/libs/elfutils'
time: package/libs/elfutils/compile#5.55#0.38#6.86
    ERROR: package/libs/elfutils failed to build.
make[2]: *** [package/Makefile:116: package/libs/elfutils/compile] Error 1
make[2]: Leaving directory '/home/runner/work/AutoBuild-Actions-BETA/AutoBuild-Actions-BETA/openwrt'
make[1]: *** [package/Makefile:110: /home/runner/work/AutoBuild-Actions-BETA/AutoBuild-Actions-BETA/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/runner/work/AutoBuild-Actions-BETA/AutoBuild-Actions-BETA/openwrt'
make: *** [/home/runner/work/AutoBuild-Actions-BETA/AutoBuild-Actions-BETA/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

newifi D2

### 详细日志

checking whether the C compiler works... no
configure: error: in `/home/runner/work/AutoBuild-Actions-BETA/AutoBuild-Actions-BETA/openwrt/build_dir/target-mipsel_24kc_musl/elfutils-0.192':
configure: error: C compiler cannot create executables
See `config.log' for more details
make[3]: *** [Makefile:129: /home/runner/work/AutoBuild-Actions-BETA/AutoBuild-Actions-BETA/openwrt/build_dir/target-mipsel_24kc_musl/elfutils-0.192/.configured_a17fb5ef857664f03cd0ce37cc5ea591] Error 77
make[3]: Leaving directory '/home/runner/work/AutoBuild-Actions-BETA/AutoBuild-Actions-BETA/openwrt/package/libs/elfutils'
time: package/libs/elfutils/compile#5.55#0.38#6.86
    ERROR: package/libs/elfutils failed to build.
make[2]: *** [package/Makefile:116: package/libs/elfutils/compile] Error 1
make[2]: Leaving directory '/home/runner/work/AutoBuild-Actions-BETA/AutoBuild-Actions-BETA/openwrt'
make[1]: *** [package/Makefile:110: /home/runner/work/AutoBuild-Actions-BETA/AutoBuild-Actions-BETA/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/runner/work/AutoBuild-Actions-BETA/AutoBuild-Actions-BETA/openwrt'
make: *** [/home/runner/work/AutoBuild-Actions-BETA/AutoBuild-Actions-BETA/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

coolsnowwolf, 2024-12-04:
已解决

closed 2024-12-04 by coolsnowwolf

jw10126121, 2024-12-11:
make[2]: *** [tools/Makefile:231: tools/elfutils/compile] Error 1
20241211最新master源码在macos下编译，还是出现这个问题。