coolsnowwolf/lede #13162: unknown option aarch64-openwrt-linux-musl-gcc

ftkey, 2024-12-04:
### 详细叙述

unknown option aarch64-openwrt-linux-musl-gcc

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

ER1

### 详细日志

2024-12-04T03:10:39.3218524Z make[5]: Leaving directory '/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/readline-8.2/doc'
2024-12-04T03:10:39.3220109Z /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/staging_dir/host/bin/install -c -m 644 /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/readline-8.2/readline.pc /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/readline-8.2/ipkg-install/usr/lib/pkgconfig/readline.pc
2024-12-04T03:10:39.3229508Z /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/staging_dir/host/bin/install -c -m 644 /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/readline-8.2/history.pc /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/readline-8.2/ipkg-install/usr/lib/pkgconfig/history.pc
2024-12-04T03:10:39.3241010Z mv /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/readline-8.2/ipkg-install/usr/lib/libreadline.a /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/readline-8.2/ipkg-install/usr/lib/libreadline.old
2024-12-04T03:10:39.3250065Z mv: cannot stat '/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/readline-8.2/ipkg-install/usr/lib/libreadline.a': No such file or directory
2024-12-04T03:10:39.3251241Z make[4]: [Makefile:255: install-static] Error 1 (ignored)
2024-12-04T03:10:39.3252671Z /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/staging_dir/host/bin/install -c -m 644 libreadline.a /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/readline-8.2/ipkg-install/usr/lib/libreadline.a
2024-12-04T03:10:39.3271258Z test -n "aarch64-openwrt-linux-musl-gcc-ranlib" && aarch64-openwrt-linux-musl-gcc-ranlib /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/readline-8.2/ipkg-install/usr/lib/libreadline.a
2024-12-04T03:10:39.3356197Z mv /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/readline-8.2/ipkg-install/usr/lib/libhistory.a /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/readline-8.2/ipkg-install/usr/lib/libhistory.old
2024-12-04T03:10:39.3364148Z mv: cannot stat '/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/readline-8.2/ipkg-install/usr/lib/libhistory.a': No such file or directory
2024-12-04T03:10:39.3365737Z make[4]: [Makefile:258: install-static] Error 1 (ignored)
2024-12-04T03:10:39.3367056Z /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/staging_dir/host/bin/install -c -m 644 libhistory.a /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/readline-8.2/ipkg-install/usr/lib/libhistory.a
2024-12-04T03:10:39.3378859Z test -n "aarch64-openwrt-linux-musl-gcc-ranlib" && aarch64-openwrt-linux-musl-gcc-ranlib /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/readline-8.2/ipkg-install/usr/lib/libhistory.a
2024-12-04T03:10:39.3416293Z test -d shlib || mkdir shlib
2024-12-04T03:10:39.3424293Z ( cd shlib ; make -w all )
2024-12-04T03:10:39.3467250Z make[5]: Entering directory '/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/readline-8.2/shlib'
2024-12-04T03:10:39.3468234Z make[5]: Nothing to be done for 'all'.
2024-12-04T03:10:39.3468931Z make[5]: Leaving directory '/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/readline-8.2/shlib'
2024-12-04T03:10:39.3470541Z ( cd shlib ; make -w DESTDIR=/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/readline-8.2/ipkg-install install )
2024-12-04T03:10:39.3499948Z make[5]: Entering directory '/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/readline-8.2/shlib'
2024-12-04T03:10:39.3501047Z /bin/sh ../support/mkdirs /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/readline-8.2/ipkg-install/usr/lib
2024-12-04T03:10:39.3509652Z /bin/sh ../support/mkdirs /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/readline-8.2/ipkg-install/usr/bin
2024-12-04T03:10:39.3626873Z mkdir /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/readline-8.2/ipkg-install/usr/bin
2024-12-04T03:10:39.3654979Z /bin/sh ../support/shlib-install -O linux-gnu -V openwrt -d /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/readline-8.2/ipkg-install/usr/lib -b /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/readline-8.2/ipkg-install/usr/bin -i "/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/staging_dir/host/bin/install -c -m 644" libhistory.so.8.2
2024-12-04T03:10:39.3762329Z /bin/sh ../support/shlib-install -O linux-gnu -V openwrt -d /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/readline-8.2/ipkg-install/usr/lib -b /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/readline-8.2/ipkg-install/usr/bin -i "/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/staging_dir/host/bin/install -c -m 644" libreadline.so.8.2
2024-12-04T03:10:39.3877665Z install: you may need to run ldconfig
2024-12-04T03:10:39.3879408Z make[5]: Leaving directory '/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/readline-8.2/shlib'
2024-12-04T03:10:39.3882949Z make[4]: Leaving directory '/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/readline-8.2'
2024-12-04T03:10:39.3884423Z touch /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/readline-8.2/.built
2024-12-04T03:10:39.3951427Z mkdir -p /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/bin/targets/qualcommax/ipq60xx/packages /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/readline-8.2/ipkg-aarch64_cortex-a53/libreadline/CONTROL /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/staging_dir/target-aarch64_cortex-a53_musl/pkginfo
2024-12-04T03:10:39.3980915Z install -d -m0755 /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/readline-8.2/ipkg-aarch64_cortex-a53/libreadline/usr/lib
2024-12-04T03:10:39.4012147Z cp -fpR /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/readline-8.2/ipkg-install/usr/lib/lib{history,readline}.so.* /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/readline-8.2/ipkg-aarch64_cortex-a53/libreadline/usr/lib/
2024-12-04T03:10:39.4063415Z find /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/readline-8.2/ipkg-aarch64_cortex-a53/libreadline -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
2024-12-04T03:10:39.4336368Z export CROSS="aarch64-openwrt-linux-musl-"  NO_RENAME=1 ; NM="aarch64-openwrt-linux-musl-nm" STRIP="/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/staging_dir/host/bin/sstrip -z" STRIP_KMOD="/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/scripts/strip-kmod.sh" PATCHELF="/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/staging_dir/host/bin/patchelf" /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/scripts/rstrip.sh /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/readline-8.2/ipkg-aarch64_cortex-a53/libreadline
2024-12-04T03:10:39.4427167Z rstrip.sh: /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/readline-8.2/ipkg-aarch64_cortex-a53/libreadline/usr/lib/libhistory.so.8.2: shared object
2024-12-04T03:10:39.4460658Z rstrip.sh: /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/readline-8.2/ipkg-aarch64_cortex-a53/libreadline/usr/lib/libhistory.so.8.2: removing rpath /usr/lib
2024-12-04T03:10:39.4500516Z rstrip.sh: /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/readline-8.2/ipkg-aarch64_cortex-a53/libreadline/usr/lib/libreadline.so.8.2: shared object
2024-12-04T03:10:39.4534616Z rstrip.sh: /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/build_dir/target-aarch64_cortex-a53_musl/readline-8.2/ipkg-aarch64_cortex-a53/libreadline/usr/lib/libreadline.so.8.2: removing rp [... cut ...]
2024-12-04T03:10:39.8243970Z make[1]: *** [package/Makefile:110: /home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
2024-12-04T03:10:39.8246630Z make[1]: Leaving directory '/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt'
2024-12-04T03:10:39.8253977Z make: *** [/home/runner/work/OpenWRT-CI/OpenWRT-CI/wrt/include/toplevel.mk:231: world] Error 2
2024-12-04T03:10:39.8273603Z ##[error]Process completed with exit code 2.
2024-12-04T03:10:39.8352564Z Post job cleanup.

coolsnowwolf, 2024-12-04:
你的 host os 缺包， readline-devel 这个是需要的

ftkey, 2024-12-04:
> 你的 host os 缺包， readline-devel 这个是需要的

好像不是这个原因环境有,  更新过[一次gcc源码](toolchain: gcc: switch default to 13) 之后,出现的问题.

sxml, 2024-12-04:
这怎么解决啊 跟gcc有关吗？云编译 也报错在这

ERROR: unknown option aarch64-openwrt-linux-musl-gcc
Try './configure --help' for more information
make[3]: *** [Makefile:62: /workdir/openwrt/build_dir/target-aarch64_generic_musl/liburing-2.3/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/libs/liburing'
time: package/feeds/packages/liburing/compile#0.15#0.05#0.18
ERROR: package/feeds/packages/liburing failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/liburing/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2

closed 2024-12-04 by coolsnowwolf