coolsnowwolf/lede #13169: 12月4日的最新代码编译r2s固件，无法启动

freeluo22, 2024-12-05:
### 详细叙述

12月4日的最新代码编译r2s固件，无法启动

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

r2s

### 详细日志

12月4日的最新代码编译r2s固件，无法启动

Jermin, 2024-12-07:
I’m having the same issue.

coolsnowwolf, 2024-12-07:
ttl 日志亏哭

freeluo22, 2024-12-08:
> ttl 日志看看？

大佬，网上查了一下，需要系统启动起来才可以看，但是我的系统没有启动起来，那还有什么方法可以查看，今天又编译了12月7日的最新代码，还是无法启动

SeeDLL, 2024-12-08:
我也遇到了同样的问题，正常启动表现为电源SYS灯等个三五秒，就骤然变亮，不是那种暗淡的灯光，然后后续就开始闪烁表示正常启动了。
现在最新代码编译的启动表现为：SYS灯就一直是那种暗淡的灯光，没有变亮，也没有闪烁，就好像卡在了从SD卡读取信息的阶段，都还没到引导阶段。
但是换了八九月份边编译的镜像，同一张SD卡又可以正常启动了。我有两张SD卡，两张都测试了。
所以怀疑是R2S或者是ARM64的镜像有问题？
同版本代码还没编译过X86版本，等这两天抽空在编译个X86测试看看。

chen-wilde, 2024-12-08:
#12964

SeeDLL, 2024-12-09:
> 12964

意思是可能是这个讨论里的那几个改动导致的开不了机？我去试试看

SeeDLL, 2024-12-11:
我测试了，用了这两个commit 来测试的，第一个commit是：Revert "rkbin: update to latest git HEAD"	cb33d17db	coolsnowwolf <coolsnowwolf@gmail.com>	2024年11月13日 23:38。
第二个commit是：rkbin: update to latest git HEAD	f0ab31724	aiamadeus <2789289348@qq.com>	2024年11月7日 21:26
第二个commit里提交的rockchip-rkbin/Makefile 里的版本从 PKG_SOURCE_DATE:=2024-02-22升级到了 PKG_SOURCE_DATE:=2024-10-23。
连带着文件从 rk33/rk3328_ddr_333MHz_v1.19.bin 升级到了：rk33/rk3328_ddr_333MHz_v1.20.bin ，到了1.20的版本，在R2S下，就完全无法启动了。
只有第一个commit里把版本降级回了 2024-02-22的版本，rk33/rk3328_ddr_333MHz_v1.20.bin降级回了 v1.19 版本，才可以正常启动。
但这个commit又在 更新的commit af207e5 被升级到1.20版本了，所以到了这里，r2s就完全没办法启动了。

closed 2024-12-19 by freeluo22

zoyo94, 2025-01-14:
> 我测试了，用了这两个commit 来测试的，第一个commit是：Revert “rkbin： update to latest git HEAD” [cb33d17](https://github.com/coolsnowwolf/lede/commit/cb33d17dbaf0d9ee191653e825b79d2286c086ad) coolsnowwolf [coolsnowwolf@gmail.com](mailto:coolsnowwolf@gmail.com) 2024年11月13日 23：38。
> 
> 第二个提交的rockchip-rkbin/Makefile 里的版本从 PKG_SOURCE_DATE：=2024-02-22升级到了 PKG_SOURCE_DATE：=2024-10-23。 连带着文件从 rk33/rk3328_ddr_333MHz_v1.19.bin 升级到了：rk33/rk3328_ddr_333MHz_v1.20.bin ，到了1.20的版本，在R2S下，就完全无法启动了。 只有第一个commit里把版本降级回了 2024-02-22的版本，rk33/rk3328_ddr_333MHz_v1.20.bin降级回了 v1.19 版本，才可以正常启动。 但这个commit又在 更新的commit [af207e5](https://github.com/coolsnowwolf/lede/commit/af207e5f62bb88247e647483c07a4ea9ca33d318) 被升级到1.20版本了，所以到了这里，r2s就完全没办法启动了。

##### 感谢编译没问题，可以进入界面。我就说嘛怎么死活进不了页面。