coolsnowwolf/lede #13173: make -j8 download编译过程好多组件源都失效了，无法编译

coolveyron, 2024-12-05:
### 详细叙述

一开始以为是网络的问题，在香港和硅谷了开了两个服务器，都是显示连接超时或者无法下载，看了看过程发现有些组件都没了。
time: package/kernel/linux/download#4.48#0.21#12.82
  0   146    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
curl: (22) The requested URL returned error: 404
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location https://mirror2.openwrt.org/sources/ubus-2024-10-20-252a9b0c.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0   153    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404
Download failed.
No more mirrors to try - giving up.
Checking out files from the git repository...
Cloning into 'ubus-2024-10-20-252a9b0c'...

Hash of the downloaded file does not match (file: a9b74b7b95acc84a5a5c33d6acf493faad8f161caca3180734d9bd383c9d823f, requested: ec8304dc06f94338e14e608a807f48e10d0987bdac4d90f235650b46994dfbd7) - deleting download.
+ curl -f --connect-timeout 20 --retry 5 --location https://mirror2.openwrt.org/sources/make-ext4fs-2020-01-05-5c201be7.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0   153    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404
Download failed.

以上是仅两个例子，ubus-2024-10-20-252a9b0c.tar.xz和make-ext4fs-2020-01-05-5c201be7.tar.xz都404了。
以为我自己加的东西太多了，make menuconfig后啥都不改动，save直接用原始编译，make download也是同样的问题。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

X86

### 详细日志

time: package/kernel/linux/download#4.48#0.21#12.82
  0   146    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
curl: (22) The requested URL returned error: 404
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location https://mirror2.openwrt.org/sources/ubus-2024-10-20-252a9b0c.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0   153    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404
Download failed.
No more mirrors to try - giving up.
Checking out files from the git repository...
Cloning into 'ubus-2024-10-20-252a9b0c'...

Hash of the downloaded file does not match (file: a9b74b7b95acc84a5a5c33d6acf493faad8f161caca3180734d9bd383c9d823f, requested: ec8304dc06f94338e14e608a807f48e10d0987bdac4d90f235650b46994dfbd7) - deleting download.
+ curl -f --connect-timeout 20 --retry 5 --location https://mirror2.openwrt.org/sources/make-ext4fs-2020-01-05-5c201be7.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0   153    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404
Download failed.

binge8, 2024-12-05:
不存在的，就是你的科学上网问题

coolveyron, 2024-12-05:
> > > 不存在的，就是你的科学上网问题
> > 
> > 
> > 没看清楚吗？是404，里面就没有目标文件,而且是香港和硅谷两个服务器，科学什么上网，香港和美国还要科学上网？
> 
> 本事不大脾气还不小嘞，有本事自己解决，在这里吠什么？硅谷香港服务器就了不得了？我也国外服务器，你上面这几个我都成功下载编译了，再说了这么大的问题为什么只有你存在？别人没一个提的？

你就挺奇怪的人，解决不了问题，你回我信息干啥？你是怎么知道有多少人全新编译和有多少人二次编译会不会出现此类问题？你挑点重点的问题攻击我行么？两个地域服务器表达了没有科学上网的问题，怎么到你嘴里就变成了我不得了了？怎么了，踩着你尾巴了？真是网络的问题，换了两个大地区，网络环境都不同，还是一样的问题出现，是你脑回路要重置一下了吧。你把整个包rm掉，重新clone试试再说。

binge8, 2024-12-05:
你自己的问题我怎么给你解决？就你那没教养的态度我不攻击你我攻击谁？还科学什么上网，不会说人话就不要说。多大人了怎么就听不明白呢，我再重复一遍，这么多人编译就只有你存在这个问题，就不知道考虑考虑自己的问题？这么多云编译的哪次不是全新环境下载编译，还什么rm掉。你两个服务器又能证明什么？比如你用同一种错误的方式部署上百个区的服务器，那这一百多个都肯定有问题啊，我只服你这脑回路，快重置下吧

binge8, 2024-12-05:
我都多余搭理你这种人，老老实实看看自己到底哪一步错了自己解决，你自己的问题别人谁也解决不了

coolveyron, 2024-12-05:
mkdir -p /home/ubuntu/lede/dl
SHELL= flock /home/ubuntu/lede/tmp/.firmware-utils-2024-10-20-4b763892.tar.xz.flock -c '  	/home/ubuntu/lede/scripts/download.pl "/home/ubuntu/lede/dl" "firmware-utils-2024-10-20-4b763892.tar.xz" "ea0e30f2b20211e4153704019ebb3f8789269893a4af6d853a85fae7c091802c" "" || (  	echo "Checking out files from the git repository..."; mkdir -p /home/ubuntu/lede/tmp/dl && cd /home/ubuntu/lede/tmp/dl && rm -rf firmware-utils-2024-10-20-4b763892 && [ \! -d firmware-utils-2024-10-20-4b763892 ] && git clone  https://git.openwrt.org/project/firmware-utils.git firmware-utils-2024-10-20-4b763892 && (cd firmware-utils-2024-10-20-4b763892 && git checkout 4b7638925d3eac03e614e40bc30cb49f5877c46d && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd firmware-utils-2024-10-20-4b763892 && git log -1 --format='\''@%ct'\''` && rm -rf firmware-utils-2024-10-20-4b763892/.git && 	tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c firmware-utils-2024-10-20-4b763892 |     xz -zc -7e > /home/ubuntu/lede/tmp/dl/firmware-utils-2024-10-20-4b763892.tar.xz && mv /home/ubuntu/lede/tmp/dl/firmware-utils-2024-10-20-4b763892.tar.xz /home/ubuntu/lede/dl/ && rm -rf firmware-utils-2024-10-20-4b763892;  )    '
+ curl -f --connect-timeout 20 --retry 5 --location https://sources.cdn.openwrt.org/firmware-utils-2024-10-20-4b763892.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0   146    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location https://sources.openwrt.org/firmware-utils-2024-10-20-4b763892.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0   146    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location https://mirror2.openwrt.org/sources/firmware-utils-2024-10-20-4b763892.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0   153    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404
Download failed.
No more mirrors to try - giving up.
Checking out files from the git repository...
Cloning into 'firmware-utils-2024-10-20-4b763892'...
fatal: unable to access 'https://git.openwrt.org/project/firmware-utils.git/': Failed to connect to git.openwrt.org port 443 after 130055 ms: Connection timed out
make[2]: *** [Makefile:22: /home/ubuntu/lede/dl/firmware-utils-2024-10-20-4b763892.tar.xz] Error 128
make[2]: Leaving directory '/home/ubuntu/lede/tools/firmware-utils'
time: tools/firmware-utils/download#0.27#0.04#131.53
    ERROR: tools/firmware-utils failed to build.
make[1]: *** [tools/Makefile:231: tools/firmware-utils/download] Error 1

download.pl里面有三个源

push @mirrors, 'https://sources.cdn.openwrt.org';
push @mirrors, 'https://sources.openwrt.org';
push @mirrors, 'https://mirror2.openwrt.org/sources';

每个都看一下
firmware-utils-2024-10-20-4b763892.tar.xz
https://sources.cdn.openwrt.org/里面有firmware-utils-2024-10-20-4b763892，只不过扩展名变成了firmware-utils-2024.10.20~4b763892.tar.zst，所以提示404
https://sources.openwrt.org里面也有firmware-utils-2024-10-20-4b763892，扩展名也是firmware-utils-2024.10.20~4b763892.tar.zst
https://mirror2.openwrt.org/sources里面没有firmware-utils-2024-10-20-4b763892，
三个源全都git不到

coolveyron, 2024-12-05:
> 我都多余搭理你这种人，老老实实看看自己到底哪一步错了自己解决，你自己的问题别人谁也解决不了

小丑，在蹦出来看看，笑死了!

binge8, 2024-12-05:
这特么还真是有臭味相投的人，还有这个没教养的傻x的这副嘴脸，我也真是服了，看看我十几分钟前刚编译成功的，而且我选的插件比你们俩多了去了，一样成功，没有报错，云编译全新环境全部重新下载

binge8, 2024-12-05:
没教养的傻x这幅嘴脸还真是不一样，看看我十几分钟前刚编译成功的，而且我选的插件比你多了去了，一样成功，没有报错，云编译全新环境全部重新下载
[image]

binge8, 2024-12-05:
自己玩你的勺子瓣去吧，没工夫和傻x浪费时间

coolveyron, 2024-12-05:
> 自己玩你的勺子瓣去吧，没工夫和傻x浪费时间

抱歉啊，我不用云编译，谢谢啊~慢慢用你的云编译吧。只会一个config，你还会什么？

目前的问题是有些组件的压缩格式变成zst，dl不了，就算能dl也解压缩不了，运行环境还要安装zstd，同时还要修改dl_github_archive.py，你玩好你的config吧。唉，现在年轻人真是，一瓶子不满，半瓶子瞎逛荡。

咱承认有些东西动不了，等L大修复吧。你继续玩你的config。

binge8, 2024-12-05:
你是真大仙，还在找库的原因，就不能查查自己的是不是哪里搞错了？你不用管我懂多少，我就用原始库，也不懂什么dl_github_archive.py，但我一样成功下载，成功解压缩，成功编译。换个角度考虑，只要有一个人用原始的库啥也没改就成功了，是不是就能证明人家库没问题？不要那么死板钻牛角尖

hicom-omni, 2024-12-06:
我也下载失败
[image]
[image]

xingshu1990, 2024-12-06:
https://sources.cdn.openwrt.org/
上面这个地址里ubus改了名字[ubus-2024.10.20~252a9b0c.tar.zst](https://sources.cdn.openwrt.org/ubus-2024.10.20~252a9b0c.tar.zst)
先试着自己下载文件到dl目录，重新编译，或者按照下方的内容安装md5sum    sha1sum    sha256sum
或者改校验码重新校验，
https://www.right.com.cn/forum/thread-4106115-1-1.html

另外前几天我捣鼓新的源码，编译网易云的时候，一直出现UPX不能下载，然后如下图操作，
如果按图操作找不到{upx}，自行修改为正确的路径。
[image]

yanxin152133, 2024-12-06:
> 你是真大仙，还在找库的原因，就不能查查自己的是不是哪里搞错了？你不用管我懂多少，我就用原始库，也不懂什么dl_github_archive.py，但我一样成功下载，成功解压缩，成功编译。换个角度考虑，只要有一个人用原始的库啥也没改就成功了，是不是就能证明人家库没问题？不要那么死板钻牛角尖
不会帮人解决问题就多练练啊哥们
[image]

binge8, 2024-12-06:
> 你是真大仙，还在找库的原因，就不能查查自己的是不是哪里搞错了？你不用管我懂多少，我就用原始库，也不懂什么dl_github_archive.py，但我一样成功下载，成功解压缩，成功编译。换个角度考虑，只要有一个人用原始的库啥也没改就成功了，是不是就能证明人家库没问题？不要那么死板钻牛角尖
> 不会帮人解决问题就多练练啊哥们 [...]
真有意思，反正我这里没问题，我当然认为没问题，你觉得有问题也觉得会解决，那你直接提pr不就完了，啥也不是

aiamadeus, 2024-12-06:
这就是我讨厌云编译党的地方，就没人发现题主提到的 ubus hash 没错，错的是 tools，浪费我时间。。
[image]

binge8, 2024-12-06:
> 这就是我讨厌云编译党的地方，就没人发现题主提到的 ubus hash 没错，错的是 tools，浪费我时间。。 [...]

你这话也不能这么说，毕竟用云编译的不算大多数吧，首先如果真的是hash错了，肯定会报错停止，其次如果真的hash错了，那么多非云编译的我不信没有一个人发现，就上面这两三个人发现的还提到的是直接是下载404，也不是hash错

aiamadeus, 2024-12-06:
https://github.com/coolsnowwolf/lede/commit/2778ccf7a348e907a55e5dbb9cfda7738aa7cb52 自己试试就知道了

yanxin152133, 2024-12-06:
> 你这话也不能这么说，毕竟用云编译的不算大多数吧，首先如果真的是hash错了，肯定会报错停止，其次如果真的hash错了，那么多非云编译的我不信没有一个人发现，就上面这两三个人发现的还提到的是直接是下载404，也不是hash错

下载404就是代理不行吗，网络问题那为什么不能是连接超时呢？你自己提的问题别人云编译的不也没发现吗？这是公共场所不是你家，人家质疑你的解决方案看你没素质的样子

binge8, 2024-12-06:
你是楼主的傻x二弟吧，他是来求助，我给个方案即使错了，提出来不就完了，你看看他怎么说的，科学什么上网，这句话带几层意思你听不出来吗，还是智商不够？真是物以类聚人以群分，再见了傻x

binge8, 2024-12-06:
> 下载404就是代理不行吗，网络问题那为什么不能是连接超时呢？你自己提的问题别人云编译的不也没发现吗？这是公共场所不是你家，人家质疑你的解决方案看你没素质的样子

你是楼主的傻x二弟吧，他是来求助，我好心给个方案即使错了，提出来不就完了，你看看他怎么说的，科学什么上网，这句话带几层意思你听不出来吗，还是智商不够？真是物以类聚人以群分，跟你这种傻x二弟废话闲的我真是，再见了傻x

closed 2024-12-06 by coolsnowwolf

yanxin152133, 2025-01-15:
https://github.com/coolsnowwolf/lede/issues/13275#issuecomment-2592756249

你说这人是不是科兴忘打了

[image]

> > 你是楼主的傻x二弟吧，他是来求助，我好心给个方案即使错了，提出来不就完了，你看看他怎么说的，科学什么上网，这句话带几层意思你听不出来吗，还是智商不够？真是物以类聚人以群分，跟你这种傻x二弟废话闲的我真是，再见了傻x
> 
> 来，小朋友，我教你一下，你看仔细了
> 
> 第一点，我提到了是无法下载的原因是404，不是网络连接的问题，而且明确指出了我选择的服务器在国外，不存在科学上网的问题，你上来就一句科学上网，你根本没看明白，或者是根本不明白，所以我直接回你科学上什么网，网络的问题已经验证完毕了，不存在问题。
> 
> 第二点，我还提到了二次变异和全新编译的问题，我所说的是单机编译，不是你们云编译，二次编译在make clean或者make dirclean之后，有些组件已经被下载过了，所以编译过程dl下载没有问题，不代表全新编译不会出问题，你让你rm掉，重新git clone，我以为你也是单机编译，弄了半天你是云编译，怪不得你不知道我在说啥，无语。
> 
> 第三点，为什么云编译正常?云编译的脚本git的是哪个版本，脚本下载的组件是哪个版本？你知道？没错，旧版本的源组件名字没改，可以正常下载，不代表新组件可以正常下载，因为部分新组件的压缩格式在源里面确实变了。简单的说，你云编译的脚本都是别人给你写好的，下载组件的位置、版本等都是别人写的，编译过程的组件版本你都不知道，你感觉正常了就正常了对吧。
> 
> 有问题就是有问题，在技术上不承认，到处找补理由，核心点是大家一起分析原因，你上来就一句科学上网，而且整个过程你一直在diss我，根本没有与解决问题相关，虽然我在跟你吵，但是我一直在找问题的原因，你看看后面几个朋友回复的？别人都在验证问题，你在diss来，diss去的。你的性格有很大问题，如果你是公司的技术岗位，挺危险的，第一个先开了你。

1WuNeiGui, 2025-01-18:
> 我也下载失败 [image] [image]一样，我新环境编译R66S也是这样，下载失败，腾讯云的香港、日本、美国都不行。但是我之前的台式机可以，我觉得可能是我台式机之前编译成功过，有缓存可能

coolveyron, 2025-01-22:
腾讯云的所有服务器已经无法正常编译openwrt，换阿里云吧。所有腾讯云的服务器都无法下载部分组件。

核心问题在git.openwrt.org上，能ping通，但是返回的是假地址，是无法从git.openwrt.org中获取正确的组件hash，所以下载都出错。

已经问过腾讯云的客服了，他们知道，说是运营商的问题，正在积极协调解决。

美区？香港？日本？都是运营商的问题？我没搞明白他们的网络出口的架构是啥样的。还是服务器都在国内，走GFW后再通过他们自己的专线到其他境外服务器？GFW给屏蔽了？但是阿里云没问题啊。

不懂腾讯云的操作是不是故意而为之。

coolveyron, 2025-01-22:
> > 我也下载失败 [image] ![屏幕截图 2024-12-06 092154](https://private-user-images.githubusercontent.com/190793815/393058322-ff898214-9f26-4c49-8ba3-821eb7a87943.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mzc1NzAxNTUsIm5iZiI6MTczNzU2OTg1NSwicGF0aCI6Ii8xOTA3OTM4MTUvMzkzMDU4MzIyLWZmODk4MjE0LTlmMjYtNGM0OS04YmEzLTgyMWViN2E4Nzk0My5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMTIyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDEyMlQxODE3MzVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wNTgwM2RlNWY5ZWY0MGI5MmQ3OTc2MTgwNzhlMmUxYTQyNGY0YjUzYTk1YWU1NWU4NGE5M2U0MzJhZTM5NWNiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.x7uZdR8tOkrCEF8XDNHObBlXP_xnFbZyu25v8p8kqZs)一样，我新环境编译R66S也是这样，下载失败，腾讯云的香港、日本、美国都不行。但是我之前的台式机可以，我觉得可能是我台式机之前编译成功过，有缓存可能

腾讯云的所有服务器已经无法正常编译openwrt，换阿里云吧。所有腾讯云的服务器都无法下载部分组件。

核心问题在git.openwrt.org上，能ping通，但是返回的是假地址，是无法从git.openwrt.org中获取正确的组件hash，所以下载都出错。

已经问过腾讯云的客服了，他们知道，说是运营商的问题，正在积极协调解决。

美区？香港？日本？都是运营商的问题？我没搞明白他们的网络出口的架构是啥样的。还是服务器都在国内，走GFW后再通过他们自己的专线到其他境外服务器？GFW给屏蔽了？但是阿里云没问题啊。

不懂腾讯云的操作是不是故意而为之。

1WuNeiGui, 2025-01-24:
> 腾讯云的所有服务器已经无法正常编译openwrt，换阿里云吧。所有腾讯云的服务器都无法下载部分组件。
>
> 核心问题在git.openwrt.org上，能ping通，但是返回的是假地址，是无法从git.openwrt.org中获取正确的组件hash，所以下载都出错。
>
> 已经问过腾讯云的客服了，他们知道，说是运营商的问题，正在积极协调解决。 [...]

我按照您说的，试了一下阿里云的服务器，确实可以编译成功了，是腾讯云的事。但是不知道腾讯云啥时候能修好，腾讯云的按量付费的抢占实例比阿里云便宜和稳定

eritpchy, 2025-01-25:
[image]
分享一下，这种情况通常是从git前面部分失败了，比如网络问题，又或者是我自己用Docker这个环境太干净导致的（缺少sha256命令)， 当然后面报那些mirror 404确实是没有的，出错的位置也不够准确给人带来误解

hicom-omni, 2025-01-26:
> 我按照您说的，试了一下阿里云的服务器，确实可以编译成功了，是腾讯云的事。但是不知道腾讯云啥时候能修好，腾讯云的按量付费的抢占实例比阿里云便宜和稳定

的确，腾讯云要比阿里云稳定和便宜，希望腾讯云早点修复问题

1WuNeiGui, 2025-02-09:
2025.2.9下午5点50测试，腾讯云香港依然不行

barton0426, 2025-03-18:
+1 也是404