coolsnowwolf/lede #13174: GCC升级了，subconverter编译报错，因为源代码没有显式声明类型，每次编译前都要往十几个文件头部添加#include <cstdint>才能完成编译，很麻烦，大佬们有好办法彻底解决吗？

kenwestern, 2024-12-06:
### 详细叙述

GCC升级了，subconverter编译报错，因为源代码没有显式声明类型，每次编译前都要往十几个文件头部添加#include <cstdint>才能完成编译，很麻烦，大佬们有好办法彻底解决吗？

源报错长度超限了，下面的只是部分报错代码，都是类似的报错，示意

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

i7 14700K

### 详细日志

/home/l/lede/build_dir/target-x86_64_musl/subconverter-0.7.2-2023-01-13-3813c392/src/handler/settings.cpp: In function 'void readTOMLConf(toml::value&)':
/home/l/lede/build_dir/target-x86_64_musl/subconverter-0.7.2-2023-01-13-3813c392/src/handler/settings.cpp:601:17: warning: possibly dangling reference to a temporary [-Wdangling-reference]
  601 |     const auto &section_common = toml::find(root, "common");
      |                 ^~~~~~~~~~~~~~
/home/l/lede/build_dir/target-x86_64_musl/subconverter-0.7.2-2023-01-13-3813c392/src/handler/settings.cpp:601:44: note: the temporary was destroyed at the end of the full expression 'toml::find<discard_comments, std::unordered_map, std::vector>((* & root), std::__cxx11::basic_string<char>(((const char*)"common"), std::allocator<char>()))'
  601 |     const auto &section_common = toml::find(root, "common");
      |                                  ~~~~~~~~~~^~~~~~~~~~~~~~~~
/home/l/lede/build_dir/target-x86_64_musl/subconverter-0.7.2-2023-01-13-3813c392/src/handler/settings.cpp:640:17: warning: possibly dangling reference to a temporary [-Wdangling-reference]
  640 |     const auto &section_node_pref = toml::find(root, "node_pref");
      |                 ^~~~~~~~~~~~~~~~~
/home/l/lede/build_dir/target-x86_64_musl/subconverter-0.7.2-2023-01-13-3813c392/src/handler/settings.cpp:640:47: note: the temporary was destroyed at the end of the full expression 'toml::find<discard_comments, std::unordered_map, std::vector>((* & root), std::__cxx11::basic_string<char>(((const char*)"node_pref"), std::allocator<char>()))'
  640 |     const auto &section_node_pref = toml::find(root, "node_pref");
      |                                     ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~
/home/l/lede/build_dir/target-x86_64_musl/subconverter-0.7.2-2023-01-13-3813c392/src/handler/settings.cpp:659:17: warning: possibly dangling reference to a temporary [-Wdangling-reference]
  659 |     const auto &section_managed = toml::find(root, "managed_config");
      |                 ^~~~~~~~~~~~~~~
/home/l/lede/build_dir/target-x86_64_musl/subconverter-0.7.2-2023-01-13-3813c392/src/handler/settings.cpp:659:45: note: the temporary was destroyed at the end of the full expression 'toml::find<discard_comments, std::unordered_map, std::vector>((* & root), std::__cxx11::basic_string<char>(((const char*)"managed_config"), std::allocator<char>()))'
  659 |     const auto &section_managed = toml::find(root, "managed_config");
      |                                   ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~
/home/l/lede/build_dir/target-x86_64_musl/subconverter-0.7.2-2023-01-13-3813c392/src/handler/settings.cpp:669:17: warning: possibly dangling reference to a temporary [-Wdangling-reference]
  669 |     const auto &section_surge_external = toml::find(root, "surge_external_proxy");
      |                 ^~~~~~~~~~~~~~~~~~~~~~
/home/l/lede/build_dir/target-x86_64_musl/subconverter-0.7.2-2023-01-13-3813c392/src/handler/settings.cpp:669:52: note: the temporary was destroyed at the end of the full expression 'toml::find<discard_comments, std::unordered_map, std::vector>((* & root), std::__cxx11::basic_string<char>(((const char*)"surge_external_proxy"), std::allocator<char>()))'
  669 |     const auto &section_surge_external = toml::find(root, "surge_external_proxy");
      |                                          ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/l/lede/build_dir/target-x86_64_musl/subconverter-0.7.2-2023-01-13-3813c392/src/handler/settings.cpp:675:17: warning: possibly dangling reference to a temporary [-Wdangling-reference]
  675 |     const auto &section_emojis = toml::find(root, "emojis");
      |                 ^~~~~~~~~~~~~~
/home/l/lede/build_dir/target-x86_64_musl/subconverter-0.7.2-2023-01-13-3813c392/src/handler/settings.cpp:675:44: note: the temporary was destroyed at the end of the full expression 'toml::find<discard_comments, std::unordered_map, std::vector>((* & root), std::__cxx11::basic_string<char>(((const char*)"emojis"), std::allocator<char>()))'
  675 |     const auto &section_emojis = toml::find(root, "emojis");
      |                                  ~~~~~~~~~~^~~~~~~~~~~~~~~~
/home/l/lede/build_dir/target-x86_64_musl/subconverter-0.7.2-2023-01-13-3813c392/src/handler/settings.cpp:690:17: warning: possibly dangling reference to a temporary [-Wdangling-reference]
  690 |     const auto &section_ruleset = toml::find(root, "ruleset");
      |                 ^~~~~~~~~~~~~~~
/home/l/lede/build_dir/target-x86_64_musl/subconverter-0.7.2-2023-01-13-3813c392/src/handler/settings.cpp:690:45: note: the temporary was destroyed at the end of the full expression 'toml::find<discard_comments, std::unordered_map, std::vector>((* & root), std::__cxx11::basic_string<char>(((const char*)"ruleset"), std::allocator<char>()))'
  690 |     const auto &section_ruleset = toml::find(root, "ruleset");
      |                                   ~~~~~~~~~~^~~~~~~~~~~~~~~~~
/home/l/lede/build_dir/target-x86_64_musl/subconverter-0.7.2-2023-01-13-3813c392/src/handler/settings.cpp:702:17: warning: possibly dangling reference to a temporary [-Wdangling-reference]
  702 |     const auto &section_template = toml::find(root, "template");
      |                 ^~~~~~~~~~~~~~~~
/home/l/lede/build_dir/target-x86_64_musl/subconverter-0.7.2-2023-01-13-3813c392/src/handler/settings.cpp:702:46: note: the temporary was destroyed at the end of the full expression 'toml::find<discard_comments, std::unordered_map, std::vector>((* & root), std::__cxx11::basic_string<char>(((const char*)"template"), std::allocator<char>()))'
  702 |     const auto &section_template = toml::find(root, "template");
      |                                    ~~~~~~~~~~^~~~~~~~~~~~~~~~~~
/home/l/lede/build_dir/target-x86_64_musl/subconverter-0.7.2-2023-01-13-3813c392/src/handler/settings.cpp:722:17: warning: possibly dangling reference to a temporary [-Wdangling-reference]
  722 |     const auto &section_server = toml::find(root, "server");
      |                 ^~~~~~~~~~~~~~
/home/l/lede/build_dir/target-x86_64_musl/subconverter-0.7.2-2023-01-13-3813c392/src/handler/settings.cpp:722:44: note: the temporary was destroyed at the end of the full expression 'toml::find<discard_comments, std::unordered_map, std::vector>((* & root), std::__cxx11::basic_string<char>(((const char*)"server"), std::allocator<char>()))'
  722 |     const auto &section_server = toml::find(root, "server");
      |                                  ~~~~~~~~~~^~~~~~~~~~~~~~~~
/home/l/lede/build_dir/target-x86_64_musl/subconverter-0.7.2-2023-01-13-3813c392/src/handler/settings.cpp:731:17: warning: possibly dangling reference to a temporary [-Wdangling-reference]
  731 |     const auto &section_advanced = toml::find(root, "advanced");
      |                 ^~~~~~~~~~~~~~~~
/home/l/lede/build_dir/target-x86_64_musl/subconverter-0.7.2-2023-01-13-3813c392/src/handler/settings.cpp:731:46: note: the temporary was destroyed at the end of the full expression 'toml::find<discard_comments, std::unordered_map, std::vector>((* & root), std::__cxx11::basic_string<char>(((const char*)"advanced"), std::allocator<char>()))'
  731 |     const auto &section_advanced = toml::find(root, "advanced");
[... 367 lines cut ...]
/home/l/lede/build_dir/target-x86_64_musl/subconverter-0.7.2-2023-01-13-3813c392/src/utils/md5/md5.cpp:295:45: error: 'BLOCK_SIZE' is not a member of 'md5'
  295 |                 process_block(stored + md5::BLOCK_SIZE);
      |                                             ^~~~~~~~~~
/home/l/lede/build_dir/target-x86_64_musl/subconverter-0.7.2-2023-01-13-3813c392/src/utils/md5/md5.cpp:298:43: error: 'signature' was not declared in this scope; did you mean 'signature_'?
  298 |             get_result(static_cast<void*>(signature));
      |                                           ^~~~~~~~~
      |                                           signature_
/home/l/lede/build_dir/target-x86_64_musl/subconverter-0.7.2-2023-01-13-3813c392/src/utils/md5/md5.cpp:301:38: error: 'str' was not declared in this scope; did you mean 'std'?
  301 |             sig_to_string(signature, str, MD5_STRING_SIZE);
      |                                      ^~~
      |                                      std
[... cut ...]
/home/l/lede/build_dir/target-x86_64_musl/subconverter-0.7.2-2023-01-13-3813c392/src/utils/md5/md5.cpp:330:32: error: 'signature' was not declared in this scope; did you mean 'signature_'?
  330 |             memcpy(signature_, signature, MD5_SIZE);
      |                                ^~~~~~~~~
      |                                signature_
/home/l/lede/build_dir/target-x86_64_musl/subconverter-0.7.2-2023-01-13-3813c392/src/utils/md5/md5.cpp:330:43: error: 'MD5_SIZE' was not declared in this scope
  330 |             memcpy(signature_, signature, MD5_SIZE);
      |                                           ^~~~~~~~

coolsnowwolf, 2024-12-06:
这个库有这个程序么？

coolsnowwolf, 2026-09-05:
当前 LEDE 主仓库及官方 feeds 中没有 `subconverter` 包，issue 日志来自外部/自定义包（路径为 `build_dir/.../subconverter-0.7.2`）；因此无法在本仓库修复其 C++ 头文件问题。建议向对应第三方 feed 提交 `#include <cstdint>` 修复，现关闭。

closed 2026-09-05 by coolsnowwolf