coolsnowwolf/lede #13175: 出现很多这个提示：Max auth tries reached - user 'is invalid'

drw2005, 2024-12-06:
### 详细叙述

最近一直出现这个提示，请问有没有影响！
感谢！

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

Thu Dec  5 17:43:57 2024 authpriv.warn dropbear[12388]: Login attempt for nonexistent user
Thu Dec  5 17:43:58 2024 authpriv.info dropbear[12388]: Exit before auth from <125.70.52.249:61400>: Max auth tries reached - user 'is invalid'
Thu Dec  5 17:43:59 2024 authpriv.info dropbear[12392]: Child connection from 125.70.52.249:61426
Thu Dec  5 17:43:59 2024 authpriv.warn dropbear[12392]: Login attempt for nonexistent user
Thu Dec  5 17:44:00 2024 authpriv.info dropbear[12392]: Exit before auth from <125.70.52.249:61426>: Max auth tries reached - user 'is invalid'
Thu Dec  5 17:44:00 2024 authpriv.info dropbear[12396]: Child connection from 125.70.52.249:61446
Thu Dec  5 17:44:00 2024 authpriv.warn dropbear[12396]: Login attempt for nonexistent user
Thu Dec  5 17:44:01 2024 authpriv.info dropbear[12396]: Exit before auth from <125.70.52.249:61446>: Max auth tries reached - user 'is invalid'
Thu Dec  5 17:44:01 2024 authpriv.info dropbear[12400]: Child connection from 125.70.52.249:61478
Thu Dec  5 17:44:02 2024 authpriv.warn dropbear[12400]: Login attempt for nonexistent user
Thu Dec  5 17:44:03 2024 authpriv.info dropbear[12400]: Exit before auth from <125.70.52.249:61478>: Max auth tries reached - user 'is invalid'
Thu Dec  5 17:44:03 2024 authpriv.info dropbear[12404]: Child connection from 125.70.52.249:61505
Thu Dec  5 17:44:03 2024 authpriv.warn dropbear[12404]: Login attempt for nonexistent user
Thu Dec  5 17:44:04 2024 authpriv.info dropbear[12404]: Exit before auth from <125.70.52.249:61505>: Max auth tries reached - user 'is invalid'
Thu Dec  5 17:44:04 2024 authpriv.info dropbear[12408]: Child connection from 125.70.52.249:61532
Thu Dec  5 17:44:04 2024 authpriv.warn dropbear[12408]: Login attempt for nonexistent user
Thu Dec  5 17:44:05 2024 authpriv.info dropbear[12408]: Exit before auth from <125.70.52.249:61532>: Max auth tries reached - user 'is invalid'
Thu Dec  5 17:44:05 2024 authpriv.info dropbear[12412]: Child connection from 125.70.52.249:61557
Thu Dec  5 17:44:06 2024 authpriv.warn dropbear[12412]: Login attempt for nonexistent user
Thu Dec  5 17:44:07 2024 authpriv.info dropbear[12412]: Exit before auth from <125.70.52.249:61557>: Max auth tries reached - user 'is invalid'
Thu Dec  5 17:44:07 2024 authpriv.info dropbear[12416]: Child connection from 125.70.52.249:61584
Thu Dec  5 17:44:07 2024 authpriv.warn dropbear[12416]: Login attempt for nonexistent user
Thu Dec  5 17:44:08 2024 authpriv.info dropbear[12416]: Exit before auth from <125.70.52.249:61584>: Error reading: Connection reset by peer

CYaNu, 2024-12-06:
被攻击了 和固件无关

drw2005, 2024-12-06:
> 被攻击了 和固件无关

原来如此，谢谢指点。

closed 2024-12-06 by drw2005