coolsnowwolf/lede #13176: 好久没编译L大的固件，今天发现想把Luci-app-vssr编译进固件里时出现了iptables依赖的错误

guolongchu, 2024-12-06:
### 详细叙述

在使用make menuconfig编译前保存config时就会提示以下错误
tmp/.config-package.in:99842:error: recursive dependency detected!
tmp/.config-package.in:99842:	symbol PACKAGE_iptables is selected by PACKAGE_miniupnpd
tmp/.config-package.in:100680:	symbol PACKAGE_miniupnpd is selected by PACKAGE_luci-app-upnp
tmp/.config-package.in:63965:	symbol PACKAGE_luci-app-upnp depends on PACKAGE_rpcd
tmp/.config-package.in:1142:	symbol PACKAGE_rpcd is selected by PACKAGE_rpcd-mod-luci
tmp/.config-package.in:60619:	symbol PACKAGE_rpcd-mod-luci is selected by PACKAGE_luci-base
tmp/.config-package.in:60994:	symbol PACKAGE_luci-base is selected by PACKAGE_luci-lua-runtime
tmp/.config-package.in:61146:	symbol PACKAGE_luci-lua-runtime is selected by PACKAGE_luci-app-netdata
tmp/.config-package.in:62518:	symbol PACKAGE_luci-app-netdata depends on PACKAGE_coreutils
tmp/.config-package.in:136821:	symbol PACKAGE_coreutils is selected by PACKAGE_luci-app-vssr
tmp/.config-package.in:64129:	symbol PACKAGE_luci-app-vssr depends on PACKAGE_iptables
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

似乎是跟iptables有关，我尝试强制编译，也没出错，但是helloword里面无法添加节点，点击添加节点页面出现bad way
哪位大佬能指点迷津？

这个插件作者已经很久没更新，但是界面似乎还是挺好用的！各位大佬帮帮忙指点一下，问题出在哪里？

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

X86

### 详细日志

tmp/.config-package.in:99842:error: recursive dependency detected!
tmp/.config-package.in:99842:	symbol PACKAGE_iptables is selected by PACKAGE_miniupnpd
tmp/.config-package.in:100680:	symbol PACKAGE_miniupnpd is selected by PACKAGE_luci-app-upnp
tmp/.config-package.in:63965:	symbol PACKAGE_luci-app-upnp depends on PACKAGE_rpcd
tmp/.config-package.in:1142:	symbol PACKAGE_rpcd is selected by PACKAGE_rpcd-mod-luci
tmp/.config-package.in:60619:	symbol PACKAGE_rpcd-mod-luci is selected by PACKAGE_luci-base
tmp/.config-package.in:60994:	symbol PACKAGE_luci-base is selected by PACKAGE_luci-lua-runtime
tmp/.config-package.in:61146:	symbol PACKAGE_luci-lua-runtime is selected by PACKAGE_luci-app-netdata
tmp/.config-package.in:62518:	symbol PACKAGE_luci-app-netdata depends on PACKAGE_coreutils
tmp/.config-package.in:136821:	symbol PACKAGE_coreutils is selected by PACKAGE_luci-app-vssr
tmp/.config-package.in:64129:	symbol PACKAGE_luci-app-vssr depends on PACKAGE_iptables

yanxin152133, 2024-12-06:
https://github.com/coolsnowwolf/lede/issues/13104

guolongchu, 2024-12-06:
> #13104

你是指将13104行注释掉？

guolongchu, 2024-12-06:
明白

closed 2025-01-27 by aiamadeus