coolsnowwolf/lede #13183: ERROR: package/feeds/helloworld/shadowsocksr-libev failed to build.

BenjaminX, 2024-12-07:
### 详细叙述

编译 R4S 时 hellworld 包报错

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

R4S 和 X86

### 详细日志

[image]
[logs_31775957436.zip](https://github.com/user-attachments/files/18047923/logs_31775957436.zip)

yanxin152133, 2024-12-09:
参考下这个？
https://github.com/fw876/helloworld/issues/1434

BenjaminX, 2024-12-09:
换了后编译过了
又提示这个

2024-12-09T06:51:51.4814193Z Collected errors:
2024-12-09T06:51:51.4815099Z  * check_data_file_clashes: Package libustream-openssl20201210 wants to install file /workdir/openwrt/build_dir/target-aarch64_generic_musl/root-rockchip/lib/libustream-ssl.so
2024-12-09T06:51:51.4816181Z 	But that file is already provided by package  * libustream-mbedtls20201210
2024-12-09T06:51:51.4816751Z  * opkg_install_cmd: Cannot install package libustream-openssl20201210.
2024-12-09T06:51:51.4818070Z  * check_data_file_clashes: Package libustream-openssl20201210 wants to install file /workdir/openwrt/build_dir/target-aarch64_generic_musl/root-rockchip/lib/libustream-ssl.so
2024-12-09T06:51:51.4819199Z 	But that file is already provided by package  * libustream-mbedtls20201210
2024-12-09T06:51:51.4819739Z  * opkg_install_cmd: Cannot install package luci-app-ssr-plus.
2024-12-09T06:51:51.4820146Z Configuring iptables-mod-tproxy.

yanxin152133, 2024-12-09:
> 换了后编译过了 又提示这个
>
> 2024-12-09T06:51:51.4814193Z Collected errors: 2024-12-09T06:51:51.4815099Z * check_data_file_clashes: Package libustream-openssl20201210 wants to install file [...]

万能大法？

closed 2024-12-17 by BenjaminX