coolsnowwolf/lede #13187: vlmcsd 编译失败

Gzxhwq, 2024-12-08:
### 详细叙述

vlmcsd 编译失败

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

ALL

### 详细日志

~~~make[3]: Entering directory '/home/runner/work/OpenWrt-Actions/OpenWrt-Actions/openwrt/feeds/packages/net/vlmcsd'
rm -f /home/runner/work/OpenWrt-Actions/OpenWrt-Actions/openwrt/build_dir/target-x86_64_musl/vlmcsd-svn1113/.built
touch /home/runner/work/OpenWrt-Actions/OpenWrt-Actions/openwrt/build_dir/target-x86_64_musl/vlmcsd-svn1113/.built_check
CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/runner/work/OpenWrt-Actions/OpenWrt-Actions/openwrt/build_dir/target-x86_64_musl/vlmcsd-svn1113=vlmcsd-svn1113 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/runner/work/OpenWrt-Actions/OpenWrt-Actions/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/usr/include -I/home/runner/work/OpenWrt-Actions/OpenWrt-Actions/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/include -I/home/runner/work/OpenWrt-Actions/OpenWrt-Actions/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/include/fortify " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/runner/work/OpenWrt-Actions/OpenWrt-Actions/openwrt/build_dir/target-x86_64_musl/vlmcsd-svn1113=vlmcsd-svn1113 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/runner/work/OpenWrt-Actions/OpenWrt-Actions/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/usr/include -I/home/runner/work/OpenWrt-Actions/OpenWrt-Actions/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/include -I/home/runner/work/OpenWrt-Actions/OpenWrt-Actions/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/include/fortify " LDFLAGS="-L/home/runner/work/OpenWrt-Actions/OpenWrt-Actions/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/usr/lib -L/home/runner/work/OpenWrt-Actions/OpenWrt-Actions/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/lib -fuse-ld=bfd -znow -zrelro " make  -C /home/runner/work/OpenWrt-Actions/OpenWrt-Actions/openwrt/build_dir/target-x86_64_musl/vlmcsd-svn1113/. AR="x86_64-openwrt-linux-musl-gcc-ar" AS="ccache x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/runner/work/OpenWrt-Actions/OpenWrt-Actions/openwrt/build_dir/target-x86_64_musl/vlmcsd-svn1113=vlmcsd-svn1113 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD="x86_64-openwrt-linux-musl-ld.bfd" NM="x86_64-openwrt-linux-musl-gcc-nm" CC="ccache x86_64-openwrt-linux-musl-gcc" GCC="ccache x86_64-openwrt-linux-musl-gcc" CXX="ccache x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CROSS="x86_64-openwrt-linux-musl-" ARCH="x86_64" -C /home/runner/work/OpenWrt-Actions/OpenWrt-Actions/openwrt/build_dir/target-x86_64_musl/vlmcsd-svn1113 ;
make[4]: Entering directory '/home/runner/work/OpenWrt-Actions/OpenWrt-Actions/openwrt/build_dir/target-x86_64_musl/vlmcsd-svn1113'
make[5]: Entering directory '/home/runner/work/OpenWrt-Actions/OpenWrt-Actions/openwrt/build_dir/target-x86_64_musl/vlmcsd-svn1113/src'
	CC	vlmcs.o <- vlmcs.c
	CC	kmsdata-full.o <- kmsdata-full.c
	CC	crypto.o <- crypto.c
	CC	kms.o <- kms.c
	CC	endian.o <- endian.c
	CC	output.o <- output.c
	CC	shared_globals.o <- shared_globals.c
x86_64-openwrt-linux-musl-gcc: fatal error: cannot specify '-o' with '-c', '-S' or '-E' with multiple files
compilation terminated.
make[5]: *** [GNUmakefile:512: ../build/kmsdata-full.o] Error 1
	CC	helpers.o <- helpers.c
make[5]: *** Waiting for unfinished jobs....
x86_64-openwrt-linux-musl-gcc: fatal error: cannot specify '-o' with '-c', '-S' or '-E' with multiple files
compilation terminated.
make[5]: *** [GNUmakefile:513: ../build/crypto.o] Error 1
x86_64-openwrt-linux-musl-gcc: fatal error: cannot specify '-o' with '-c', '-S' or '-E' with multiple files
compilation terminated.
make[5]: *** [GNUmakefile:513: ../build/vlmcs.o] Error 1
x86_64-openwrt-linux-musl-gcc: fatal error: cannot specify '-o' with '-c', '-S' or '-E' with multiple files
compilation terminated.
make[5]: *** [GNUmakefile:513: ../build/endian.o] Error 1
x86_64-openwrt-linux-musl-gcc: fatal error: cannot specify '-o' with '-c', '-S' or '-E' with multiple files
compilation terminated.
make[5]: *** [GNUmakefile:513: ../build/output.o] Error 1
x86_64-openwrt-linux-musl-gcc: fatal error: cannot specify '-o' with '-c', '-S' or '-E' with multiple files
compilation terminated.
make[5]: *** [GNUmakefile:513: ../build/helpers.o] Error 1
x86_64-openwrt-linux-musl-gcc: fatal error: cannot specify '-o' with '-c', '-S' or '-E' with multiple files
compilation terminated.
make[5]: *** [GNUmakefile:513: ../build/kms.o] Error 1
x86_64-openwrt-linux-musl-gcc: fatal error: cannot specify '-o' with '-c', '-S' or '-E' with multiple files
compilation terminated.
make[5]: *** [GNUmakefile:513: ../build/shared_globals.o] Error 1
make[5]: Leaving directory '/home/runner/work/OpenWrt-Actions/OpenWrt-Actions/openwrt/build_dir/target-x86_64_musl/vlmcsd-svn1113/src'
make[4]: *** [GNUmakefile:111: all] Error 2
make[4]: Leaving directory '/home/runner/work/OpenWrt-Actions/OpenWrt-Actions/openwrt/build_dir/target-x86_64_musl/vlmcsd-svn1113'
make[3]: *** [Makefile:44: /home/runner/work/OpenWrt-Actions/OpenWrt-Actions/openwrt/build_dir/target-x86_64_musl/vlmcsd-svn1113/.built] Error 2
make[3]: Leaving directory '/home/runner/work/OpenWrt-Actions/OpenWrt-Actions/openwrt/feeds/packages/net/vlmcsd'
time: package/feeds/packages/vlmcsd/compile#0.31#0.13#0.25
    ERROR: package/feeds/packages/vlmcsd failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/vlmcsd/compile] Error 1
make[2]: Leaving directory '/home/runner/work/OpenWrt-Actions/OpenWrt-Actions/openwrt'
make[1]: *** [package/Makefile:110: /home/runner/work/OpenWrt-Actions/OpenWrt-Actions/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/runner/work/OpenWrt-Actions/OpenWrt-Actions/openwrt'
make: *** [/home/runner/work/OpenWrt-Actions/OpenWrt-Actions/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

liwenjie119, 2024-12-08:
同样错误，github action

closed 2024-12-08 by aiamadeus

ZqinKing, 2024-12-26:
开启ccache后会出现这个问题。
报错GNUmakefile为：
+@$(CC) -x$(COMPILER_LANGUAGE) $(PLATFORMFLAGS) $(BASECFLAGS) $(CFLAGS) $(PLATFORMFLAGS) -c $< -o $@
实际指令如下：
ccache aarch64-openwrt-linux-musl-gcc -xc  -DVLMCSD_COMPILER="ccache aarch64-openwrt-linux-musl-gcc" -DVLMCSD_PLATFORM="aarch64-openwrt-linux-musl" -DCONFIG="config.h" -DBUILD_TIME=1735192751 -g -Os -fno-strict-aliasing -fomit-frame-pointer -ffunction-sections -fdata-sections -fvisibility=hidden -pipe -fno-common -fno-exceptions -fno-stack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fmerge-all-constants -flto -Wall -D_CRYPTO_INTERNAL -O2 -pipe -mcpu=cortex-a53+crc+crypto -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/zqinking/wrt_relese/imm-nss/build_dir/target-aarch64_cortex-a53_musl/vlmcsd-2020.03.30~e5990804=vlmcsd-2020.03.30~e5990804 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/zqinking/wrt_relese/imm-nss/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/usr/include -I/home/zqinking/wrt_relese/imm-nss/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/include -I/home/zqinking/wrt_relese/imm-nss/staging_dir/toolchain-aarch64_cortex-a53_gcc-13.3.0_musl/include/fortify   -c vlmcs.c -o ../build/vlmcs.o