coolsnowwolf/lede #13191: E20C 编译不通过，看错误似乎是uboot问题

doyoman, 2024-12-09:
### 详细叙述

actions里面编译，报错如下，请佬指点一下

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

radxa e20c

### 详细日志

echo "uboot-envtools" >> /workdir/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/uboot-envtools.default.install
make[3]: Leaving directory '/workdir/openwrt/package/boot/uboot-envtools'
time: package/boot/uboot-envtools/compile#0.24#0.25#0.37
make[3]: Entering directory '/workdir/openwrt/package/boot/uboot-rk35xx'
rm -f /workdir/openwrt/build_dir/target-aarch64_generic_musl/uboot-rk35xx-evb-rk3528/uboot-rk35xx-2024-10-29-27398f1e/.built
touch /workdir/openwrt/build_dir/target-aarch64_generic_musl/uboot-rk35xx-evb-rk3528/uboot-rk35xx-2024-10-29-27398f1e/.built_check
make  -C /workdir/openwrt/build_dir/target-aarch64_generic_musl/uboot-rk35xx-evb-rk3528/uboot-rk35xx-2024-10-29-27398f1e CROSS_COMPILE=aarch64-openwrt-linux-musl-  HOSTCC="/workdir/openwrt/staging_dir/host/bin/gcc" HOSTCFLAGS="-O2 -I/workdir/openwrt/staging_dir/host/include -I/workdir/openwrt/staging_dir/hostpkg/include -I/workdir/openwrt/staging_dir/target-aarch64_generic_musl/host/include -I/workdir/openwrt/staging_dir/host/include -I/workdir/openwrt/staging_dir/hostpkg/include -I/workdir/openwrt/staging_dir/target-aarch64_generic_musl/host/include -std=gnu11" HOSTLDFLAGS="-L/workdir/openwrt/staging_dir/host/lib -L/workdir/openwrt/staging_dir/hostpkg/lib -L/workdir/openwrt/staging_dir/target-aarch64_generic_musl/host/lib" LOCALVERSION="-OpenWrt-r7087-e66ffaf37" STAGING_PREFIX="/workdir/openwrt/staging_dir/host" PKG_CONFIG_PATH="/workdir/openwrt/staging_dir/host/lib/pkgconfig" PKG_CONFIG_LIBDIR="/workdir/openwrt/staging_dir/host/lib/pkgconfig" PKG_CONFIG_EXTRAARGS="--static" V='' BL31=/workdir/openwrt/stagin
make[4]: Entering directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/uboot-rk35xx-evb-rk3528/uboot-rk35xx-2024-10-29-27398f1e'
  CHK     include/config/uboot.release
  CHK     include/generated/version_autogenerated.h
  CHK     include/generated/timestamp_autogenerated.h
  CHK     include/generated/generic-asm-offsets.h
  CHK     include/generated/asm-offsets.h
  CC      common/edid.o
In function 'do_hdmi_vsdb_modes',
    inlined from 'add_cea_modes' at common/edid.c:3617:12,
    inlined from 'drm_add_edid_modes' at common/edid.c:5542:15:
common/edid.c:2934:17: error: 'hdmi_len' may be used uninitialized [-Werror=maybe-uninitialized]
 2[934](https://github.com/doyoman/Actions-OpenWrt/actions/runs/12228172199/job/34105993602#step:11:935) |         if (len < (8 + offset + 2))
      |             ~~~~^~~~~~~~~~~~~~~~~~
common/edid.c: In function 'drm_add_edid_modes':
common/edid.c:3582:17: note: 'hdmi_len' was declared here
 3582 |         u8 dbl, hdmi_len, video_len = 0;
      |                 ^~~~~~~~
cc1: all warnings being treated as errors
make[5]: *** [scripts/Makefile.build:281: common/edid.o] Error 1
make[4]: *** [Makefile:1314: common] Error 2
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/uboot-rk35xx-evb-rk3528/uboot-rk35xx-2024-10-29-27398f1e'
make[3]: *** [Makefile:89: /workdir/openwrt/build_dir/target-aarch64_generic_musl/uboot-rk35xx-evb-rk3528/uboot-rk35xx-2024-10-29-27398f1e/.built] Error 2
make[3]: Leaving directory '/workdir/openwrt/package/boot/uboot-rk35xx'
time: package/boot/uboot-rk35xx/evb-rk3528/compile#1.58#0.39#1.90
    ERROR: package/boot/uboot-rk35xx failed to build (build variant: evb-rk3528).
make[2]: *** [package/Makefile:116: package/boot/uboot-rk35xx/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2

yanxin152133, 2024-12-09:
https://github.com/coolsnowwolf/lede/issues/12411

doyoman, 2024-12-09:
> #12411
不是这个问题，Makefile早就被修改了很多版了，这个里面提到的这行已经没有了，这个貌似使用的uboot源码有问题

closed 2024-12-09 by aiamadeus