coolsnowwolf/lede #13192: 最新源码编译完成开机后系统日志中报大量`modprobe segfault`

devnakx, 2024-12-09:
### 详细叙述

[image]
详细日志在下面

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86-64 PVE VM

### 详细日志

[segfault.log](https://github.com/user-attachments/files/18056511/segfault.log)

devnakx, 2024-12-09:
而且还在不停地报，到现在已经刷了很多在日志里了

coolsnowwolf, 2024-12-10:
这个是你第三方的 kmod 的问题了，例如上网过滤之类

devnakx, 2024-12-10:
[config.log](https://github.com/user-attachments/files/18083256/config.log)
@coolsnowwolf 我编译用的config，因为github限制后缀名，所以改成.log了

devnakx, 2024-12-12:
@coolsnowwolf 我把`.config`删了重做了就没报了，但是每次`reboot`或者`poweroff`也会报一个`netifd segfault`
这个之前就一直有，我之前编译勾选了`openssh-sftp-avahi-service`后就没报了，但现在这个选上了也依然报
[image]
[image]

closed 2024-12-14 by devnakx