coolsnowwolf/lede #13194: luci首页显示可以优化一下。

yulinsoft, 2024-12-09:
### 详细叙述

luci首页显示可以优化一下。现在显示有两个温度。
[image]

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

N1

### 详细日志

界面优化

xingshu1990, 2024-12-10:
可自行修改：/usr/lib/lua/luci/view/admin_status/index.htm
谷歌搜索：温度 /usr/lib/lua/luci/view/admin_status/index.htm 或者openwrt /usr/lib/lua/luci/view/admin_status/index.htm

位置
[image]

yulinsoft, 2024-12-11:
感谢你的回复。

> 可自行修改：/usr/lib/lua/luci/view/admin_status/index.htm 谷歌搜索 /usr/lib/lua/luci/view/admin_status/index.htm 或者openwrt /usr/lib/lua/luci/view/admin_status/index.htm 至于里面具体怎么修改，要么看别人的教程，要么先全文丢给AI让其翻译，或者直接修改 然后先自行测试。 注意此文档先备份，以便后面覆盖。 免得修改后，不知道修改了什么地方，又不知道怎么修复。

closed 2024-12-11 by yulinsoft