coolsnowwolf/lede #13197: make download还是有问题

hicom-omni, 2024-12-10:
### 详细叙述

好多天了，每次到下载这2个包就不行

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86-64

### 详细日志

[image]
[image]

yanxin152133, 2024-12-10:
这个问题应该前几天刚提过
https://github.com/coolsnowwolf/lede/issues/13173#issuecomment-2521911246

xingshu1990, 2024-12-10:
> 这个问题应该前几天刚提过 [#13173 (comment)](https://github.com/coolsnowwolf/lede/issues/13173#issuecomment-2521911246)

是提过，不过我不知道这个编译是哪个插件，所以不知道上面我说的内容，是否可行_,

yanxin152133, 2024-12-10:
[image]

hicom-omni, 2024-12-10:
> > 这个问题应该前几天刚提过 [#13173 (comment)](https://github.com/coolsnowwolf/lede/issues/13173#issuecomment-2521911246)
> 
> 是提过，不过我不知道这个编译是哪个插件，所以不知道上面我说的内容，是否可行_,

怎么解决的？

xingshu1990, 2024-12-10:
> > > 这个问题应该前几天刚提过 [#13173 (comment)](https://github.com/coolsnowwolf/lede/issues/13173#issuecomment-2521911246)
> > 
> > 
> > 是提过，不过我不知道这个编译是哪个插件，所以不知道上面我说的内容，是否可行_,
> 
> 怎么解决的？

就是上面另外那位朋友发的，我在已经结贴的帖子里的回复：自行尝试解压，单独编译。

coolsnowwolf, 2024-12-10:
这个是你用了第三方的库的问题

aiamadeus, 2024-12-10:
[image]
单纯你网络问题

hicom-omni, 2024-12-11:
> 这个是你用了第三方的库的问题

腾讯云硅谷或HK的机器，16核32G，22.04 Ubuntu系统

安装依赖时出现2次粉色屏幕全部直接回车

在cd lede后输入2条命令：
sed -i "/helloworld/d" "feeds.conf.default"
echo "src-git helloworld https://github.com/fw876/helloworld.git;main" >> "feeds.conf.default"

进入make menuconfig后未修改任何内容直接保存

make download -j8还是报错

hicom-omni, 2024-12-11:
> 这个是你用了第三方的库的问题
换用阿里云下载成功了，哎

closed 2024-12-13 by aiamadeus

coolveyron, 2024-12-19:
> > 这个是你用了第三方的库的问题
> > 换用阿里云下载成功了，哎

我已经找到原因了，与第三方库关系也不大，主要问题还在腾讯云上，你试试直接用OpenWrt官方的包进行编译看看，什么也不改，git下来直接按照步骤编译，你就知道为啥了。嘿嘿!

hicom-omni, 2024-12-23:
> > > 这个是你用了第三方的库的问题
> > > 换用阿里云下载成功了，哎
> 
> 我已经找到原因了，与第三方库关系也不大，主要问题还在腾讯云上，你试试直接用OpenWrt官方的包进行编译看看，什么也不改，git下来直接按照步骤编译，你就知道为啥了。嘿嘿!

年纪大了，懒得搞了，啥原因啊？

coolveyron, 2025-01-22:
> > > > 这个是你用了第三方的库的问题
> > > > 换用阿里云下载成功了，哎
> > 
> > 
> > 我已经找到原因了，与第三方库关系也不大，主要问题还在腾讯云上，你试试直接用OpenWrt官方的包进行编译看看，什么也不改，git下来直接按照步骤编译，你就知道为啥了。嘿嘿!
> 
> 年纪大了，懒得搞了，啥原因啊？

腾讯云的所有服务器已经无法正常编译openwrt，换阿里云吧。所有腾讯云的服务器都无法下载部分组件。

核心问题在git.openwrt.org上，能ping通，但是返回的是假地址，是无法从git.openwrt.org中获取正确的组件hash，所以下载都出错。

已经问过腾讯云的客服了，他们知道，说是运营商的问题，正在积极协调解决。

美区？香港？日本？都是运营商的问题？我没搞明白他们的网络出口的架构是啥样的。还是服务器都在国内，走GFW后再通过他们自己的专线到其他境外服务器？GFW给屏蔽了？但是阿里云没问题啊。

不懂腾讯云的操作是不是故意而为之。